esp8266/Arduino #1381: LIGHT_SLEEP HOW ?

donnib, 2016-01-07:
Hi,
How is it possible to light sleep with arduino framework on the ESP8266, i know how to deepsleep and it works but i am interested to do LIGHT_SLEEP so i can benefit from "low power" equal to 0.5mA and have GPIO wake up enabled with gpio_pin_wakeup_enable. 

I can't seem to find examples how to do this, anyone tried it ?

donnib

stevescot, 2016-01-08:
Here's what I currently have, it runs, but does not go to Light_sleep.

___________________________________________________

#include "ESP8266WiFi.h"
#include "gpio.h"
extern "C" {
#include "user_interface.h"
  uint16 readvdd33(void);
  bool wifi_set_sleep_type(sleep_type_t);
  sleep_type_t wifi_get_sleep_type(void);
}
bool sleep = false;

int switchPin = 2;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("before Sleep");
  goToLightSleep();
  Serial.println("Woken up");
}

void loop() {
  // put your main code here, to run repeatedly:

}

void goToLightSleep()
{
  if (!sleep)
  {
    Serial.println("switching off Wifi Modem and CPU");
    //stop any clients
    ///thisclient.stop();
    delay(1000);
    wifi_set_sleep_type(LIGHT_SLEEP_T);
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    // how do we call this: 
    //gpio_pin_wakeup_enable(GPIO_ID_PIN(switchPin),GPIO_PIN_INTR_NEGEDGE);
    wifi_fpm_open();
    wifi_fpm_do_sleep(26843455);
    //if(WiFi.forceSleepBegin(26843455)) sleep = true;
  }
}

antelder, 2016-01-08:
From what i understand light sleep is on by default but only happens when the ESP is in station mode and connected to an access point. The default mode is WIFI_AP_STA so you need to call WiFi.mode(WIFI_STA) and then connect to an access point. I did test this out and it did seem to work and reduce power consumption to a bit less than 1 mA  during inactivity, but that was quite a while back now and i've not verified it still works with the lastest ESP/Arduino code and SDK.

stevescot, 2016-01-08:
antelder - I think you might be talking about modem_sleep where power use goes down to 20mA, - that is on by default but has no effect for me, so I force modem sleep when I don't need wifi.

Light_sleep requires a pin interrupt to wake it, and switches off the CPU and WiFi, so it  can't wake itself, 

http://bbs.espressif.com/viewtopic.php?t=133

antelder, 2016-01-08:
I don't think so but i admit all this area can be confusing and is not well documented so i may be mistaken. However,  i'm pretty sure i've tried both those and what i described above is how i got light sleep to work and the power consumption was matching what they say it should be at a bit under 1 mA. Where have you seen it said that light sleep requires a pin interrupt to wake up?

Links2004, 2016-01-08:
Modem sleep is working an implemented in staring / git.
usage:
when you want to sleep:
```cpp
WiFi.forceSleepBegin();
```

and to enable WiFi again:
```cpp
WiFi.forceSleepWake();
```

more info staring here:
https://github.com/esp8266/Arduino/issues/460#issuecomment-168775857
#460

donnib, 2016-01-08:
@stevescot yes exactly what i am looking for. I can't find somebody that got this to work yet.

torntrousers, 2016-01-08:
I dug up the old sketch i had for testing light sleep and tried it again with the current 2.0.0 release. WIth this sketch light sleep definately only seems to kick in when in station mode and Wifi is connected and i see a minimum current of about 1 mA. Its a bit hard to see the current use on a multimeter as the current dances around (which is what you'd expect with light sleep, right?, as it keeps waking up every 300 milliseconds or so)  but its most of the time showing well under 10mA:
```
#include <ESP8266WiFi.h>
extern "C" {
  #include "user_interface.h"
}

const char* ssid = "BTHub5-72W5";
const char* password = "xxxxxxxxxx";

void setup() {
  Serial.begin(115200);
  Serial.println();
}

void loop() {

  initWifi();
  
  Serial.println("Light sleep:");
  wifi_set_sleep_type(LIGHT_SLEEP_T);
  doDelays();
  
  Serial.println("None sleep:");
  wifi_set_sleep_type(NONE_SLEEP_T);
  doDelays();

  WiFi.disconnect();
  Serial.print("WiFi disconnected, IP address: "); Serial.println(WiFi.localIP());
  Serial.println("Light sleep:");
  wifi_set_sleep_type(LIGHT_SLEEP_T);
  doDelays();
  
}

void doDelays() {
  Serial.println("Yield for 20 sec");
  long endMs = millis() + 20000;
  while (millis() < endMs) {
     yield();
  }

  Serial.println("Delay for 20 sec");
  delay(20000);
}

void initWifi() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while ((WiFi.status() != WL_CONNECTED)) {
     delay(500);
     Serial.print(".");
  }
  Serial.println("");
  Serial.print("WiFi connected, IP address: "); Serial.println(WiFi.localIP());
}
```

stevescot, 2016-01-08:
That sounds great - will try it...

Links2004, 2016-01-09:
Note: sleep in AP mode is not possible, since the connected client can send something to any time.
in STA mode the WiFi standard allows the client to sleep (the ESP)

stevescot, 2016-01-10:
Ran this script findings were:

with light_sleep, when delay is called the modem, and cpu will shut down (going down to 1mA and below for small periods between beacons, )
however - the yield time shows 70mA

stevescot, 2016-01-10:
Setting to modem_sleep - still when delay is called then power drops to 20mA, given that sleep only happens when you have called delay (then you are not using the CPU)- no reason not to set light sleep.

WereCatf, 2016-01-11:
I am sorry for butting in on the conversation, but I would also be interested in learning how to make use of light-sleep. I just did some testing I can get modem-sleep to work just fine, but light-sleep only works when the ESP is actively connected to the WIFI-network -- I can't get light-sleep to work if I disconnect from the network.

If the ESP is connected to the network the current-draw keeps jumping up and down during delay() and overall current-draw won't be anywhere near Espressif's claims. On the other hand, disconnecting from the network and forcing modem-sleep works fine -- current-draw stays stable <17mA during delay(). Disconnecting from the network and trying to enter light-sleep either draws current as if the was no power-management at all enabled, ie. >50mA or current-draw stays at <17mA if I force modem-sleep after disconnect.

Is it even possible to get light-sleep to work when disconnected so as to get power-draw down during longer periods of sleep?

donnib, 2016-01-11:
I still don't get it, how can light_sleep work without having the GPIO
enable wake ? In the manual
http://russ.russmathis.com/wp-content/uploads/2015/09/9B-ESP8266__Sleep__Function-Description__EN_v1.0.pdf
it is stated that the CPU is in suspend and the clock is off so one need to
wake up the device with GPIO out of light_sleep or ?

stevescot, 2016-01-11:
Yeah that confused me - my assumption is that light sleep as implemented by the core code uses the real-time clock to do an interrupt - so it is not running code and not using WiFi during these periods. However if you manually put the chip into light sleep then you will need to generate an interrupt to wake. That is all empirical - haven't seen documents to tell me that its just what seems to make sense.

WereCatf, 2016-01-11:
@donnib: The PDF says "Under the light-sleep mode, the CPU will not respond to the signal and interrupt from the peripheral hardware interface under the pause state. Therefore, the ESP8266 need to be waked up via external GPIO, and the wake process is less than 3 ms."

I read that as them saying that any interrupts you've attached to GPIO-pin state-changes won't work under light-sleep, but you can designate one GPIO-pin that the ESP will monitor in order to wake up from light-sleep and then it will be able to handle those interrupts. Internal mechanisms based on WIFI and RTC, however, will keep running and working even during light-sleep.

My interpretation may be completely off the base, but that's how I read it.

mihino, 2016-01-16:
@torntrousers Thanks! I've spent lot of time to lower current draw by LIGHT_SLEEP_T and the trick is to NOT use WiFi.disconnect();
finally my working the code is:
```
// Required for LIGHT_SLEEP_T delay mode
extern "C" {
#include "user_interface.h"
}

...

    //WiFi.disconnect(); // DO NOT DISCONNECT WIFI IF YOU WANT TO LOWER YOUR POWER !
    wifi_set_sleep_type(LIGHT_SLEEP_T);
    delay(20000); // 20s
```

pigglet, 2016-02-05:
After some head scratching, many many cups of tea and some beer this is my implementation of a stable mA light sleep with gpio_pin_wakeup_enable(GPIO_ID_PIN(1), GPIO_PIN_INTR_HILEVEL);
uses  staring / git. for WiFi.forceSleepBegin();
`#include  <ESP8266WiFi.h>
//include  ESP8266WiFi.h needed doesnt show up on post code
extern "C" {
  #include "e:/gpio.h" //change to where you have saved gpio.h listed below
}
extern "C" {
  #include "user_interface.h"
}
const char* ssid = "SKY8606B";
const char* password = "*******";
int val = 0;

WiFiClient client;
void setup() {
  // put your setup code here, to run once:
pinMode(5, INPUT);           // set pin to input
digitalWrite(5, LOW);       // turn off pullup resistors

Serial.begin(115200);
delay(10);

WiFi.begin(ssid, password);

Serial.println();
Serial.println();
Serial.print("Connecting to ");
Serial.println(ssid);

WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("WiFi connected");
//gpio_pin_wakeup_disable();
gpio_pin_wakeup_enable(GPIO_ID_PIN(1), GPIO_PIN_INTR_HILEVEL);
wifi_set_sleep_type(LIGHT_SLEEP_T);
}

void loop() {
  // put your main code here, to run repeatedly:
val = digitalRead(5);
while (val == 1){
  // do something repetitive
Serial.println("Pin 5 high"); 
delay (20000);
}
WiFi.forceSleepBegin();
delay(20000); // 20s
//ESP.deepSleep(60000000, WAKE_RF_DEFAULT); // Sleep for 60 seconds
}`
Code needed for gpio_pin_wakeup_enable(GPIO_ID_PIN(1), GPIO_PIN_INTR_HILEVEL); needs to be saved as gpio.h

`/*
 * copyright (c) Espressif System 2010
 *
 */

#ifndef _GPIO_H_
#define _GPIO_H_

#define GPIO_PIN_ADDR(i) (GPIO_PIN0_ADDRESS + i*4)

#define GPIO_ID_IS_PIN_REGISTER(reg_id) \
    ((reg_id >= GPIO_ID_PIN0) && (reg_id <= GPIO_ID_PIN(GPIO_PIN_COUNT-1)))

#define GPIO_REGID_TO_PINIDX(reg_id) ((reg_id) - GPIO_ID_PIN0)

typedef enum {
    GPIO_PIN_INTR_DISABLE = 0,
    GPIO_PIN_INTR_POSEDGE = 1,
    GPIO_PIN_INTR_NEGEDGE = 2,
    GPIO_PIN_INTR_ANYEDGE = 3,
    GPIO_PIN_INTR_LOLEVEL = 4,
    GPIO_PIN_INTR_HILEVEL = 5
} GPIO_INT_TYPE;

#define GPIO_OUTPUT_SET(gpio_no, bit_value) \
    gpio_output_set((bit_value)<<gpio_no, ((~(bit_value))&0x01)<<gpio_no, 1<<gpio_no,0)
#define GPIO_DIS_OUTPUT(gpio_no) 	gpio_output_set(0,0,0, 1<<gpio_no)
#define GPIO_INPUT_GET(gpio_no)     ((gpio_input_get()>>gpio_no)&BIT0)

/* GPIO interrupt handler, registered through gpio_intr_handler_register */
typedef void (* gpio_intr_handler_fn_t)(uint32 intr_mask, void *arg);

/*
 * Initialize GPIO.  This includes reading the GPIO Configuration DataSet
 * to initialize "output enables" and pin configurations for each gpio pin.
 * Must be called once during startup.
 */
void gpio_init(void);

/*
 * Change GPIO pin output by setting, clearing, or disabling pins.
 * In general, it is expected that a bit will be set in at most one
 * of these masks.  If a bit is clear in all masks, the output state
 * remains unchanged.
 *
 * There is no particular ordering guaranteed; so if the order of
 * writes is significant, calling code should divide a single call
 * into multiple calls.
 */
void gpio_output_set(uint32 set_mask,
                     uint32 clear_mask,
                     uint32 enable_mask,
                     uint32 disable_mask);

/*
 * Sample the value of GPIO input pins and returns a bitmask.
 */
uint32 gpio_input_get(void);

/*
 * Set the specified GPIO register to the specified value.
 * This is a very general and powerful interface that is not
 * expected to be used during normal operation.  It is intended
 * mainly for debug, or for unusual requirements.
 */
void gpio_register_set(uint32 reg_id, uint32 value);

/* Get the current value of the specified GPIO register. */
uint32 gpio_register_get(uint32 reg_id);

/*
 * Register an application-specific interrupt handler for GPIO pin
 * interrupts.  Once the interrupt handler is called, it will not
 * be called again until after a call to gpio_intr_ack.  Any GPIO
 * interrupts that occur during the interim are masked.
 *
 * The application-specific handler is called with a mask of
 * pending GPIO interrupts.  After processing pin interrupts, the
 * application-specific handler may wish to use gpio_intr_pending
 * to check for any additional pending interrupts before it returns.
 */
void gpio_intr_handler_register(gpio_intr_handler_fn_t fn, void *arg);

/* Determine which GPIO interrupts are pending. */
uint32 gpio_intr_pending(void);

/*
 * Acknowledge GPIO interrupts.
 * Intended to be called from the gpio_intr_handler_fn.
 */
void gpio_intr_ack(uint32 ack_mask);

void gpio_pin_wakeup_enable(uint32 i, GPIO_INT_TYPE intr_state);

void gpio_pin_wakeup_disable();

void gpio_pin_intr_state_set(uint32 i, GPIO_INT_TYPE intr_state);

#endif // _GPIO_H_`

The reason for setting  gpio_pin_wakeup_enable(GPIO_ID_PIN(1), for pin 5
[http://esp8266.co.uk/tutorials/introduction-to-the-gpio-api/](url)
Yours pigglet

koravadi, 2016-02-26:
@ pigglet thanks for posting the code for light sleep with GPIO. Could you please let me know regarding 
a. do we need to keep polling the GPIO pin to check the IO status?
b. link http://esp8266.co.uk/tutorials/introduction-to-the-gpio-api/ is not working
c. The GPIO code looks like part of lib, will there be a conflict if we use another gpio.h file in our project? "copyright (c) Espressif System 2010"
d. In my application I am developing I want to be continuously stay in light sleep mode <1-2 mA current and wakeup with MQTT message or GPIO level high, please advice how it could be achieved?

Regards, kk

mikeyoyoyo, 2016-03-07:
For anyone else still wondering, the above posters code didn't work for me. I was however able to get light sleep working with a button interrupt for wake up, current draw dropped to 0 mA on my bench power supply. The gpio.h the above poster provided is exactly the same as the one provided in the ESP8266 Arduino library so shouldn't need to save your own.

```C
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

extern "C" {
  #include "gpio.h"
}

extern "C" {
  #include "user_interface.h"
}
void setup()
{
   /* connect to WiFi/normal setup stuff */

  gpio_init(); // Initilise GPIO pins
  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // set sleep type, the above posters wifi_set_sleep_type() didnt seem to work for me although it did let me compile and upload with no errors 
}

void loop()
{
     /* do whatever you want */

    gpio_pin_wakeup_enable(GPIO_ID_PIN(2), GPIO_PIN_INTR_LOLEVEL); // GPIO_ID_PIN(2) corresponds to GPIO2 on ESP8266-01 , GPIO_PIN_INTR_LOLEVEL for a logic low, can also do other interrupts, see gpio.h above
    wifi_fpm_open(); // Enables force sleep
    wifi_fpm_do_sleep(0xFFFFFFF); // Sleep for longest possible time
 }
  
  
}
```

The WiFi.forceSleepBegin() function seems to set the sleep type to MODEM_SLEEP_T which would explain why people have had trouble getting into light sleep. (from ...AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp)

```C
bool ESP8266WiFiGenericClass::forceSleepBegin(uint32 sleepUs) {
    _forceSleepLastMode = getMode();
    if(!mode(WIFI_OFF)) {
        return false;
    }

    if(sleepUs == 0) {
        sleepUs = 0xFFFFFFF;
    }

    wifi_fpm_set_sleep_type(MODEM_SLEEP_T);
    wifi_fpm_open();
    return (wifi_fpm_do_sleep(sleepUs) == 0);
}
```

TomTom101, 2016-03-07:
Do you know whether it restarts the whole device on wakeup like deep sleep does or does it continue in the loop()? I would assume keeping a wireless connection does not work during light sleep, right?

mikeyoyoyo, 2016-03-07:
It actually does keep the wireless connection and doesn't restart, just continues in the loop(). I had it sending UDP packets on button press (same GPIO pin that woke up the device) and it didn't need to reset or reconnect.

koravadi, 2016-03-08:
@mikeyoyoyo  
Is it possible to make it light sleep till the device get MQTT message or interrupt triggered?

mikeyoyoyo, 2016-03-08:
I haven't tried but the documentation would imply not, "Under the light-sleep mode, the CPU will not respond to the signal and interrupt from the peripheral hardware 
interface under the pause state. Therefore, the ESP8266 need to be waked up via external GPIO,". (from Sleep_Function_Description_V1 http://espressif.com/sites/default/files/documentation/9b-esp8266_sleep_function_description_en_v1.0.pdf)

pigglet, 2016-03-09:
koravadi  the code i posted was just for a stable ma test, to get the cpu to sleep change the code to pinMode(0, INPUT);           // set pin to input
digitalWrite(0, LOW);       // turn off pullup resistors
and 
gpio_pin_wakeup_enable(GPIO_ID_PIN(3), GPIO_PIN_INTR_HILEVEL);
this makes GPIO.0 the wake up pin when taken high (i only experienced a wake up event when setting the pin to wake from low to high) 

changing this bit of code tests that the cpu does in fact sleep and will only run when you take GPIO.0 high
val = digitalRead(0); 
while (val == 1){
  // do something repetitive
Serial.println("Pin 0 high"); 
delay (20000);
}

The modem stays awake during light sleep so the current draw ramps up and down despite the cpu going to sleep
All the best pigglet

TomTom101, 2016-03-10:
Not much luck for me, I can get it to sleep, but it won't wake up on the designated pin (GPIO4, made sure with an LED that I was using the correct one), not on GPIO_PIN_INTR_LOLEVEL or GPIO_PIN_INTR_HILEVEL or …ANY
GPIO2 and CH_PD are connected as well as RESET and GPIO16 which is supposedly required to enable deep_sleep (which seems to work, although I can't really say that waking up from deep sleep is any faster than a "cold start", it runs through setup() as well).

mikeyoyoyo, 2016-03-11:
You couldn't wake it from light sleep or deep sleep? If light what was your code?

TomTom101, 2016-03-11:
I realized I also gave up on deep sleep some time ago and are actually powering down. So the restart is no surprise.

I slight changed to code above to this:

```c
void setup() {
  Serial.write("Setup!");
}

void loop() {
  sleep();
  delay(100);
  Serial.write("Woke up!");
}

void sleep()
{
    gpio_pin_wakeup_enable(GPIO_ID_PIN(4), GPIO_PIN_INTR_HILEVEL); // or LO/ANYLEVEL, no change
    wifi_fpm_open(); 
    wifi_fpm_do_sleep(0xFFFFFFF);
 }
```

Without the delay in the loop, I saw continuous writes to Serial which reminded me that I read somewhere that the CPU keeps going for a few more cycles after being put to sleep. With the delay, I don't get to see the "woke up" message anymore and I assume real sleep. Then however, it neither will wake up when I pull up Pin4 to 3.3V.

So if it needs to be sent to sleep in a loop() continuously, this seems strange and I don't understand why you still saw 0mA.

Can you confirm the loop() does in fact stop, without a delay() on the first call to wifi_fpm_do_sleep() in your case?

mikeyoyoyo, 2016-03-11:
Did you set
```c
wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
```
anywhere?

I tried your code, it wouldn't compile unless I moved sleep() above loop() after I did, it just seemed to make my chip reset until I added a delay before the sleep() function in loop, but otherwise this worked (I used gpio 2 and low level though)

```C
extern "C" {
  #include "gpio.h"
}

extern "C" {
  #include "user_interface.h"
}

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.write("Setup!");
  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
}

void sleep()
{
    gpio_pin_wakeup_enable(GPIO_ID_PIN(2), GPIO_PIN_INTR_LOLEVEL); // or LO/ANYLEVEL, no change
    wifi_fpm_open(); 
    wifi_fpm_do_sleep(0xFFFFFFF);
}

void loop() {
  delay(1000);
  sleep();
  delay(100);
  Serial.write("Woke up!");
}
```

And yes if I removed the delay after sleep() it would still execute Serial.write("Woke up!"); before it went to sleep.

TomTom101, 2016-03-11:
@mikeyoyoyo Thanks! That was a sloppy job of mine, I did not copy and paste real code but wrote it on the fly from memory; I basically put your code from loop() in a separate function and added Serial.write()s and delay().

In your example, there's no more ```gpio_init()``` – maybe I'll try it without!

TomTom101, 2016-03-11:
@Yoshi20 writes in  #557:

> The problem now is, that light sleep won't be activated unless the ESP8266 is connected to a WLAN (or was connected once -> it seems to save the ssid and the ssid-pw internaly!)

@mikeyoyoyo It seems to work for you w/o being connected to an AP, right?

pigglet, 2016-03-12:
TomTom101 hes copied my code i posted the other day all be it with altering such as woke up as opposed to waking up, but hes still got it wrong. To wake up after the cpu has gone to sleep requires an interrupt on GPIO.0 and your tristate and port pins set accordingly. I am still unsure until a full data sheet showing all registers is available if using this device is viable or not

mikeyoyoyo, 2016-03-12:
@TomTom101

No worries, yeah I found gpio_init() was needed if I wanted to use digitalRead() or something in my actual code but not needed just to wake it up, I assume pinMode() or the arduino functions call it somewhere.

Yeah I wondered initially if that was a problem (not connected to an AP) but it worked without it being connected so I guess not. Not mentioned in the documentation either (like it does for modem sleep).

TomTom101, 2016-03-12:
I sort of see some sleeping now with this

```c
#include <ESP8266WiFi.h>

extern "C" {
#include "gpio.h"
}

extern "C" {
#include "user_interface.h"
}

#define LIGHT_WAKE_PIN 4

void setup()
{
  Serial.begin(115200);
  Serial.println("Setup");
  delay(100);
}

void loop()
{
  sleep();
  delay(100);
  Serial.println("Woke up form sleep");
}
void sleep()
{

  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
  wifi_fpm_open();
  gpio_pin_wakeup_enable(GPIO_ID_PIN(LIGHT_WAKE_PIN), GPIO_PIN_INTR_HILEVEL);
  wifi_fpm_do_sleep(0xFFFFFFF);
}
```

I pulled down PIN2 and it wakes on high level. However I see around 970µA of power consumption, not the 500µA as mentioned in #460 :-/

TomTom101, 2016-03-12:
When I connect to an AP in setup(), it consumes 3.3mA during sleep, not quite battery compatible …

MattiaDurli, 2016-03-15:
@TomTom101 @mikeyoyoyo 
Hello, I've given up with deep sleep, because it works for some times then it hangs on reset (gpio 0 and 2 up, 15 down, 16 to RST), as it is reported in many other forums.

I've made modem sleep work, it activates and deactivates, but it passes from 76mA to 22mA only:

```
 WiFi.disconnect();
 WiFi.mode(WIFI_OFF);
 delay(1); 
 WiFi.forceSleepBegin();
 delay(15000);
```

when in delay stays at 22mA and then loops.

I've managed to enter light sleep, and it goes from 76mA to 7mA, but I'm unable to wake it up:

```
wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
wifi_fpm_open(); // Enables force sleep
Serial.println("A");
wifi_fpm_do_sleep(0xFFFFFFF); // Sleep for longest possible time
delay(1);
Serial.println("A");
```
it never gets to "A", and it stays at 7mA

Doing any combination of wait with both the wifi_fpm_do_sleep argument or with the delay:
```
wifi_fpm_do_sleep(10000000); // Sleep for longest possible time
delay(10000);
Serial.println("A");
```
it gets to print "A", but never goes to 7mA, always 76mA.

Any suggestion?
At this point going to 7mA and back would be great.

shirish47, 2016-03-15:
Hi check this out. Light sleep requires External signal to wake up that is what I understand from this doc.  Still 7mA seems more may be you have power led on board. By the way anyone has a well documented code? 
http://espressif.com/sites/default/files/documentation/9b-esp8266_sleep_function_description_en_v1.0.pdf

shirish47, 2016-03-17:
@torntrousers  I used your code for testing that device goes in sleep or not and to find the activity. I am using a 0.1Ohm Resistor in my system(I can't show you schematic, contain other things of company) but has ESP8266 as Network processor. so I tested your code on it and thus found that your code works in delay for light sleep and around 20mA of power is consumed. We have a efficient DC-DC convertor for 3.3 volt but Signals have lots of noise I am not sure why(We have a very indian made oscilloscope can trust on filtering :P ) bt here is the [video](https://www.youtube.com/watch?v=Zc1jGw88C6Q&ab_channel=ShirishJadav).

chaeplin, 2016-03-18:
@shirish47 Can you check my code ? I have 1mA at light sleep.
```
  Serial.println("going to light sleep / 0xFFFFFFF");
  delay(100);
  wifi_station_disconnect();
  wifi_set_opmode(NULL_MODE);
  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
  wifi_fpm_open();
  gpio_pin_wakeup_enable(GPIO_ID_PIN(DOOR_INT_PIN), GPIO_PIN_INTR_HILEVEL);
  wifi_fpm_do_sleep(0xFFFFFFF);
  delay(200);

  gpio_pin_wakeup_disable();
  Serial.println("wake up / 0xFFFFFFF");
  wifi_fpm_close();
  wifi_set_opmode(STATION_MODE);
  wifi_station_connect();
  wifi_connect();
  delay(100);
```

shirish47, 2016-03-18:
@chaeplin I will do it bt right now my hardware has other controller on board so I can't give to exact current. I am working on making a power meter to do such things. :) that will give use better comparision intead of just pick current. My board might be taking more current because of other reasons but I would make a seperate ESP board to do current test.

chaeplin, 2016-03-18:
@shirish47 I have removed all the LEDs(20mA). I use ucurrent gold to check current.
graph is here : https://github.com/chaeplin/esp8266_and_arduino/tree/master/_48-door-alarm-deepsleep

shirish47, 2016-03-18:
Oh thats an awesome tool. your graph looks cool . My ESP8266 is ESP-12E module(I am not sure of this but it only has one LED connected to GPIO pin and PCB antenna) 
Btw It would be great if we can measure power for 1min operation in each mode. that kind of test can give better view that is what I am thinking of. so working on that.

TheAustrian, 2016-04-05:
Sorry to maybe ask a dumb question, but if I just want to get into light sleep during delays, I just call     "wifi_set_sleep_type(LIGHT_SLEEP_T);" in setup() and it will go into light sleep whenever I call delays?

mikeyoyoyo, 2016-04-05:
No not quite, you call "wifi_set_sleep_type(LIGHT_SLEEP_T);" in setup(), then you have to put it to sleep by calling wifi_fpm_do_sleep(0xFFFFFFF)  where 0xFFFFFFF is the length of sleep (not sure which units its in) and you generally have to put a delay call after the sleep call as it take a little bit of time to go to sleep (to stop it running code after the do_sleep call but before it goes to sleep).

TheAustrian, 2016-04-05:
Thanks! I could get it to work with that, but any hexadecimal value other than 0xFFFFFFF / 0cFFFFFF produces WDT resets or gibberish for me, so it seems it's unfortunately not useable for what I wanted to do, i.e. wake up after a minute or so of sleep.

mikeyoyoyo, 2016-04-05:
I see that's a bit odd I wonder why, I know the length of your program has to be longer than the beacon time of the router for it to enable light sleep. That doesn't really explain resets though.

TheAustrian, 2016-04-06:
I searched around and according to http://bbs.espressif.com/download/file.php?id=1294  wifi_set_sleep_type(MODEM_SLEEP_T) should be enough to use light sleep during delays etc, strangely. I unfortunately don't have a way to test the current consumption at those values, otherwise I'd test that. Would be a nice, clean and easy way if it worked.

tmorford, 2016-05-20:
Hello all I am new to this and have been looking in to doing just this, what I am trying to accomplish is enter light sleep mode and use a GPIO port to wake the device and run a function. I don't understand when I do run this script  

#include <ESP8266WiFi.h>

extern "C" {
#include "gpio.h"
}

extern "C" {
#include "user_interface.h"
}

#define LIGHT_WAKE_PIN 4

void sleep()
{

  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
  wifi_fpm_open();
  gpio_pin_wakeup_enable(GPIO_ID_PIN(LIGHT_WAKE_PIN), GPIO_PIN_INTR_HILEVEL);
  wifi_fpm_do_sleep(0xFFFFFFF);
}

void setup()
{
  Serial.begin(115200);
  delay(100);
  Serial.println("scanning for wifi");
  WiFi.scanNetworks();
}

void loop()
{
  sleep();
  delay(100);
}

that it just loops and resets, how do you tell when it wakes and when it does wake how can you tell it to run a function (VOID)?

Thanks any help would be great!

shirish47, 2016-05-28:
hey light sleep does not make the ESP sleep it just makes wifi radio turned on and off based on DTMI interval. you just check that out in datasheet. I am not sure because its long I forgot stuff. but I think you need deep sleep instead of light.

TheAustrian, 2016-06-25:
I now have a USB ammeter to test the current and tried to get light sleep to work using the code provided in the SDK documentation, I even tried adding delays copiously, but no dice... Current consumption stays the same and the loop continues, even though the callback function seems to fire after the specified 60 seconds... (I'm using the latest stable release, before it would not compile anyway)

Code:
```
#include <ESP8266WiFi.h>
extern "C" {
#include "user_interface.h" 
}

const char* ssid = "***";
const char* password = "***";
const char* espname = "D1-Sleep";
int counter = 0;

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
  //wifi_set_sleep_type(LIGHT_SLEEP_T); // LIGHT_SLEEP_T for sleep, NONE_SLEEP_T FOR NO SLEEP
  WiFi.hostname(espname);
  WiFi.mode(WIFI_STA);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    delay(500);
  }

}

void loop() {
Serial.println(counter);
counter++;

if (counter % 10 == 9) {
  sleepNow();
  
  }

delay(2000);
}

void fpm_wakup_cb_func1(void)  { 
wifi_fpm_close(); // disable force sleep function  
wifi_set_opmode(STATION_MODE); // set station mode 
wifi_station_connect();
Serial.println("Woken up...");
}

void sleepNow() { 
Serial.println("Going to sleep...");
wifi_station_disconnect(); 
wifi_set_opmode(NULL_MODE); 
wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // light sleep  
wifi_fpm_open(); // enable force sleep  
wifi_fpm_set_wakeup_cb(fpm_wakup_cb_func1); // Set wakeup callback 
delay(1000);  
wifi_fpm_do_sleep(60*1000*1000);  
delay(1000);
}

```

Serial output:
```
0
1
2
3
4
5
6
7
8
Going to sleep...
9
10
11
12
13
14
15
16
17
18
Going to sleep...
19
20
21
22
23
24
25
26
27
28
Going to sleep...
29
30
31
32
33
34
35
Woken up...
36
37
38
Going to sleep...
39
40
41
42
43
44
45
46
47
48
Going to sleep...
49
50
51
52
53
54
55
56
57
58
Going to sleep...
59
60
61
62
63
64
65
Woken up...
66
67
68
Going to sleep...
69
70
71
72
73
74
75
76
77
78
Going to sleep...
79
80
81
82
83
84
85
86
87
88
Going to sleep...
89
90
91
92
93
94
95
Woken up...
96
97
98
Going to sleep...
99
100
101
102
103
104
105
106
107
108
Going to sleep...
109
110
111
112
113
114
115
116
117
118
Going to sleep...
119
120
121
122
123
124
125
Woken up...
126
```

JohnHeath2, 2016-08-12:
Is it possible to leave ESP8266 in light sleep (as STA) and still periodically respond to messages from a smartphone App that is on the same AP? Could the App send out a set of redundant messages, which the ESP could periodically wake up and listen for? Or would I have to buffer the instructions on a server which ESP would query 1x per minute?

I have reliable communication between ESP / phone, but only tried it in full power mode. I need average current <= 1 mA.

 Would really appreciate advice / feedback.

mikeyoyoyo, 2016-08-12:
I think you could set it to light sleep for a set period of time and then
wake up and be able to receive messages. If you want current average less
than 1mA it would have to be asleep for large percentage of time though.

asafteirobert, 2016-08-12:
This sleep has been quite an adventure. I wanted rather precise sleep. When the device goes in light sleep the main counter goes slower and thus things like `millis()` are slower. Normally it's ok, but actually it causes `delay()` to wake up about 10% later witch is rather stupid. I tried to use the internal RTC(witch I think stands for real time counter, not clock), but after fiddling with the calibration value, it was still quite imprecise. So I did some measurements and calculated how much i have to subtract from the sleep delay to get the delay I want.
In the end, my code looks something like this.

```
void setup()
{
	WiFi.mode(WIFI_STA);
	WiFi.begin(ssid, password);
	WiFi.setSleepMode(WIFI_LIGHT_SLEEP);
	wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
       // ...
}
void loop()
{
        // ...
	if (toSleep > 0)
	{
		wifi_set_sleep_type(LIGHT_SLEEP_T);
		unsigned long extraSleep = toSleep * 0.011;
		delay(toSleep - extraSleep);
		previousMillis -= extraSleep;
	}
}
```

Now this works almost perfectly except that after some time (~20 hours) `delay()` stops sleeping, the ESP starts using full power and `delay()` becomes very precise again.
Did someone have similar issues? Any help?

simonliu009, 2016-09-23:
I'm using the following code and it can only be wake up for one time then doesn't response to GPIO2 state changes.
`
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <Ethernet.h>

extern "C" {
#include "gpio.h"
}

extern "C" {
#include "user_interface.h"
}

#define ioctrl 14
#define LIGHT_WAKEUP 2  //define GPIO5 as wake up pin for light sleep
IPAddress broadcastIp(255, 255, 255, 255);

unsigned int relay1Port = 8267; //remote udp port of the relay1
char  packetBuffer[10]; //only read 10 bytes of the udp packet
char  onBuffer[] = "LAMPON";
char  offBuffer[] = "LAMPOFF"; 
int   onflag = 0;
WiFiUDP Udp;
WiFiClient client;

void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
}

void setup() {
  // put your setup code here, to run once:
//    gpio_init(); // Initilise GPIO pins
    Serial.begin(115200);
    Serial.println();
    pinMode(ioctrl, INPUT);
    pinMode(LIGHT_WAKEUP, INPUT);
  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;

  //exit after config instead of connecting
    wifiManager.setBreakAfterConfig(true);
    wifiManager.setAPCallback(configModeCallback);
    wifiManager.autoConnect();  
    while (WiFi.status() != WL_CONNECTED)//
    {
        delay(500);
        Serial.print(".");

 
  //if you get here you have connected to the WiFi
    Serial.println("Wifi Connected :)");
    Serial.println("Local Ip Address");
    Serial.println(WiFi.localIP());
    delay(100);

}

void loop() 
{
       
        Serial.println("Ready to go into light sleep...");
        delay(1000);
        Serial.println("3...");
        delay(1000);
        Serial.println("2...");
        delay(1000);
        Serial.println("1...");
        sleepNow();
    //some udp code here...
}

void fpm_wakup_cb_func1(void)  { 
wifi_fpm_close(); // disable force sleep function  
wifi_set_opmode(STATION_MODE); // set station mode 
wifi_station_connect();
Serial.println("Woke up from sleep...");
}

void sleepNow() { 
Serial.println("Going into light sleep...");
wifi_station_disconnect(); 
wifi_set_opmode(NULL_MODE); 
wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // light sleep  
gpio_pin_wakeup_enable(GPIO_ID_PIN(LIGHT_WAKEUP), GPIO_PIN_INTR_LOLEVEL);
wifi_fpm_open(); // enable force sleep  
delay(100);
wifi_fpm_set_wakeup_cb(fpm_wakup_cb_func1); // Set wakeup callback  
wifi_fpm_do_sleep(0xFFFFFFF);  
delay(100);
}

`

elle123456, 2016-09-23:
I am finding difficulty in waking esp from deep sleep. it worked on GPIO 4 pin but on GPIO pin its not responding to the signal. I actually changed the library line from NONE_LIGHT_SLEEP to LIGHT_SLEEP.

if((digitalRead(tiltupdate) == HIGH)&&(digitalRead(updateline) == HIGH)){
 // digitalWrite(wakestate,LOW);
  delay(10);
  gpio_pin_wakeup_enable(GPIO_ID_PIN(16), GPIO_PIN_INTR_LOLEVEL);
 delay(20);
  WiFi.forceSleepBegin(0xFFFFFFF);
  delay(50);
  WiFi.forceSleepWake();
}

davescherler, 2016-11-04:
Hi All - I've read through this thread a bunch and it's been super helpful! Unfortunately, I'm also having a difficult time getting the ESP to wake up from light sleep. I can get it to sleep but it seems like the interrupt on my pin never registers. I'm using Arduino IDE on a Huzzah ESP8266 breakout. I have a PIR sensor connected to two pins: Pin 4 to listen and Pin 0 for the interrupt. Once it goes to sleep, it never wakes up however, I have noticed that the built-in LED (on gpio0) still blinks when I pass in front of the PIR even in sleep. That tells me something is registering and I'm not that far off. But can some help me understand what I'm missing?

`#include <ESP8266WiFi.h>

extern "C" {
#include "gpio.h"
}
extern "C" {
#include "user_interface.h"
}

int state = LOW;
const char* ssid = "myNetwork";
const char* password = "myPassword";
WiFiClient client;

void setup() {
  //WifiClient client;
  Serial.begin(115200);
  Serial.print("initializing GPIOs");
  gpio_init();
  pinMode(4, INPUT); //this pin is connected to the PIR sensor
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  delay(1000);

  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
  gpio_pin_wakeup_enable(GPIO_ID_PIN(0), GPIO_PIN_INTR_HILEVEL); //I've also connected the PIR sensor to pin 0. 
}

void loop() {
  
  delay(3000);
  Serial.println("reading PIR");
  int state = digitalRead(4);
  if (state == HIGH) {
    Serial.println("Motion detected");
    state = LOW;
  } else {
    Serial.print("about to go to sleep");
    wifi_fpm_open();
    wifi_fpm_do_sleep(0xFFFFFFF);
    delay(5000);
    return;
  }
}`

mikeyoyoyo, 2016-11-04:
Do any of the serial prints happen? Or it just seems to never wake up?

davescherler, 2016-11-05:
Hey Mike - thanks for the reply. I have good news! On a whim, I decided to use some of Simonliu009 callback code and everything seems to be working as I expected. I think what I wasn't putting together is that I needed to declare `wifi_fpm_set_wakeup_cb(some_wakeup_function);` before I called `wifi_fpm_do_sleep(0xFFFFFFF);` so that it knows what to do when the interrupt registers on the Pin. Now that I type that out it's like, duh, but I guess I had a hard time making sense of the comments in the gpio header file. Anyway, I wanted to share my successful code since I'm able to put the ESP to sleep, and wake it up using a PIR sensor, without having to pull any gpio's up or down, or have RST connected to 16 or EN. So a pretty simple hardware solution to use as a starting place for others. Thanks to everyone for helpful thread! I learned a lot. 

```
#include <ESP8266WiFi.h

extern "C" {
#include "gpio.h"
}
extern "C" {
#include "user_interface.h"
}

const char* ssid = "myNetwork";
const char* password = "myPassword";
WiFiClient client;

void setup() {
  Serial.begin(115200);
  Serial.print("initializing GPIOs");
  gpio_init();
  pinMode(0, INPUT); // this pin is connected to the PIR sensor.
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  delay(1000);
}

void loop() {
  Serial.println("Ready to go into light sleep...");
  delay(1000);
  Serial.println("3...");
  delay(1000);
  Serial.println("2...");
  delay(1000);
  Serial.println("1...");
  sleepNow();
}

void sleepNow() {
  Serial.println("going to light sleep...");
  wifi_station_disconnect();
  wifi_set_opmode(NULL_MODE);
  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); //light sleep mode
  gpio_pin_wakeup_enable(GPIO_ID_PIN(0), GPIO_PIN_INTR_HILEVEL); //set the interrupt to look for HIGH pulses on Pin 0 (the PIR).
  wifi_fpm_open();
  delay(100);
  wifi_fpm_set_wakeup_cb(wakeupFromMotion); //wakeup callback
  wifi_fpm_do_sleep(0xFFFFFFF); 
  delay(100);
}

void wakeupFromMotion(void) {
  wifi_fpm_close;
  wifi_set_opmode(STATION_MODE);
  wifi_station_connect();
  Serial.println("Woke up from sleep");
}

```

mikeyoyoyo, 2016-11-05:
Good to hear Dave. Although you don't have to do a callback function as I never had to use one, I assume it just re-enters the line after the do_sleep() call. Nicer to use a function though so thanks for sharing.

torntrousers, 2016-11-07:
@davescherler that sounds great, I'll give your code a try. Do you know what current consumption you get while its sleeping?

davescherler, 2016-11-07:
hey @torntrousers I haven't been able to  find my multimeter anywhere - so no, unfortunately I don't know the consumption rate. Were you able to get the code to work? I'm curious because after several successful tests, I decided it was time to incorporate the code into my larger project - which I just did this afternoon. But now, the callback function (wakeupFromMotion) doesn't seem to get called. Once it goes to sleep, it's asleep and can't be woken up. I'm using the same exact pins and header files and everything. So bizarre. I'm running out of ideas on why this is happening. I was hoping you might have given it a go and have some insight? Thanks!

[43 more replies]

closed 2018-10-09 by devyte