esp8266/Arduino #1382: PWM not working always HIGH or LOW 2.1 git version

tontito, 2016-01-07:
Hi,

PWM not working always HIGH or LOW 2.1 git version 
SDK 1.5

int ledPin = 5;    // LED connected to digital pin 5
void setup() {
  pinMode(ledPin, OUTPUT);
  analogWriteRange(255);
}

void loop() {
  for (int fadeValue = 0 ; fadeValue <= 255; fadeValue += 5) {
    analogWrite(ledPin, fadeValue);
    delay(30);
  }

  for (int fadeValue = 255 ; fadeValue >= 0; fadeValue -= 5) {
    analogWrite(ledPin, fadeValue);
    delay(30);
  }
}

Led high or low

SmartArduino, 2016-01-08:
It's worked in my case using your code. I use ESP-13 based module(ESPduino).

tontito, 2016-01-08:
İ'm using esp-07 and esp-12 

Not working pwm

Analogwrite (50-100-200) always High
Analogwrite (0) low

SmartArduino, 2016-01-08:
Change to 2.0 version may solve your prolem.
As I'm using pwm to control a rgb led light via tcp client, I found it worked well in 2.0 version(sdk is 1.3), but something not good under 2.1 version(sdk is 1.5).

tontito, 2016-01-08:
i tested 2.0 stable version it`s worked but 2.1 github version not working.

tablatronix, 2016-01-08:
Same problem here, spent all day thinking I was doing something wrong.
Rolled back to 2.0 and its working fine.

tablatronix, 2016-01-08:
revert 1ae423021a3327b416919b95c17de6eafeffcffc

tablatronix, 2016-01-08:
typo fix works for me now
https://github.com/esp8266/Arduino/commit/1ae423021a3327b416919b95c17de6eafeffcffc#commitcomment-15344434

tontito, 2016-01-08:
fix works for me now
1ae4230#commitcomment-15344434

closed 2016-01-08 by tontito

tablatronix, 2016-01-23:
Can we get this fix merged?