esp8266/Arduino #1384: TCP server ESP8266 resets client connection

hakeemta, 2016-01-07:
I need to make my ESP8266 Mod12 have more than one socket.

At least I want it to act as a client connecting to a cloud server via websocket. In this way, it joins an existing Internet gateway (or AP). 

Also, I want to have use ESP8266 softAP and create some server sockets listening for client from other devices.

I have the following specifications/schemes:
1. The ESP configured or setup in STA+AP mode.
2. As a Station, it connects to an existing Internet AP.
3. As a "soft" AP, it accepts connection from other WiFi-enabled device.
4. A client socket to a cloud server (websocket)
5. Server socket listening to two or three clients' connection.

I succeeded in establishing all these but a moment later after testing, my phone looses connection with the Server socket running on the ESP; logging ECONNRESET error. Since then, I've not been able to establish a connection with the server even in just a single socket.

Pls assist.

nouser2013, 2016-01-14:
if still open, attach your code / a [MWE](https://en.wikipedia.org/wiki/Minimal_Working_Example) and state the versions of IDE and arduino-esp variant used (2.0.0, 2.1.0RC, GIT, ...).

hakeemta, 2016-01-14:
Thank you so much. Below is the code, I decided to use this sample below and still the same issue persists.

IDE:  Arduino 1.65 (also Visual Micro + Atmel Studio)
Ard-ESP Variant: 1.6.5.1160-gef26c5f

hakeemta, 2016-01-14:
#include <ESP8266WiFi.h>
#include <WebSocketClient/WebSocketClient.h>
#include <WiFiClient.h>

/* Set these to your desired credentials. */
const char *AP_ssid = "AP_chas";
const char *AP_passkey = "12345678";

int status = WL_IDLE_STATUS;
boolean alreadyConnected[2] ; // whether or not the client was connected previously

WiFiServer server(6000);
WiFiClient client[2];

#define Config_Led D2

void setup()
{
	delay(1000);
	Serial.begin(115200);
	Serial.println();
	Serial.println("\nConfiguring access point...");
		
	WiFi.softAP(AP_ssid, AP_passkey);
	delay(10000);
	IPAddress myIP = WiFi.softAPIP();
	Serial.print("AP IP address: ");
	Serial.println(myIP);

	Serial.println("\nStarting server...");
	// start the server:
	server.begin();	
	Serial.println("\nServer Started.");
}

void loop()
{
	unsigned int i;
	for (i= 0; i<2; i++)
	{
		// wait for a new client:
		if (!client[i])
		{
			client[i] = server.available();
		}
		else
		{
			if (client[i].status() == CLOSED)
			{
				client[i].stop();
				Serial.print("\nConnection closed on client: "  );
				Serial.println(i);
				Serial.println();
				alreadyConnected[i] = false;
			}
			
			else
			{
				if(!alreadyConnected[i]) // when the client sends the first byte, say hello:
				{
					// clean out the input buffer:
					client[i].flush();
					Serial.print("\nNew client: ");
					client[i].println("Hello, client !");
					Serial.println(i);
					alreadyConnected[i] = true;
				}
				
				if (client[i].available())
				{
					// read the bytes incoming from the client:
					String data = client[i].readString();
					// echo the bytes back to the client:
					client[i].print(data);
					// echo the bytes to the server as well:
					Serial.println(data);
				}				
			}
		}
	}
}

nouser2013, 2016-01-14:
well, I dont see anything wrong with the code. You can try with one single client first, though that should not make a difference. Also I suggest that you upgrade the IDE to 1.6.7 and the variant to at least 2.0.0 or even the 2.1.0-rc1.

hakeemta, 2016-01-14:
Thanks. I'll try that and get back with results.

nima72, 2016-01-24:
I had the same problem, solved by adding the following line:

WiFi.mode(WIFI_AP);

sounds like if you don't add this line, it won't notice data coming from AP...

hakeemta, 2016-01-24:
That's really incredible! That solves it, you guys here are geniuses.

Thanks @nouser2013, @nima72 and others. I can move on with my work now.

closed 2016-01-24 by hakeemta

ByTorrez, 2016-07-28:
Hello nima72, where do you include this 
WiFi.mode(WIFI_AP);
Thanks a lot.

MeFisto94, 2016-12-07:
Hey Guys, I am re-bumping this, since this is still happening on 2.3.0, @igrr 
I had the problem that my `WiFiClient`'s `status` changed from 4 to 0 (`CLOSED`) and a `FIN` packet was sent. This might be related to the `WiFiClient#close` calls I was seeing, but I can't verify that.
So this meant that my connection was killed after a few bytes and then I had to reestablish it.

@ByTorrez You just need to specify `WiFi.mode(WIFI_AP);` before the `WiFi.softAP` calls. #529 might have been related to that, so you'd have to specify it after softAP, but this seems to be fixed indeed.

Another thing worth mentioning: Unlike in this Issue, I did explictly specifiy `WiFi.mode(WIFI_STA_AP);` (not knowing what I was doing, actually, I think the WebServerExample also does this wrong?).

So to me it seems that something in `STA_AP` mode is killing the connections, when you spawn an AP but don't connect to an AP on your own (that's at least what I could experience)

Thanks for the awesome work, guys!

sprabu37, 2018-01-02:
hello, hakeemta? are u create local server hakeemta?. i want to listen another cloud  server  data ,this data sent to connected client what i do?

youmebangbang, 2018-04-04:
Thank you bachi76, changing my mode to WIFI_AP solved all my server connection issues!