esp8266/Arduino #1385: "fmod" causes linker error undefined reference to `__ieee754_remainder'

bmitov, 2016-01-08:
This simple Sketch fails to compile in 2.1.0-rc1:

void setup() {
  // put your setup code here, to run once:

  float ATest2 = 3.3;
  float ATest1 = fmod( ATest2, 1 );
}

void loop() {
  // put your main code here, to run repeatedly:
}

It produces:

e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'

error.
This prevents a number of Visuino components from compiling for ESP8266.
Please let me know if there is a workaround, but I think this is a very serious compiler/linker/library bug.

SalvorinFex, 2016-01-08:
I can confirm this in the 'stable" version 2.0.0, running WIN 8.1 and arduino IDE 1.6.5.

```c:/users/WINUSER/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o):(.literal+0x4): undefined reference to `__ieee754_remainder'
c:/users/WINUSER/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':
e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'
collect2.exe: error: ld returned 1 exit status
Error compiling.```

igrr, 2016-01-08:
This is tracked as #612, let's keep the discussion there.

closed 2016-01-08 by igrr