esp8266/Arduino #1386: ESP Strange Reset

dariocdj, 2016-01-08:
Hi all, 
i've a Souliss node on esp-07 that work for some seconds and then reset with : 

ets Jan  8 2013,rst cause:1, boot mode:(3,6) 

What is the reason ? what's the meaning of this code ? 

thanks
Dario

Links2004, 2016-01-08:
rst cause:1 is normal boot (when the power supply turned on)
check the stability of the 3.3V, may add some capacitors >=47µF

more info about the different boot messages can be found here:
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-messages-and-modes

DedeHai, 2016-01-11:
I would also place my bet on instable 3.3V supply. AMS1117 for 3.3V for a test would be good. The small sot23-5 regulators do not have enough power in my experience, they drop down on high current peaks. For testing put a big ass capacitor on the 3.3V rail, like 2200µF. Or use an external 3.3V supply, like a Lab powersupply.

dariocdj, 2016-01-11:
I've put a 1000mf at the start... Today has not crashed, bah

hallard, 2016-01-12:
yeah, classic problem, I often see this one on lot of chinese NodeMCU because you think you got a real ASM1117 3V3 on board but it's just a clone with really poor caps, I've burned some putting only 14V on vin and AMS datasheet says reg can go up to 20V (NodeMCU doc says same) so, first bet power supply, that'a a shame because you can find really good regs for some cents more !!!

dariocdj, 2016-01-14:
Hallard... you're right, but how do you recognize fake one ?

hallard, 2016-01-14:
To be honest, you can consider all as fake until you buy a well none manufacturer of ESP8266 boards such as Adafruit Huzzah, Sparkfun Things, Digistump OAK, ...
I've got plenty of Node MCU, I put before some [cheap DC/DC](http://www.ebay.com/itm/Mini-MP1584EN-DC-DC-BUCK-Adjustable-Step-Down-Module-Effect-Higher-Than-LM2596-/301668096806) now to avoid problem

dariocdj, 2016-01-14:
Ehm... and who told you that this DC/DC is not fake :P

dariocdj, 2016-01-14:
Also if after i use also the regulator of electrodragon adapter... 
so from 12v with fake ams down to 5v and then from 5v to 3.3v with electrodragon board integrated adapter.... http://www.electrodragon.com/product/esp8266-smd-adapter-board/

hallard, 2016-01-14:
DC/DC sure made with cheap components, but surprising, they're working fine ;-)
Yeah got some of them from electrodragon also ;-)

igrr, 2016-02-29:
@dariocdj is the issue resolved for you?

dariocdj, 2016-02-29:
Now I'm using ams from 5 to 3.3 or directly electrodragon adapter and no more reset

closed 2016-02-29 by igrr