esp8266/Arduino #1387: Board is unknown with Git version and 1.6.7

networxnet, 2016-01-08:
When i use Git version with 1.6.7 as described in the newest read.me i still have 
"Board nodemcuv2 (platform esp8266, package esp8266) is unknown" error.

For my understanding it should work with 1.6.7 now?!

Can somebody please verfiy.

Links2004, 2016-01-08:
check your arduino dir you may have staring and stable mixed

networxnet, 2016-01-08:
C:\Program Files (x86)\Arduino only?

Links2004, 2016-01-08:
some think like: C:\Users\[username]\AppData\Roaming\Arduino15\packages\esp8266\hardware

networxnet, 2016-01-08:
Sorry maybe a stupid question but i'm confused now. Readme is a bit minimalistic at this part and says:

    Install Arduino 1.6.7
    Go to Arduino directory
    Clone this repository into hardware/esp8266com/esp8266 directory (or clone it elsewhere and create a symlink)

What is the full path ? 
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266
or
C:\Users[username]\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266com\esp8266

andig, 2016-01-08:
Duplicate of https://github.com/esp8266/Arduino/issues/1377 ?

igrr, 2016-01-10:
The issue seems to come from different FQBN's when using boards manager package version and git version.
In boards manager package, FQBN is `esp8266:esp8266:nodemcuv2`.
In git version, if you follow instructions in readme, it will be `esp8266com:esp8266:nodemcuv2`.
So if Arduino has saved the latest board selection as `esp8266:esp8266:nodemcuv2`, then once you install git version, IDE will no longer be able to find the board. I think selecting a different board (like one of the AVR boards) and then selecting NodeMCU again should fix this.

Regarding the installation directory, you should place the git copy into Arduino program directory (which may be in Program Files or in another place, depending on where you have placed your copy of Arduino software).

The path mentioned by @Links2004 is where Arduino IDE places downloaded boards manager packages itself. If you use git version, make sure you don't have the esp8266 package in that directory, or it will actually be used instead of your git copy.

igrr, 2016-02-29:
@networxnet any feedback?

tectiv3, 2016-03-23:
I have the same issue:

> Board generic (platform esp8266, package esp8266) is unknown
> Error compiling for board Generic ESP8266 Module.
OSX Arduino 1.6.8, 2.1 stable or 2.0 - doesn't matter. But folder Documents/Arduino/Hardware is empty.

ericcastro, 2016-04-02:
having the same issue in Mac OS X

anhhuy0501, 2016-04-03:
I have same issue on Windows 10, Arduino 1.6.8, Installing with Boards Manager. Version 2.1 stable or -rc2 is the same. The error when compile:

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\nguyen\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\nguyen\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\nguyen\Documents\Arduino\libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10608 -build-path "C:\Users\nguyen\AppData\Local\Temp\buildb2853698f701ebc78fb285fce0ae6bdb.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\nguyen\Documents\Arduino\read_flash_id\read_flash_id.ino"

Board generic (platform esp8266, package esp8266) is unknown
Error compiling for board Generic ESP8266 Module.
```
It was ok when I use it in the Arduino 1.6.7. The problem appear when upgrade to 1.6.8. There is no folder name
C:\Users[username]\AppData\Roaming\Arduino15\packages\esp8266\hardware
or:
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266

Any suggestion is appreciate.

anhhuy0501, 2016-04-03:
I lucky solve the problem by:

Go to the folder:

`C:\Users\[username]\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\
`

In this folder, in my case, there are 2 version folder. I delete the older ver folder and everything working fine again.

matroll, 2016-04-05:
thanks @anhhuy0501 !!
i'be been getting frustrated with this for about an hour and was about to go back to 1.6.6 
glad i found this post - cheers

grantnlee, 2016-04-08:
For reference...  Had the same problem with my boards manager installation using IDE 1.6.8 on OSX.  I had previously used several older versions like 1.6.5 and 1.6.5 on the same system.  It was not obvious to me how or where to delete the ESP8266 board files.  My setup points my sketches to a Dropbox folder, but previously had them in Downloads.  Tried using board manager to auto remove and replace the ESP8266 board, tried doing so with restarts in between actions as well as selecting an UNO in between remove and replace...  Notta.  At one point I even had two complete listings of the various ESP8266 board within the IDE drop down menus.  Ended up manually deleting anything related to hardware:ESP8266 within 1.) my Documents folder (had Arduino and Arduinosketches folders) as well as 2.) Dropbox:Arduino and 3.) also the hidden Arduino15 folder.  Once all traces were remove, then I reinstalled with boards manager and all worked right away...

anhhuy0501, 2016-04-08:
@matroll U welcom. :)

About the Arduino 15 folder, in Arduino ver 1.6.8, I see that we could find it by: File->Preferences, in the bottom of the References window, there is a link like this: 
`C:\Users\[user]\AppData\Local\Arduino15\preferences.txt.`

Hope that will help.

biol75, 2016-05-18:
Thanks grantnlee for this, I had two directories in ~/Library/Arduino15/packages/esp8266/hardware/esp8266/

I deleted 2.2.0 and left 2.3.0-rc1and then all compiled ok

kozoke, 2016-05-21:
Thanks @biol75 and others.  removing 2.2.0 worked for me too.

culbeda, 2016-06-25:
I had multiple versions of the libraries under %userprofile%\Appdata\Local\Arduino15\packages\esp8266\hardware.  Removing one fixed the issue immediately for me.

(Note:  Uninstalling and reinstalling the boards in board manager won't fix it.  Uninstalling and reinstalling the Arduino IDE won't fix it.  You have to remove the 2nd version in the path above to fix, even on 1.6.9.

tablatronix, 2016-06-29:
Same problem, must be a bug in board manager

tectiv3, 2016-07-02:
I switched to platformio and never looked back

electronicsguy, 2016-07-19:
@culbeda Thanks...that fixed it!

- [ ]

aguirre689, 2016-12-08:
You are a genious ! It works perfect. I lost 2 days looking for the problem.
I use Windows 7. I had an "installed" version of Arduino 1.5.4.
Since that version worked well with all my previous sketches, I did not want to change it. Then I installed Arduino 1.6.7 "as portable". When I followed your advice, in the directory of the installed version were the two folders mentioned. I deleted the "old" and everything worked ok. Thank you very much.

Bruce-Ferjulian, 2016-12-11:
I got myself stuck here too, I am on OSX.

Tried to take the staging version to poke at spiffs filesystem.  After trying to revert to the stable version I began to receive the same message.

To fix this I did the following.

1. Check what is in your library [ ls ~//library/Arduino15/packages/esp8266/hardware/esp8266 ].
2. In my case there were two libraries [ 2.3.0 && 2.3.0-rc2 ]
3. I removed unstable one and kept the stable [ 2.3.0 ] version and I was back in business.

ArsenioDev, 2016-12-18:
Confirmed, removing the 2.1.0 board file fixed the compile issue for me too

devyte, 2017-10-19:
Switching between package manager and git version is documented. Closing.

closed 2017-10-19 by devyte

gearsngenes, 2017-11-05:
This suggestion is remarkable!!!
BTW in that folder: C:\Users\Owner\AppData\Local\Arduino15\packages\esp8266 
there should be only ONE FOLDER preferably with version 2.3.

Then it works like a charm.

tarisont, 2018-11-25:
how do i add ESPWiFi.h library to my arduino environment

tarisont, 2018-11-25:
[image]