esp8266/Arduino #1392: fatal error: ESP8266HTTPClient.h: 2.1.0-rc1 with arduino 1.6.5

mmiscool, 2016-01-08:
I installed using the board mangers link for the staging version.

removed the old arduino15 directory from the appdata roaming folder before install to start fresh. Getting this error when compiling my project. 

C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266httpUpdate\src/ESP8266httpUpdate.h:33:31: fatal error: ESP8266HTTPClient.h: No such file or directory
 #include <ESP8266HTTPClient.h>
                               ^
compilation terminated.
Error compiling.

Links2004, 2016-01-09:
you need to include the ESP8266HTTPClient lib in you sketch
```cpp
#include <ESP8266HTTPClient.h>
```

mmiscool, 2016-01-10:
I added that include to my sketch. Now I get the following error. 

This did work in the last staging version 

In file included from ESP8266Basic.ino:29:0:
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:129:20: error: 'String' has not been declared
         void begin(String url, String httpsFingerprint = "");
                    ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:129:32: error: 'String' has not been declared
         void begin(String url, String httpsFingerprint = "");
                                ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:131:38: error: 'uint16_t' has not been declared
         void begin(const char *host, uint16_t port, const char * url  = "/", bool https = false, const char * httpsFingerprint = "");
                                      ^
In file included from ESP8266Basic.ino:29:0:
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:132:20: error: 'String' has not been declared
         void begin(String host, uint16_t port, String url = "/", bool https = false, String httpsFingerprint = "");
                    ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:132:33: error: 'uint16_t' has not been declared
         void begin(String host, uint16_t port, String url = "/", bool https = false, String httpsFingerprint = "");
                                 ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:132:48: error: 'String' has not been declared
         void begin(String host, uint16_t port, String url = "/", bool https = false, String httpsFingerprint = "");
                                                ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:132:86: error: 'String' has not been declared
         void begin(String host, uint16_t port, String url = "/", bool https = false, String httpsFingerprint = "");
                                                                                      ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:142:25: error: 'uint16_t' has not been declared
         void setTimeout(uint16_t timeout);
                         ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:148:18: error: 'uint8_t' has not been declared
         int POST(uint8_t * payload, size_t size);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:148:37: error: 'size_t' has not been declared
         int POST(uint8_t * payload, size_t size);
                                     ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:149:18: error: 'String' has not been declared
         int POST(String payload);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:150:44: error: 'uint8_t' has not been declared
         int sendRequest(const char * type, uint8_t * payload = NULL, size_t size = 0);
                                            ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:150:70: error: 'size_t' has not been declared
         int sendRequest(const char * type, uint8_t * payload = NULL, size_t size = 0);
                                                                      ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:151:44: error: 'Stream' has not been declared
         int sendRequest(const char * type, Stream * stream, size_t size = 0);
                                            ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:151:61: error: 'size_t' has not been declared
         int sendRequest(const char * type, Stream * stream, size_t size = 0);
                                                             ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:151:13: error: 'int HTTPClient::sendRequest(const char*, int*, int)' cannot be overloaded
         int sendRequest(const char * type, Stream * stream, size_t size = 0);
             ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:150:13: error: with 'int HTTPClient::sendRequest(const char*, int*, int)'
         int sendRequest(const char * type, uint8_t * payload = NULL, size_t size = 0);
             ^
[... cut ...]
         void begin(String host, uint16_t port, String url = "/", bool https = false, String httpsFingerprint = "");
                                                                                                                ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:150:64: error: 'NULL' was not declared in this scope
         int sendRequest(const char * type, uint8_t * payload = NULL, size_t size = 0);
                                                                ^
Error compiling.

Links2004, 2016-01-10:
what IDE you use for compiling?
eclipse need:
```
#include <Arduino.h>
```
before.

the Arduino IDE does it in background for you.

mmiscool, 2016-01-10:
I am using arduino ide 1.6.5

mmiscool, 2016-02-25:
Any word on this one?

Links2004, 2016-02-25:
this is woking for me in Arduino 1.6.5
with 2.1.0-rc1 and 2.1.0-rc2 with git version too.
try to remove all core versions that you may have and install them new.
```cpp
/**
 * httpUpdate.ino
 *
 *  Created on: 27.11.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    USE_SERIAL.begin(115200);
    // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("SSID", "PASSWORD");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        t_httpUpdate_return ret = ESPhttpUpdate.update("http://server/file.bin");
        //t_httpUpdate_return  ret = ESPhttpUpdate.update("https://server/file.bin");

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                break;

            case HTTP_UPDATE_NO_UPDATES:
                USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                USE_SERIAL.println("HTTP_UPDATE_OK");
                break;
        }
    }
}
```

kmat, 2016-09-08:
i have same error.
Anyone help us please.

devyte, 2017-10-19:
Can't reproduce with latest git. Closing.

closed 2017-10-19 by devyte

pikolo84, 2018-10-17:
I have this:
.piolibdeps\FirebaseArduino\src/ESP8266HTTPClient.h:138:10: error:   initializing argument 2 of 'bool HTTPClient::begin(String, uint16_t, String)' [-fpermissive]
bool begin(String host, uint16_t port, String uri = "/");
if add firebase to project with library httpupdater.

ahmadasmndr, 2019-09-06:
> 
> 
> you need to include the ESP8266HTTPClient lib in you sketch
> 
> ```c++
> #include <ESP8266HTTPClient.h>
> ```

thanks it worked for platform IO