esp8266/Arduino #1393: Incorrect firmware loaded into ESP 01 after replacing  flash

DimIsaev, 2016-01-09:
after replacing the flash on ESP 01
not working sewn sketches
WR is dimly lit and pollutes the entire broadcast WiFi

andig, 2016-01-09:
When I've seen dim led before that was due to insufficient power supply. Usb may not be enough to support the board when transmitting over wifi.

DimIsaev, 2016-01-09:
the problem is not power DC
DC =  3.3V 
Other operating firmware for example Homes-smart.ru
firmware and an blank_1mb
As you press the reset s LED bright lights
Something with increased power transmitters WiFi

marvinroger, 2016-01-09:
It's not about the voltage, it's about the amps. I'm using an 1A power supply and I don't have any issues, but I had with less powerfull power supplies.

DimIsaev, 2016-01-09:
Yes. Power suplay  supply 1A
Other firmware work  is ok

marvinroger, 2016-01-10:
Have you tried http://esp8266.github.io/Arduino/versions/2.0.0/doc/boards.html#minimal or even better http://esp8266.github.io/Arduino/versions/2.0.0/doc/boards.html#improved-stability?

DimIsaev, 2016-01-10:
@marvinroger  `
firmware upload  ok
After upload not correct work module.

ESP01  not  GPIO15 !

marvinroger, 2016-01-10:
What version are you using? 2.0? 2.1-rc1? Git rev?

DimIsaev, 2016-01-10:
@marvinroger I tried both versions

DimIsaev, 2016-01-10:
@igrr

igrr, 2016-01-10:
You have mentioned something about replacing flash,

    after replacing the flash on ESP 01

Could you please elaborate, have you replaced the flash chip? What is the size of the new chip? What flash size do you have selected in the IDE? Did you try erasing the chip completely? Is there any serial output from the chip? Any exception info? Did you try enabling debug output from Serial?

DimIsaev, 2016-01-11:
1. in the CAP 01 is replaced by Chip 25G32FSV  (datasheet http://lib.chipdip.ru/093/DOC001093215.pdf)
2. Erases the chip completely using the form on file 1m
[image]
after upload LED ок 

3. Setting for uploads

[image]

output  debug upload
```
C:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1/tools/sdk/include -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\cores\esp8266 -IC:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\variants\generic C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\CheckFlashConfig.cpp -o C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\CheckFlashConfig.cpp.o 
C:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1/tools/sdk/include -c -g -x assembler-with-cpp -MMD -mlongcalls -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\cores\esp8266 -IC:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\variants\generic C:\Users\dim\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\cores\esp8266\cont.S -o C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\cont.S.o 
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\cont_util.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_eboot_command.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_flash_utils.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_i2s.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_noniso.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_phy.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_postmortem.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_si2c.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_timer.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_wiring.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_wiring_analog.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_wiring_digital.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_wiring_pulse.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_wiring_pwm.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\core_esp8266_wiring_shift.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\cdecode.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\cencode.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\libc_replacements.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\spiffs_cache.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\spiffs_check.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\spiffs_gc.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\spiffs_hydrogen.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\spiffs_nucleus.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\time.c.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\abi.cpp.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\base64.cpp.o
Используем предварительно скомпилированный файл: C:\Users\dim\AppData\Local\Temp\build717341743733632456.tmp\cbuf.cpp.o
[... cut ...]
....................................................................................................................................................................................................................................
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
```

igrr, 2016-01-11:
Looks good so far, except for the "4Mbit" setting in the flash download
tool. But that probably doesn't make much difference.

Is there any output on the serial port from the ESP? Check at 115200 and
74880 baud.

DimIsaev, 2016-01-11:
74880 
```
PORT OPEN 74880

Communication with MCU...
.....Got answer! AutoDetect firmware...

Can't autodetect firmware, because proper answer not received.
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

```

DimIsaev, 2016-01-11:
@igrr  more information needed?

igrr, 2016-01-11:
Looks like the firmware is uploaded successfully. 
Frankly, I have no other ideas left what might be wrong.

DimIsaev, 2016-01-11:
[Solved]
problem is power supllay

```
Flash real id:   001640EF
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  QIO
Flash Chip configuration ok.
```

igrr, 2016-01-11:
Just as @marvinroger said :)

closed 2016-01-11 by igrr

DimIsaev, 2016-01-11:
:)  yes

ESP12 correct work with the power source
ESP01  consumes more than ESP12

andig, 2016-01-11:
> problem is power supply

Which is what I've said...