esp8266/Arduino #1400: help with the installation of libraries

Amayak777, 2016-01-10:
doing everything so as described here https://github.com/esp8266/Arduino#installing-with-boards-manager, before the library was in the folder -С:\Users\ХХХ\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries. I tried to download GitHab project gives many errors at compile time. Compared with the downloaded ZIP archive GitHab and the library project that I had - I found a lot of fixes and changes in the structure of folders. I suspect that this is a completely different project. What to do? I can not build the project https://gist.github.com/4refr0nt/829052635894da0f5715

Amayak777, 2016-01-10:
You may need to reinstall the library, how to do it?

Links2004, 2016-01-10:
```С:\Users\ХХХ\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries```
is for core library only if you add own stuff there it will be deleted on every update of the core you do.

if you want to add libraries to your Arduino follow this guide:
https://www.arduino.cc/en/Guide/Libraries

Amayak777, 2016-01-16:
I installed all the libraries by keyword ESP. Don't know what else to do? Probably a library, esp not updated.
here is what I have installed, see figure
[image]

Amayak777, 2016-01-16:
when you remove the IDE library are not deleted. How to delete all settings and files? to try to re-install ESP the library

Amayak777, 2016-01-17:
1) Deleted all the folder:C:\Users\User\AppData\Roaming\Arduino15, E:\Users\User\Documents\Arduino and the IDE. 2) Downloaded from the IDE website and re-installed version 1.6.5. 3)write in URL: http://arduino.esp8266.com/package_esp8266com_index.json. 4) now in the library Manager does not find the project titled esp8266. What to do?

Amayak777, 2016-01-17:
here is my project Manager with filter esp8266
[image]

Amayak777, 2016-01-24:
Кто-нибудь попробуйте сами установить ядро.

igrr, 2016-02-29:
The first three steps you did were correct, but you need to use Boards Manager to install the package, not Library Manager. It's in `Tools > Board > Boards Manager...` menu.
Should look like this:

[image]

closed 2017-10-19 by devyte