esp8266/Arduino #1404: data transfer from arduino to other device by the ESP8266

kapilpawar, 2016-01-11:
I am new with esp8266 and arduino.
so can any buddy help me to write the code....

thanks in advance..

Links2004, 2016-01-11:
you not need a uno, you can program the ESP8266 direct with the arduino IDE.
to send data to a mobile phone there are many ways but its based on what you have running on the phone?
do you write an app or you only use the Webbrowser of the phone, what phone os?
one very generic way which work for PC too is using websockets.
https://github.com/Links2004/arduinoWebSockets/blob/master/examples/WebSocketServer_LEDcontrol/WebSocketServer_LEDcontrol.ino

drmpf, 2016-01-11:
If you are using Android, you can use the free pfodDesigner to design  a custom menu to switch ESP outputs and generate the code for you that will work the pfodApp (www.pfod.com.au).
Also works for Uno using the ESP as a WiFi shield (http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html)

itsjustvenky, 2016-01-12:
https://github.com/tuanpmt/espduino

kapilpawar, 2016-01-13:
i will sent data to the android phone.
help me in writing the server code in the arduino.

Duality4Y, 2016-01-13:
At a forum they might be able to help you beter and further on that.

devyte, 2017-10-18:
Not the right place to ask for beginner help. Closing per #3655 .

closed 2017-10-18 by devyte