esp8266/Arduino #1405: Error ESP8266 send data to webserver:

binhdv92, 2016-01-11:
Hi all,
I want to send real-time data to webserver, I modified example HelloServer at (https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/HelloServer)
but: It appear error. 
Error appear on Terminal Serial when I debug it:
/*        ERROR       */
.......
Connected to Song Phuc
IP address: 192.168.50.61
MDNS responder started
HTTP server started

Exception (29):
epc1=0x40208059 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff0a80 end: 3fff0d40 offset: 01a0

>>>stack>>>
3fff0c20:  3fffbfb0 4010683a 3fff0cd0 40208055  
3fff0c30:  3fff2008 4020a892 00000000 3ffe8538  
3fff0c40:  3fff2030 00000002 40206a44 40206a28  
3fff0c50:  4020a8c4 3fff0ca0 3ffef228 3fff0cec  
3fff0c60:  3fff0cec 3ffe8538 3ffef228 402080cc  
3fff0c70:  3fff1ff0 4020a892 00000000 3ffef228  
3fff0c80:  3fff2018 3ffef218 40206a44 40206a28  
3fff0c90:  00000000 3ffef218 3ffe8538 4020b9ee  
3fff0ca0:  4010684c 3ffef228 3ffe8538 40208105  
3fff0cb0:  3fff2000 4020a892 40206a44 40206a28  
3fff0cc0:  3ffe86e4 3ffef218 3ffef228 40206c7a  
3fff0cd0:  3fff1ce8 00000000 40206a44 40206a28  
3fff0ce0:  3fffbfc0 0000000f 00000002 00000000  
3fff0cf0:  00000000 00000000 3fff2000 0000000f  
3fff0d00:  00000002 00000000 00000016 40101941  
3fff0d10:  4020a9ad 00000000 00000000 3ffefd10  
3fff0d20:  3fffdc20 00000000 3ffefd09 4020a9d5  
3fff0d30:  00000000 00000000 3ffefd20 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
/*    The End ERROR
/*   Code myESP8266 here: */
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "Song Phuc";
const char* password = "anhkhongbiet";

ESP8266WebServer server(80);

const int led = 2;
uint8_t i;
String s;
//void handleRoot() {
//  digitalWrite(led, 1);
//  server.send(200, "text/plain", "hello from esp8266!");
//  digitalWrite(led, 0);
//}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }
//  server.on("/", handleRoot);

//  server.on("/inline", [](){
//    server.send(200, "text/plain", "this works as well");
//  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  //prepare mydatai  
  i++;
  if(i>100){
    i=0;  
  }  
  //modify mydata
  s="";
  s=String(i)+" "+String(i+10)+" "+String(i+20);
  
  //update data to webserver.
  server.on("/", [](){
    server.send(200, "text/plain", s);
  });
  server.handleClient();
  //
  delay(10);
}
/*   The end Code MyESP8266    */

Thanks,
Best,
binhdv92

Links2004, 2016-01-11:
the 
```
 //update data to webserver.
  server.on("/", [](){
    server.send(200, "text/plain", s);
  });
```
need to be in the setup.
you create a new event handler every loop, this overload the ESP.

binhdv92, 2016-01-11:
Hi Links2004,
Thanks for your advice, Are you want to me using Interrupt?
If I create the
//update data to webserver.
  server.on("/", [](){
    server.send(200, "text/plain", s);
  });
in function "void setup()" then I don't update real-time data to webserver. :-/
I Thinks It is overloaded  ESP's RAM (leak memory) but I don't know how to reset or refresh ESP's Memory.
Best,
binhdv92.

Links2004, 2016-01-11:
why it shut not be updated when you have it in setup?
String s; is global
when you change s the next request gets the new data.

binhdv92, 2016-01-11:
Hi Links2004,
Thank you very much, I have followed your advice and I see result very good. :+1: 
and here I have adviced mycode for ESP8266:
/*         code for eSP8266   */
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "DVB_LOBE";
const char* password = "google.com";

ESP8266WebServer server(80);

const int led = 2;
uint8_t i;
String s;
void handleRoot() {
  digitalWrite(led, 1);
  //modify mydata
  s="";
  s=String(i)+" "+String(i+10)+" "+String(i+20)+"10";
  server.send(200, "text/plain", s);
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }
  server.on("/", handleRoot);

//  server.on("/inline", [](){
//    server.send(200, "text/plain", "this works as well");
//  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  //prepare mydata 
  i++;
  if(i>600){
    i=0;  
  }  
   
  //update data to webserver.
  server.handleClient();
  delay(1);
}
/*     the end code   */

Best,
binhdv92.

closed 2016-02-29 by igrr