esp8266/Arduino #1406: Multiple libraries were found for "OneWire.h"

andig, 2016-01-11:
Bundled OneWire conflicts with default one. Unfortunately the IDE chooses the standard lib instead of the bundled one:

	Multiple libraries were found for "OneWire.h"
	 Used: C:\Users\xx\Documents\Arduino\libraries\OneWire
	 Not used: C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\OneWire

Is there a clean solution (without renaming the standard lib) that would allow me to keep using IDE for non-ESP8266 projects, too?

sej7278, 2016-01-15:
replace the standard lib with the esp one, still works on avr's etc.

if you look, its just the standard OneWire.h with an ESP8266 ifdef block added:

https://github.com/esp8266/Arduino/blob/master/libraries/OneWire/OneWire.h#L107

really it needs to get merged into the upstream lib i guess

andig, 2016-01-15:
Good to know, thanks!

andig, 2016-01-16:
I've just seen that Paul has actually updated OneWire: https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.h#L122 and release 2.3.1: https://github.com/PaulStoffregen/OneWire/issues/7

Wouldn't it make sense to avoid further confusion to remove OneWire from this repo as it's fixed upstream?

sej7278, 2016-01-16:
actually he's released 2.3.2 and the esp8266 changes got merged in a couple of months back:

https://github.com/PaulStoffregen/OneWire/commit/4ae7f068be9dab4ef2287512af695c20aa6c54b4

so yes, probably better to remove it from this repo and maybe just mention a link to it in the readme

closed 2016-01-18 by igrr