esp8266/Arduino #1410: yield() in delay()

gioblu, 2016-01-12:
How much coherent with the requested value is the delay time produced, and how much is affected by yield() and so execution of other tasks? Can this lead to overshoot the requested time?
If this happens maybe yield() function should be called consciously by user and not forced inside delay(), not to influence some use cases like bit-banging.

igrr, 2016-01-12:
Short answer: delay actually operates by doing a context switch (via
esp_yield), it's just how it is implemented. Other tasks may affect timing,
but on a single core network processor this is pretty much inevitable.

delayMicroseconds doesn't do a context switch, so if you want bitbanging
you should probably use that instead of delay.

gioblu, 2016-01-12:
Perfect, is the answer I was hoping for :)
Thank you!

closed 2016-01-12 by igrr

asetyde, 2016-01-12:
Sorry Igor but I  not clear clearly , context switch can be a factor of block device ? Sometimes my device it block with connector MySQL , I use a conter to escape from delay in function but to work connector need pauses , which of this function Delay or delay micro. Is best to do this more safely possible

Links2004, 2016-01-12:
if you need to "pauses" be course you wait for data you need to use delay or yield,
or you will never get data (the network stack need to run).

asetyde, 2016-01-12:
But difference delay() and yield