esp8266/Arduino #1411: Can't copile example and My sketch (Arduino 1.6.7) and Git Repo....

Lstt2005, 2016-01-12:
After compile https://github.com/esp8266/Arduino/blob/342c4ae6fb847bfc787f80b89a2bb888d942dc32/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino
have this erros:
"
In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.h:26:0,
                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:26:
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiType.h:34:14: error: multiple definition of 'enum WiFiMode'
 typedef enum WiFiMode {
              ^
In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:25:0:
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFi.h:39:6: error: previous definition here
 enum WiFiMode { WIFI_OFF = 0, WIFI_STA = 1, WIFI_AP = 2, WIFI_AP_STA = 3 };
      ^
In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.h:26:0,
                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:26:
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiType.h:36:13: error: invalid type in declaration before ';' token
 } WiFiMode_t;
             ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp: In member function 'bool ESP8266WiFiAPClass::softAP(const char*, const char*, int, int)':
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:86:14: error: 'class ESP8266WiFiClass' has no member named 'enableAP'
     if(!WiFi.enableAP(true)) {
              ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:88:45: error: 'DEBUG_WIFI' was not declared in this scope
         DEBUG_WIFI("[AP] enableAP failed!\n");
                                             ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:94:54: error: 'DEBUG_WIFI' was not declared in this scope
         DEBUG_WIFI("[AP] SSID too long or missing!\n");
                                                      ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:100:62: error: 'DEBUG_WIFI' was not declared in this scope
         DEBUG_WIFI("[AP] fail passphrase to long or short!\n");
                                                             ^
In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:25:0:
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFi.h:390:10: error: 'bool ESP8266WiFiClass::_persistent' is protected
     bool _persistent;
          ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:127:17: error: within this context
         if(WiFi._persistent) {
                 ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:135:51: error: 'DEBUG_WIFI' was not declared in this scope
             DEBUG_WIFI("[AP] set_config failed!\n");
                                                   ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:140:52: error: 'DEBUG_WIFI' was not declared in this scope
         DEBUG_WIFI("[AP] softap config unchanged\n");
                                                    ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:144:58: error: 'DEBUG_WIFI' was not declared in this scope
         DEBUG_WIFI("[AP] DHCP not started, starting...\n");
                                                          ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:156:63: error: 'DEBUG_WIFI' was not declared in this scope
             DEBUG_WIFI("[AP] IP config Invalid resetting...\n");
                                                               ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:165:53: error: 'DEBUG_WIFI' was not declared in this scope
         DEBUG_WIFI("[AP] wifi_get_ip_info failed!\n");
                                                     ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp: In member function 'bool ESP8266WiFiAPClass::softAPConfig(IPAddress, IPAddress, IPAddress)':
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:180:150: error: 'DEBUG_WIFI' was not declared in this scope
     DEBUG_WIFI("[APConfig] local_ip: %s gateway: %s subnet: %s\n", local_ip.toString().c_str(), gateway.toString().c_str(), subnet.toString().c_str());
                                                                                                                                                  ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:181:14: error: 'class ESP8266WiFiClass' has no member named 'enableAP'
     if(!WiFi.enableAP(true)) {
              ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:218:45: error: 'wifi_softap_set_dhcps_lease_time' was not declared in this scope
     if(!wifi_softap_set_dhcps_lease_time(720)) {
                                             ^
In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:25:0:
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFi.h: In member function 'bool ESP8266WiFiAPClass::softAPdisconnect(bool)':
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFi.h:390:10: error: 'bool ESP8266WiFiClass::_persistent' is protected
     bool _persistent;
          ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:265:13: error: within this context
     if(WiFi._persistent) {
             ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:273:57: error: 'DEBUG_WIFI' was not declared in this scope
         DEBUG_WIFI("[APdisconnect] set_config failed!\n");
                                                         ^
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:277:20: error: 'class ESP8266WiFiClass' has no member named 'enableAP'
         ret = WiFi.enableAP(false);
                    ^
Используем библиотеку ESP8266WiFi версии 1.0 из папки: C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
exit status 1
Ошибка компиляции.
"

Links2004, 2016-01-12:
look like something gets mixed up.
ESP8266WiFi.h has no enum at line 39 in git.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L39

try to delete all an clone the repo again and run the get.py.

Lstt2005, 2016-01-12:
Can you say to newby where is try to delete mixed NO ENUM? :)

Links2004, 2016-01-12:
goto:
```C:\Program Files (x86)\Arduino\hardware\esp8266com```
delete the ```esp8266``` dir

and run 
```bat
git clone https://github.com/esp8266/Arduino.git esp8266
cd esp8266/tools
python get.py
```

https://github.com/esp8266/Arduino#using-git-version-

Lstt2005, 2016-01-12:
I just allready run this procedure twice with reinstalling Arduino (clean install)..Seems this error in master REPO?

Links2004, 2016-01-12:
the master on git is ok see:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L39
no enums there, the problem is local at your setup.

have you the staring or stable still installed too?
if yes delete them too.

Lstt2005, 2016-01-12:
Absolutly clean install on other PC (Windows 8.1) - the same error...(

Links2004, 2016-01-12:
tested on Win 8.1 x64 (core: latest master)
Arduino 1.6.5:
[image]

Arduino 1.6.7:
[image]

working fine here.

Lstt2005, 2016-01-12:
Thanks for all!

closed 2016-01-12 by Lstt2005