esp8266/Arduino #1417: PWM output with analogWrite changes frequency instead of duty cycle

tgmann, 2016-01-12:
With 2.1.0-rc1 I see that when using analogWrite with different values changes actually the frequency on the output, with always a 50% duty cycle, while I was expecting a fixed frequency, and a changing duty cycle.

e.g. I get a frequency on 40 kHz for a analogWrite(2, 1013);, and a frequency of 505 Hz with analogWrite(2, 11); , all with a duty cycle of 50%.

When I switch back to release 2.0.0-rc2, the behaviour is as expected.

This could be reproduced on NodeMCU 1.0 and ESP-01 modules, using Arduino 1.6.5. toolchain.

tgmann, 2016-01-13:
This is probably a duplicate of #1382 .

devyte, 2017-10-18:
AnalogWrite() works for me on latest git on a WittyCloud. Also, the referenced duplicate issue is closed.
Closing as can't reproduce.

closed 2017-10-18 by devyte