esp8266/Arduino #1420: client.connect() takes too long of server is unavailable...

syalujjwal, 2016-01-13:
In case the server is unavailable the client.connect() takes too long to return false come out of the block. This takes 1.5 secs approx. which is too long for my code. Is there a way to get past this problem? Can there be a timeout set in case the client cannot connect within N secs?

MarcusF2015, 2016-01-18:
If have got a similar problem. Sometime the Client.connect() of a WIFISecureClient takes too long to return and causes a WDT reset.

syalujjwal, 2016-02-02:
Hi...any solution for this?

syalujjwal, 2016-02-11:
Hi...still waiting. Please help!!!

marvinroger, 2016-03-17:
connect() takes 5 seconds for me, but no WDT reset.

battuashwik, 2016-05-14:
Any update on this. This is becoming a blocker for applications that need to do other jobs when not connected to the server, something like saving state to eeprom.

igrr, 2016-05-14:
Please quantify "too long". Current behavior is to wait until either "connected" or "connection refused", but not more than 5 seconds. I can change this to the same timeout which is set using `client.setTimeout`, will this help?

syalujjwal, 2016-05-14:
This should help. I mean any feature that allows the coder to manually set
number seconds to "wait for client or move on" is necessary in many cases.

Please let up know what can be done for this. It's required urgently as
some finished products are facing this issue.

Thanks

battuashwik, 2016-05-14:
@igrr, that would help a bit but a better way would be to make it asynchronous and let the user monitor a status variable to check if the connection is successful. This way the code is non blocking and use cases where there are other tasks to perform while waiting for the connection can also be done.

Ex.

```
void loop() {
if (CLIENT_STATUS == "WAITING") {
 // meanwhile can do other jobs also
} else if (CLIENT_STATUS ==  "CONNECTED"){
 // do watever needs to be done after connection
} else {
 client.connect(IP_ADDRESS, MASTER_PORT);
}
}
```

igrr, 2016-05-14:
@battuashwik If you need asynchronous operation, then you should probably use [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP) library.

battuashwik, 2016-05-14:
Completely missed this in the mentioned libraries. Thanks a lot for the help.

syalujjwal, 2016-05-16:
Its great that now this is into consideration. Just wanted to know how can
I come to know about the new updates and development on this, and when the
new feature is alive?

igrr, 2016-05-16:
@syalujjwal could you please clarify what are you referring to in your last question: ESPAsyncTCP library, or changing `client.connect` timeout to the same value as `client.setTimeout` ?

syalujjwal, 2016-05-16:
both are good enough solutions for me. I wish to use a library that allows
me to get control over how much time the client.connect blocks the code
for. Whatever it is that you are working on first, i would love to know
more about it and its availability. :)

igrr, 2016-05-16:
You can totally use [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP) library right now.

syalujjwal, 2016-05-16:
so cool. Thank you :)

szpght, 2016-09-11:
I have the same problem. I would be grateful for changing `client.setTimeout` behavior, or a little tip where to look in the code :)

terryjmyers, 2017-04-16:
While I understand the 5 second behavior, I don't want an asyncronous library for my application.  What would it take to get a the client.setTimeout function working as one would expect it to?

closed 2017-05-19 by igrr

jarnolbrown, 2017-10-04:
If this was done in May, how can I update my Arduino IDE installation to include it? I seem to be stuck on 2.3, and I can see from viewing the source files that the client.connect does not use the timeout set by setTimeout, bt comparing with the src from the repository. I tried the "staging" option, but that seemed to break everything again, and after a day of uninstalling, re-installing and rebooting, I'm loathe to try it again. I'm not stupid, I've been writing code for over 40 years, but I'm not au fait with all the Git stuff.
Any advice appeciated.

papadenis, 2020-08-14:
Occasionally it happens that the `request->client()->close()` command fails on the AsyncWebServer, the connection remains open and the client sticks. It's needed to take care of resetting this connection from the client side by timeout. Something like this:

```
if (client.connected()) {
    client.print(tail);
    int http_code = 0;
    
    unsigned long cnnct_timeout = 5 * ONE_SEC;
    if (!millis_fails(cnnct_timeout, global.millis_range)) {
        unsigned long cnnct_timestamp = millis();

        while ((client.connected() || client.available())
                && millis() < cnnct_timestamp + cnnct_timeout) {
            if (client.available()) {
                String line = client.readStringUntil('\n');
                if (line.substring(0, 8) == "HTTP/1.1")
                    http_code = line.substring(9, 12).toInt();
            }
        }
        if (millis() >= cnnct_timestamp + cnnct_timeout)
            logOutput(global.chipid, "Released by timeout");
    }

    if (http_code != 200)
        uploaded = false;
}
```

PR #3257 (2017-05-19): Configurable timeout for WiFiClient connect and write operations
WiFiClient inherits `Stream::setTimeout` method. The value passed to `setTimeout` is used by `Stream` methods like `readBytes`, `readStringUntil` and so on. Other operations with WiFiClient such as `connect` and `write` previously didn't have configurable timeouts. `connect` would time out after 5 seconds, and `write` would return only if LwIP reported an error.

This change implements timeout during `connect` and `write`. For `connect`, same timeout value is used for name resolution and the actual connection, so it is possible t [... cut]