esp8266/Arduino #1422: OTA End Failed on spiffs upload

hallard, 2016-01-14:
Hi there,
I'm using OTA for long time without any problem in command line with espota.py.

Starting few days, I'm not able to upload SPIFFS anymore (sketch upload works fine) and I didn't changed anything in my code related to OTA
```
build0c3ea91603fe6795abd4c568f834b507.tmp # ota -i 192.168.1.229 -r s- -f ../build0c3ea91603fe6795abd4c568f834b507.spiffs/MultiSensor.spiffs.bin -a ch2i_MultiSensor
Uploading: [============================================================] 100% Done...

12:36:29 [ERROR]: No Result!
build0c3ea91603fe6795abd4c568f834b507.tmp # s
```

reaper7, 2016-01-14:
did you update ESP8266FS to 0.2.0?
https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system

https://github.com/esp8266/Arduino/commit/b4ef8cf0fab1f28ae1297f8d08512df92352328f

hallard, 2016-01-14:
@reaper7 
Nope, I will thanks to mention
But not sure it will solve my problem, because I did not mention but SPIFFS upload from Arduino IDE (Serial) works fine, it's just from OTA with command line with espota that I've got this problem (can't test OTA from IDE device does not appears on my Windows)

hallard, 2016-01-14:
Updated, but as I suspected, same thing SPIFFS over OTA ends with error, so strange...

andig, 2016-01-16:
I've just noted this: https://github.com/esp8266/Arduino/issues/1450 However I can only reproduce if the esp is busy.

hallard, 2016-01-23:
yeah, correct if browser is making request to ESP during upload it sometimes fail closing browser and this problem went away :-)

closed 2016-01-23 by hallard