esp8266/Arduino #1427: Best way to process and avoid lock into "long" user while loop

hallard, 2016-01-14:
I've got one sensor reading that can take to about 300ms (this is not timing critical, so even if it's longer, does not matter) to issue measurement and I'm wondering how to get this "delay" the most optimized to avoid any lockup on webserver/websocket/OTA routine (all network stuff)

Of course I could state machine all of this and manage in main loop() but I'm too lazy right now, and I want to keep things simple as possible. So today I'm doing like using only a delay (that yield() into) this but I'm sure it's not the best way to do. 

```arduino
// True if okay, false if timed out
bool sht1x_waitForData(void)
{
  bool data=1;
  long start = millis();

  pinMode(SHT1x_DATA_PIN, INPUT);
 
  // until ok or timeout (320 ms max with 14bits, we take 330 ms)
  while ( data==1 && millis()-start <= 330 ) {
    delay(5);
    data = digitalRead( SHT1x_DATA_PIN );
  } 

  return !data;
}
```
So I'm wondering if it will be safe calling same networks process as "loop()" process during the while loop such as something I like more avoiding overhead of delay timer and processing all that I need. Note this one does not have any delay(), it's checked also by ms elapsed.

```arduino
// True if okay, false if timed out
bool sht1x_waitForData(void)
{
  bool data=1;
  long start = millis();

  pinMode(SHT1x_DATA_PIN, INPUT);
 
  // until ok or timeout (320 ms max with 14bits, we take 330 ms)
  while ( data==1 && millis()-start <= 330 ) {
    // Do all related network stuff
    server.handleClient();
    ArduinoOTA.handle();
    data = digitalRead( SHT1x_DATA_PIN );
  } 

  return !data;
}
```

Any hints on the best way to do validated by ESP's guru will be appreciated ;-)

Links2004, 2016-01-14:
how look the rest of the code?
may you can make it async.
like:
```cpp
if( digitalRead( SHT1x_DATA_PIN )) {
 // handle data here
}

if((millis()-start) > 330){
 // timeout
}
```

hallard, 2016-01-14:
@Links2004 
yeah, sure I can do like you say, async, it's better (and mainly the best way) but add some coding and all other should also be async, when you have some it can be harder to manage ;-)

But my question was also for my knowledge of "best practices" on this target during some while loop (let's says >100ms), sometime adding just 2/3 calls (so lines) can be easy do to and to read ;-)

WereCatf, 2016-01-14:
"Do I need any delay() or yield() in this one ?" -- Well, yes, server.handleClient() and ArduinoOTA.handle() doesn't make the need for delay() or yield() go away -- they don't do the same things. I would also do it async, like Links2004 says, but at least keep a yield() there in your loop if you wanna keep the structure as-is.

hallard, 2016-01-23:
@WereCatf 
Thanks, that's what I was assuming, I need to handle net and yield ;-)
- OTA.handle()
- server.handle()
- delay(x) or yield()

closed 2016-01-23 by hallard