esp8266/Arduino #1428: ESP8266WebServer returning empty requests

tzapu, 2016-01-14:
i ve noticed this issue using the latest github, all pages served in WiFiManager return empty pages after commit e8d60be62a9b08457bed597d721ff44afa57de67
another user reported it here
https://github.com/tzapu/WiFiManager/issues/58

i updated to the latest github and reverted just that once change 
```
-    if (_contentLength == CONTENT_LENGTH_NOT_SET) {
-        sendHeader("Content-Length", String(contentLength));
-    } else if (_contentLength != CONTENT_LENGTH_UNKNOWN) {

```
to
```
+    if (_contentLength != CONTENT_LENGTH_UNKNOWN && _contentLength != CONTENT_LENGTH_NOT_SET) {
         sendHeader("Content-Length", String(_contentLength));
     }
+    else if (contentLength > 0){
+    }
```

as it was before, and pages are served properly again...

igrr, 2016-01-14:
/cc @shmuelzon

igrr, 2016-01-14:
On the second thought, i think your code relied on the previous bad behaviour (content length not being sent for empty payloads).
You need to update the code to include a call to 
```c++
server->setContentLength(CONTENT_LENGTH_UNKNOWN);`
```
before you do `server->send(...`.

shmuelzon, 2016-01-14:
For what it's worth, I agree with Ivan. As far was web servers are concerned, you should always send the content length.
If you do not know what it is before hand, set it to UNKNOWN. The better option would be to use chunked encoding but I don't know if the current API supports that. Should be simple enough to add if not.

tzapu, 2016-01-14:
cool, thanks guys, will update and report

tzapu, 2016-01-14:
seems to work fine guys, thank you very much
do you think this change would be backwards compatible with stable 2.0.0 ?
cheers

igrr, 2016-01-14:
Should be compatible, we have similar code in SDWebServer example.

tzapu, 2016-01-14:
ok thank you, updated my code

closed 2016-01-14 by tzapu