esp8266/Arduino #1429: SPI on ESP12e CLK is missing

Pfannex, 2016-01-14:
Hi,
I use this little Sketch to test the SPI output (MOSI) on my 12e.

```
  #include <ESP8266WiFi.h>
  #include <SPI.h>             //SPI

void setup() {

  Serial.begin(115200);

  SPI.begin();
    SPI.setDataMode(SPI_MODE1);
    SPI.setClockDivider(SPI_CLOCK_DIV128);

  //SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE0));  
  //SPI.setFrequency(1000000);
}

void loop() {
  SPI.transfer(B01010101);
  delay(1000);
}
```

I expected SPI on the following Pins of the 12e:

Pin 13 - DATA-Output
Pin 14 - Clocksignal
Pin 9 - ChipSelect

Now I miss the Clocksignal on Pin14?
Also I wonder why the MISO Pin also changes?

[image]

[image]

tbnobody, 2016-01-14:
The pins 9 - 14 are just to access the integrated eeprom. You should not use them. The SPI functions are avialable on the pins 6 - 8.
Please look at this description:
https://github.com/esp8266/Arduino/blob/master/doc/esp12.png?raw=true

Pfannex, 2016-01-15:
@tbnobody 

OK, thanks for this hint!

Should the code from above run without any changes?
Or is it necessary to declare the SPI pins?

Did you have a short example?

Pfannex, 2016-01-15:
SPI works now!

Is it correct that the CS must be set manually?

[image]

```
  #include <ESP8266WiFi.h>
  #include <SPI.h>             //SPI

  #define CS 15

void setup() {
  pinMode(CS, OUTPUT);
  Serial.begin(115200);

  digitalWrite(CS, HIGH);
  SPI.begin();
    //SPI.setDataMode(SPI_MODE3);
    //SPI.setClockDivider(SPI_CLOCK_DIV128);
    //SPI.setFrequency(1000000);
    //SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE0));  
    //SPI.setFrequency(1000000);
}

void loop() {
  digitalWrite(CS, LOW); 
  SPI.transfer(B01010101);
  digitalWrite(CS, HIGH);
  delay(500);
  
  digitalWrite(CS, LOW); 
  SPI.transfer(B10101010);
  digitalWrite(CS, HIGH);
  delay(500);
}
```

chaeplin, 2016-01-15:
I have a nrf24l01p(spi radio) with esp12.

RF24 radio(CE_PIN, CS_PIN);
RF24 radio(3, 15);

So I think manual CS is correct. 
https://github.com/TMRh20/RF24/blob/master/RF24.cpp#L60-L82

Links2004, 2016-01-15:
a automatic CS is possible, but not standard in arduino so its disabled by default.
(auto CS is on GPIO15)
https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.h#L58
https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L63-L74

Pfannex, 2016-01-15:
Great Guys!!

The same problem like ever......
hardly done properly, it did work
Thanks for your support....

[image]

```
  #include <ESP8266WiFi.h>
  #include <SPI.h>             //SPI

void setup() {
  Serial.begin(115200);

  SPI.begin();
    SPI.setHwCs(true);
    //SPI.setDataMode(SPI_MODE3);
    //SPI.setClockDivider(SPI_CLOCK_DIV128);
    //SPI.setFrequency(1000000);
    //SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE0));  
    //SPI.setFrequency(1000000);
}

void loop() {
  SPI.transfer(B01010101);
  delay(500);
 
  SPI.transfer(B10101010);
  delay(500);
}
```

closed 2016-01-15 by Pfannex

sarwadenj, 2016-09-07:
I'm using esp12e, GPIO 12,13,14 pins  already i have connected to motors using mosfet.  And my other Gpio's are also busy in doing another task. 
Now i want to connect SPI bus so what can i do?
as gpio 12,13,14,15 having alternate function for spi and as gpio so how i configure it.

Netoperz, 2016-12-02:
As spa standard is telling, Hardcoded pins can be used together on many devices, let's say 4 spy dev. only ss and rat must be declared as unique for every device.

I've been reading that the ss pin should be pull up usia 10k resistor to assure that device that is not needed is not activated, in case of a few spy devices and a few SS pins they should not be left floating.

Correct me if i'm wrong that SS is active low, so pulling up assures deactivation of the device for the time that SPI is used to talk to another device.

Btw. why we cannot use the GPIO9 and GPIO10 as standard GPIO ?
And exactly why we cannot use SPI  on additional pins designed for spiced reader for other devices than SD ?

That would be very handy to have at least 2 GPIO's if not full second SPI.

MAMARR, 2017-02-03:
Hi Everyone,

I have ESP-12E only and i want to use the HSPI pins of ESP8266 ,i have seen  similar issue with you which you made working so can you please suggest which  #include <ESP8266WiFi.h>you have added in your project to make it work.

I tried another  #include <ESP8266WiFi.h> but it did not work.

Please guide me to resolve the issue.

Regards,
Vipin