esp8266/Arduino #1431: WiFiServer::close() does not close properly

mytrain, 2016-01-15:
When closing a tcp server connection with WiFiServer::close() then reopening it later with WiFiServer::begin() the tcp_bind failed with error code -13.

This is occuring only if some client connections were style alive when server was close.

nouser2013, 2016-01-15:
Close the connections individually on each client object as workaround for the time being, if you want to shutdown the server.

mytrain, 2016-01-20:
Unfortunately I'm not owner of client application (websocket).
In fact when client is:
-chrome on IOS via SAP it's closing correctly
-chrome on Windows 10 via STA it's NOT closing correctly

I'll run extra test Windows on SAP and IOS on STA to detect if issue is due to client of STA engine of ESP.

Keep in touch

igrr, 2016-06-13:
Fixed in #2140

closed 2016-06-23 by igrr