esp8266/Arduino #1435: mac 662 when sending UDP message "burst"

mvdbro, 2016-01-15:
I'm running an application that sends UDP messages to other ESP units (sort of ESP mesh network)

When I send an UDP message to all (currently 12) remote ESP units, once every second, it crashes within 5 to 10 minutes ending with a "mac 662" error. This mac 662 SDK messages seems related to TX issues.

I'm already using a delay(10) after each message transmit and adding more delay seems to solve the issue. But is there no smarter way to avoid the mac 662 errors?

Can the application check if the system is too busy?
Is there a queue for sending messages and can we read the length?

I'm not so keen on adding not so intelligent fixed delays in the application.

(And I'm not using UDP broadcast IP address here because it seems less reliable than using unicast, don't know why...)

igrr, 2016-02-29:
Does your application disable interrupts occasionally? If not, which libraries does it use?
This error usually happens when interrupts are disabled for too long.

igrr, 2016-02-29:
Is this related to/duplicate of #1229?

mkeyno, 2016-05-13:
hi @mvdbro any success on your plan , ? do you use the modem-router ? I've had same plan to dispense big amount of data(like 150-250kb) between dozen of ESP module, I use processing to  split  my big file  and add IP to each of chunks and send it to the my home router which is all esp connected to , but I'm not sure whether UDP can handle all of that in proper time , do you have any Idea abut that ?also I really thank you if you could share your codes

drmpf, 2016-05-13:
You can with a lot of coding, see my comment on #1009.  I cannot supply the code as I don't have a copy and it was proprietary.  But you can work it out for yourself in a few years.

devyte, 2017-10-18:
@mvdbro is this issue still valid with latest git? with PR #3362 ?

devyte, 2018-01-08:
Closing due to lack of feedback in over two months, lack of info in original post, and age.

closed 2018-01-08 by devyte