esp8266/Arduino #1436: Missing uart defines in tools/sdk/include/user_interface.h

tbnobody, 2016-01-15:
I tried to control a WS1812 LED using Serial1. To do this, it's required to invert the uart signal. 

In the original file (I think, https://raw.githubusercontent.com/espressif/ESP8266_IOT_PLATFORM/master/include/driver/uart_register.h) there is a define called 
```#define UART_TXD_INV                        (BIT(22))```
to invert the uart signal.
In the file for the arduino environment which is located here: tools/sdk/include/user_interface.h
this define, and also all other invert defines are missing.

Was this done on purpose or is this a mistake?

Links2004, 2016-01-15:
its there but not in the SDK notation
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/esp8266_peri.h#L262

me-no-dev, 2016-01-15:
```cpp
U0C0 |= (1 << USTXI);
```

tbnobody, 2016-01-15:
Oh sorry. Didn't see this define when looking at the source. Will close that issue.

closed 2016-01-15 by tbnobody