esp8266/Arduino #1438: How to get the AP configuration with ESP8266 classes? (AT+CWJAP_DEF?)

odilonafonso, 2016-01-15:
I want to know if the ESP-12e has already set the SSID and PASSW settings.
With a AT firmware I use the AT+CWJAP_DEF? command from arduino sketch.
Now I want to do the same on ESP8266 ESP-12e module, with the ESP8266WiFiClass (and similar)
The goal is to start a ESP module, if there is a Station access set use it; if not, set the AP mode, load a webserver in 192.168.4.1 and get the credentials (ssid and password).
Then, restart and use the credentials.
But, some times this works, sometimes not.

andig, 2016-01-15:
Take a look at the ESPWiFi.h. file- you should find all functions you need.

odilonafonso, 2016-01-15:
It has almost nothing there. Is that I'm with the wrong library?

mtnbrit, 2016-01-15:
I recommend the WiFimanager library by @tzapu.

andig, 2016-01-15:
Ist this what you need: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h

odilonafonso, 2016-01-15:
Yes !!

closed 2016-01-17 by odilonafonso