esp8266/Arduino #1441: UDP multicast : ethernet frame check sequence incorrect

Lolok33, 2016-01-15:
I send UDP multicast packet (100 bytes per packet every seconds) with the Esp8266.
When I use sdk 1.5.1, I have (with Wireshark) on each packet this error : ethernet frame check sequence incorrect.
With sdk 1.3.0, All packets are ok.
Is-it a bug of the sdk 1.5.1 ?

Links2004, 2016-01-15:
possible, can you provide a simple code that triggers the problem?

Lolok33, 2016-01-15:
An extract of my code :
 ```
void loop() {
  beginMulticastUDPPacket();
  // this loop is to check multiple call to write multicast
  for (int i = 0;  i < 10; i++)
  {
    cpt++;
    temps = millis();
    cptLecture = 0;
    buffer[cptLecture] = (byte)((cpt >> 24) & 0xFF);
    buffer[cptLecture + 1] = (byte)((cpt >> 16) & 0xFF);
    buffer[cptLecture + 2] = (byte)((cpt >> 8) & 0xFF);
    buffer[cptLecture + 3] = (byte)(cpt & 0xFF);
    buffer[cptLecture + 4] = '-';
    buffer[cptLecture + 5] = (byte)((temps >> 24) & 0xFF);
    buffer[cptLecture + 6] = (byte)((temps >> 16) & 0xFF);
    buffer[cptLecture + 7] = (byte)((temps >> 8) & 0xFF);
    buffer[cptLecture + 8] = (byte)(temps & 0xFF);
    buffer[cptLecture + 9] = ';';
    cptLecture += 10;
    writeMulticastUDPPacket(buffer, cptLecture);
  }
  endMulticastUDPPacket();
  
  delay(1000);
 } 
```
with
```
// UDP Packet begin sending function
// ---------------------------
void beginMulticastUDPPacket()
{
  udpTransponder.beginPacketMulticast(broadcastIP, UDP_TRANSPONDER_PORT, softAPIP);
}

// UDP Packet write sending function
// ---------------------------
void writeMulticastUDPPacket(byte* buffer, int size)
{
  udpTransponder.write(buffer, size);
}

// UDP Packet endd sending function
// ---------------------------
void endMulticastUDPPacket()
{
  udpTransponder.endPacket();
}
```
and
```
byte setup_eth_wifi()
{
   debug_println("Test Esp8266...");
  debug_println("");
  
  IPAddress IP_AP(192,168,4,1);
  IPAddress IP_Gateway(192,168,4,1);
  IPAddress IP_Subnet(255,255,255,0);

  Serial.setDebugOutput(false);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(espSSID);
  WiFi.softAPConfig(IP_AP, IP_Gateway, IP_Subnet);
  debug_println("ESP8266 Wi-Fi init done!");
  debug_println("");
  softAPIP = WiFi.softAPIP();
  debug_print("IP address: ");
  debug_println(softAPIP);
  debug_print("SDK Version: ");
  debug_println(system_get_sdk_version());
  
  return INIT_SUCCESS;
}
```

themindfactory, 2017-01-22:
I am also sending UDP in Arduino 2.3.0rc2 and get the same this, I am simply sending packets to the x.x.x.255 broadcast address and I get:

120	18.422965	192.168.4.1	192.168.4.255	UDP	69	8080→8080 Len=11 [ETHERNET FRAME CHECK SEQUENCE INCORRECT]

error in Wireshark also....

RichardS

igrr, 2017-05-05:
This has been fixed in ESP8266 non-OS SDK 2.1.0. Will try to do SDK update soon.

johnrford, 2017-06-16:
Hi, I have seen a similar problem but wanted to check a few details to verify that my issue is the same as this before opening a new case.
The issue I see is that frames are sent in Broadcast with ETHERNET FRAME CHECK SEQUENCE INCORRECT as shown in Wireshark. However, this only seems to be the case with frames of more than 8 bytes when using another ESP8266 as the AP. If I use my SKY router as the AP then every frame is sent correctly with no FCS error irrespective of length. In both cases I am using the sketch that was posted in a previous case to resolve an issue with Broadcast so I am confident that the sketch is correct.
Additionally I see that there are UDP frames not sent by the ESP8266 despite the sendPacket() sequence returning a 1 to indicate a send has been carried out. This occurs using both ESP and SKY routers as the AP.

I am in a good position to do any further testing or documenting is required.

John

j07rdi, 2017-10-13:
I have the same issue here. I'm sending messages with >25 bytes and I get "ETHERNET FRAME CHECK SEQUENCE INCORRECT". This issue will not happen if I send short amounts of bytes. 

I'm using Arduino with "Generic ESP8266 module".

devyte, 2017-10-18:
@Lolok33 @j07rdi please try 2.4.0-rc2 release candidate, or latest git.

johnrford, 2017-10-22:
Hi,

I'm am in the US at the moment so will try the code suggested when I return next month.

Best regards

j07rdi, 2017-10-24:
@Lolok33 It works, not more errors on Wireshark!! 

But is kind of laggy compared to direct UDP (i'm sending 63bytes at 10Hz). 

Thanks!

devyte, 2017-10-31:
@igrr this is reported to no longer happen with 2.4.0-rc2. Should it be closed, or not yet?

j07rdi, 2017-11-22:
@devyte @igrr This is no longer happening as per my last testing, but there is a very bad "lag" issue compared to direct UDP transmission.

devyte, 2017-11-22:
Please open a new issue to track that, and add timing details there.

closed 2018-01-02 by igrr