esp8266/Arduino #1442: can't connect to wss:// protocol

qblacksheep, 2016-01-16:
Hello,
at first i would like to say, im just started with ESP8266. I tried simple program form github:
https://github.com/morrissinger/ESP8266-Websocket/blob/master/examples/WebSocketClient_Demo/WebSocketClient_Demo.ino

When i am connecting with echo.websocket.org its working fine, i can send and read data from server. But i cant connect with server whose adress is: wss://example/device. Does anyone know the solution of this problem?

Thanks for help
Sorry for my english

penfold42, 2016-01-16:
Is the host "example" alive and dns resolvable on your network ?

marvinroger, 2016-01-16:
Try replacing `WiFiClient client;` by `WiFiClientSecure client;`.

Links2004, 2016-01-16:
use this one:
https://github.com/Links2004/arduinoWebSockets
it has ssl suport
https://github.com/Links2004/arduinoWebSockets/blob/master/examples/WebSocketClientSSL/WebSocketClientSSL.ino

qblacksheep, 2016-01-16:
Hi, thanks  for help but i cant complie this Example. 
[image]

Links2004, 2016-01-16:
you need the staring or git version of the ESP core

qblacksheep, 2016-01-16:
Thanks, its compiling now. But i still can't connect. In method webSocket.beginSSL should i write webSocket.beginSSL("wss://adress/device") or just webSocket.beginSSL("adress/device")? Defualt port is 443?

qblacksheep, 2016-01-16:
Do u know any test SSL server with can i test this program?

qblacksheep, 2016-01-16:
When i am trying to connect echo.websocket.org i get an exception: please start sntp first !

igrr, 2016-01-16:
This isn't an exception, just a debug message from the SDK. You can safely
ignore it.

If you are really getting an exception, it is likely unrelated to this
message.

qblacksheep, 2016-01-16:
Ok. But i still can't connect to any wss:// server. I was trying to connect to echo.websocket.org. 
[image]

qblacksheep, 2016-01-16:
. In method webSocket.beginSSL should i write webSocket.beginSSL("wss://echo.websocket.org") or just webSocket.beginSSL("echo.websocket.org")?

Links2004, 2016-01-16:
```webSocket.beginSSL("echo.websocket.org", 80);```

qblacksheep, 2016-01-16:
Now im getting this: 
[image]

Links2004, 2016-01-16:
see: #1375
the Certificate size / key length is too big

qblacksheep, 2016-01-16:
Do you know any test SSL server with path where i can test websockets? i can't connect with my server maybe couse i have wss://webrouter.server/path. What do you think?

Links2004, 2016-01-16:
```webSocket.beginSSL("webrouter.server", 443, "/path");```

qblacksheep, 2016-01-16:
Thanks for help guys. But unfortunately i still can't connect to server. Has that debug message:: "please start sntp first !" have impact to conneting the server?
[image]

Links2004, 2016-01-16:
normally not.
add ```configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");``` to the setup and it will be gone.
try to enable debug to see the reason for the disconect.
https://github.com/Links2004/arduinoWebSockets/blob/master/src/WebSockets.h#L30

qblacksheep, 2016-01-17:
[image]

Is that client or server failure?

Links2004, 2016-01-17:
the server returns 503 header so its a server problem, check the server logs.

qblacksheep, 2016-01-17:
Thanks a lot! :)

qblacksheep, 2016-01-18:
Did someone try to send and get ws from server on herokuapp.com? I can easily send and recive data on server which standing on local web but when i trying to connect to this same server standing on herokuapp.com its returns 503.

qblacksheep, 2016-01-18:
from server which i CAN connect i recive header connection:upgrade, from server which i CAN'T connect i recive header connection:keep-alive. How to change it? maybe there is a problem.

Links2004, 2016-01-18:
the server not accepts the WebSocket connection,
connection:upgrade is required for ws.
do your webserver really supports ws?
what software do you use as server?

https://tools.ietf.org/html/rfc6455#section-1.3

qblacksheep, 2016-01-18:
I use Netty deployed on Heroku. Server logs are:
Exception caught in Netty
org.jboss.netty.handler.codec.http.websocketx.WebSocketHandshakeException: Requested subprotocol(s) not supported: arduino

When I run server locally on Tomcat, it works fine

Links2004, 2016-01-18:
```Requested subprotocol(s) not supported: arduino```
is a good hint.
the client sends:
```
Sec-WebSocket-Protocol: arduino
```
the Server seams to not accept this.
you can try to remove this line:
https://github.com/Links2004/arduinoWebSockets/blob/master/src/WebSocketsClient.cpp#L369
if its working i can a option to disable/override it, then you can add something your server allows.

qblacksheep, 2016-01-18:
Its working now. Thanks a lot!!

qblacksheep, 2016-01-18:
It would be nice to have option to not include this header. Look at netty implementation http://netty.io/3.6/xref/org/jboss/netty/handler/codec/http/websocketx/WebSocketServerHandshaker13.html 
line 160-169. It throws exception only if there is unrecognized "Sec-WebSocket-Protocol" value. If there is no such header, netty uses default protocol and continues execution.

closed 2016-02-29 by igrr

selvabha, 2016-12-10:
Hello All,
I am facing issues with web-sockets.My node MCU works as a web-socket client.It connects fine and able to do a bi-directional communication but after some time my client is dead silently without notification.

  webSocket.beginSSL("YYYYYYYYYYYYYYYYYYYYYYYYY", 443, "/UUUUUUUUUUUUU");
  webSocket.setAuthorization("CCCCCCCCCCCCCCCCCC");
  webSocket.onEvent(webSocketEvent);

in the loop()
{
I was listening for the websocket event in the loop
}

even though it fails silently without notification. I am in the last phase of the project with a big issue.Please throw some light.

CarloSandoval, 2019-03-29:
hey there, I'm having exactly the same issue. I seem to be unable of removing the line of code that its emitting the Arduino protocol. Here is what I'm getting:

[WS-Client][sendHeader] sending header...
[WS-Client][sendHeader] handshake GET /chat HTTP/1.1
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: lJhypUjVIuYxris9IlVfjw==
Sec-WebSocket-Protocol: arduino
Origin: file://
User-Agent: arduino-WebSocket-Client

[write] n: 249 t: 125964
[WS-Client][sendHeader] sending header... Done (24231us).
[WS-Client][handleHeader] RX: HTTP/1.1 502 Bad Gateway
[WS-Client][handleHeader] RX: Server: nginx/1.10.3 (Ubuntu)
[WS-Client][handleHeader] RX: Date: Fri, 29 Mar 2019 06:19:52 GMT
[WS-Client][handleHeader] RX: Content-Type: text/html
[WS-Client][handleHeader] RX: Content-Length: 182
[WS-Client][handleHeader] RX: Connection: keep-alive
[WS-Client][handleHeader] Header read fin.
[WS-Client][handleHeader] Client settings:
[WS-Client][handleHeader]  - cURL: /chat
[WS-Client][handleHeader]  - cKey: lJhypUjVIuYxris9IlVfjw==
[WS-Client][handleHeader] Server header:
[WS-Client][handleHeader]  - cCode: 502
[WS-Client][handleHeader]  - cIsUpgrade: 0
[WS-Client][handleHeader]  - cIsWebsocket: 0
[WS-Client][handleHeader]  - cAccept: 
[WS-Client][handleHeader]  - cProtocol: arduino
[WS-Client][handleHeader]  - cExtensions: 
[WS-Client][handleHeader]  - cVersion: 0
[WS-Client][handleHeader]  - cSessionId: 
[WS-Client][handleHeader] no Websocket connection close.
[write] n: 27 t: 127137
[WS-Client] client disconnected.
[WSc] Disconnected!

here are the logs for the nginx whicih is running as a reverse proxy server:

[29/Mar/2019:06:15:47 +0000] "GET /chat HTTP/1.1" 502 182 "-" "arduino-WebSocket-Client"
[29/Mar/2019:06:15:47 +0000] "This is a webSocket client!" 400 182 "-" "-"

and the error.log file:

upstream prematurely closed connection while reading response header from upstream, request: "GET /chat HTTP/1.1"

my theory so far is that the upstream nodejs server is returning a 502 bad request due to the Sec-WebSocket-Protocol: arduino sent. I would appreciate if you could give me some advice.

CarloSandoval, 2019-03-30:
> 
> 
> Its working now. Thanks a lot!!

how did you removed the Sec-WebSocket-Protocol: arduino?

CarloSandoval, 2019-03-30:
> `Requested subprotocol(s) not supported: arduino`
> is a good hint.
> the client sends:
>
> ```
> Sec-WebSocket-Protocol: arduino
> ```
>
> the Server seams to not accept this.
> you can try to remove this line: [...]

I'm having the same issue but I'm unable to remove it. does it have an option to disable it? or which line should I remove? I'm using latest version so perhaps it changed.