esp8266/Arduino #1444: MQTT security

probonopd, 2016-01-16:
What is the best way to implement MQTT security (only authenticated ESP8266 devices may publish and/or subscribe)?

According to http://mosquitto.org/man/mosquitto-conf-5.html, 

> MQTT provides username/password authentication as part of the protocol. Use the password_file option to define the valid usernames and passwords. Be sure to use network encryption if you are using this option otherwise the username and password will be vulnerable to interception.

Since it is clearly a very bad idea to send around passwords unencrypted, there should be a way to do this securely. The Mosquitto broker supports TLS out of the box, and provides authentication either via username/password, pre-shared keys or TLS client certificates.

Links2004, 2016-01-16:
the ESP supports SSL/TLS encryption.
```WiFiClientSecure client;```
its up to the library programmer to add support for it or not.

TLS is sure possible.
pre-shared key/client certificates is may possible its not well tested.

probonopd, 2016-01-16:
So i guess it's up to pubsubclient to make use of this. https://github.com/Imroy/pubsubclient/issues/18

probonopd, 2016-01-16:
Ladyada added an example on how to use it: https://github.com/adafruit/Adafruit_MQTT_Library/commit/e77be5b9ac0200bdb5036194e05576c3cca868e0

closed 2016-01-16 by probonopd