esp8266/Arduino #1446: GPIO9 and GPIO10 unuseable

WereCatf, 2016-01-16:
I have a Nodemcu 1.0 - board that has GPIO-pins 9 and 10 available and I've been wanting to make use of them, but it seems nothing I do works. There is constantly traffic on pin 10, even when I'm not using it, and doing pinMode(9, OUTPUT) causes watchdog-reset.

GPIO 9 and 10 should be available for use if you are driving your Flash in DIO-mode and boards.txt does appear to be using Flash in DIO-mode on Nodemcu, but if I had to hazard a guess something is still forcing it to operate in QIO and not honoring the settings in boards.txt

Any ideas?

WereCatf, 2016-01-16:
Okay, I suppose all that I've read saying you don't need to cut anything has been wrong. I modified boards.txt to use QIO and the device works fine, still -- ESP.getFlashChipMode() confirms it is running in QIO and that shouldn't be possible if the wires were, indeed, cut.

That also means I won't ever be able to use 9 and 10 :( Closing the issue.

closed 2016-01-16 by WereCatf

asetyde, 2016-01-18:
No way to use 9 pin , 10 it’s ok for me to digital read

robotiko, 2016-01-18:
After several weeks looking and asking, It seems there are several nodemcu devkit 1.0 boards. 
Some do support DIO mode correctly and can use GPIO9 and GPIO10 but most don't. 
I have 4 boards from different suppliers bought with several months between each order and none them can use those pins :(.

WereCatf, 2016-01-18:
@robotiko: That's interesting. I have a "LoLin Nodemcu V3" - board and, as I said, there is always traffic on GPIO10 and GPIO9 resets the board, and QIO-mode is working on my board (it shouldn't!). Kind of stupid, since one of the selling-points of Nodemcu is the availability of GPIO9/10.

Maybe this should be documented somewhere. It'd also be nice to know which manufacturers (if any) do produce the boards where those pins are useable.

asetyde, 2016-01-18:
http://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/

WereCatf, 2016-01-18:
@asetyde: that page doesn't say anything, whatsoever, about GPIO9 or GPIO10 and it's, as such, completely irrelevant to this discussion.

asetyde, 2016-01-18:
ok

robotiko, 2016-01-18:
@WereCatf , you are right. 
One of the main advantages of devkit 1 is the two extra gpios, however it seems that most boards around are not using the proper esp8266  module and doesn't work. 
I have the Amica and DoitAm. 
I have seen some guys that state that after setting DIO mode it works (be aware that it implies flashing in DIO mode and then restart).
The only one I was able to get proper details of the board, seem to have an amica from seedstudio. 

Just in case it helps. 

My conclusion: use a port expander.

WereCatf, 2016-01-18:
@robotiko: Yeah, I have already added an I2C - port expander on my shopping-list, but the downside with those is that you have to keep polling pin-states on them whereas with regular pins you can tie an interrupt to them for any changes. Polling is.. inefficient and crude.

Oh well, I hear ESP32 will have more GPIO-pins, I'll be sure to buy me some and love them to pieces once they're available.

Duality4Y, 2016-01-18:
if you choose your pinexpander well you can buy one with a IRQ output, and attach a interrupt to it :)

neilyoung, 2017-01-24:
The really strange thing about this is: If you are using Lua together with OpenESP SDK then it is absolutely possible to use GPIO9 as output and GPIO10 as input. To attempt the same with the Arduino libs ends up in WDT resets.

pardhu002, 2017-06-24:
for me GPIO10 only working as DIO but if define GPIO9 as DIO then it crashes on NodeMCU

neilyoung, 2017-06-24:
@pardhu002 Right. Found it by myself a while ago. GPIO9 is internally used to control the flash memory. GPIO10 can be used as input only.