esp8266/Arduino #1447: BasicOTA.ino does not seem to work

probonopd, 2016-01-16:
The way I understand it, running BasicOTA.ino should make the device available as a port in the Arduino IDE so that sketches can be uploaded over the air.

Running the `BasicOTA.ino` sketch with staging and the hourly Arduino IDE build, the sketch does not seem to advertise itself using dns-sd, at least it never shows up in the IDE as a port and using `avahi-browse -a` I also never see the device.

@celsworth added on gitter.im:
> for what its worth I don't seem to get any MDNS announcements from ArduinoOTA.begin() either, just never noticed before cos I don't use the MDNS part of it
> a bit of error checking might not go amiss: https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/ArduinoOTA.cpp#L108
there's no check whether MDNS actually worked

WereCatf, 2016-01-16:
The mDNS-responder doesn't understand mDNS-queries for all services. You have to do 'avahi-browse _arduino._tcp' to see the device, ie. query for a specific service. Also, dns-sd != mDNS.

WereCatf, 2016-01-16:
Oh, and I, at least, have to restart the IDE for it to see the device. After that it works fine, though.

probonopd, 2016-01-16:
> The mDNS-responder doesn't understand mDNS-queries for all services.

Here is a patch written my @me-no-dev that solves this:
[patch.txt](https://github.com/esp8266/Arduino/files/93044/patch.txt)

Also, @me-no-dev pointed out that when using the latest hourly Arduino IDE builds, one does __NOT__ have to select "OTA" but "Serial" when doing OTA (yes!).

me-no-dev, 2016-01-17:
Actually on any build 1.6.7 or above, you do not need to select OTA in upload method options.
As a matter of fact, if you do and you are running on a different port or have password enabled, it will fail to upload the new firmware.
Maybe option should be removed all together now that we have it supported officially and that to be noted in the OTA instructions.

WereCatf, 2016-01-17:
@me-no-dev: That's a good bit of information to know! I've been wondering why I can't get password-enabled OTA to work. I do heartily support the idea of better documentation -- both regarding this and in general ;)

probonopd, 2016-01-17:
So:
- [x] Write patch to make it show up in the IDE menu reliably (Thanks @me-no-dev)
- [ ] Discuss and merge that patch upstream, https://github.com/esp8266/Arduino/pull/1451
- [ ] Remove "OTA" option from the menu since it is now misleading
- [ ] Document how to get password-protected OTA in the BasicOTA.ino example sketch

probonopd, 2016-02-06:
Looks like the only thing left to do in order to close this is "Remove OTA option from the menu since it is now misleading"

igrr, 2016-02-29:
Created #1702 to track the remaining task, closing this one.

closed 2016-02-29 by igrr