esp8266/Arduino #1449: How to save wifi credentials = AT+UART_DEF="ssid","psk"

odilonafonso, 2016-01-16:
How to save the ssid and psk of the wifi I want to connect next time WITHOUT send ssid and psk.
With the AT firmware I have the AT+UART_DEF="ssid","psk" command. 
Doing this, the next time I restart the module, it will autoconnect to the wifi.

WereCatf, 2016-01-16:
You store the SSID and password in EEPROM. See e.g. the files in https://github.com/esp8266/Arduino/tree/master/libraries/DNSServer/examples/CaptivePortalAdvanced and especially the credentials.ino and adapt the code to your sketch.

odilonafonso, 2016-01-16:
wow... i'm going back to AT firmware. much more stable !!

igrr, 2016-01-16:
Just call WiFi.begin and pass it your credentials. These will be saved, so
next time your sketch runs, it will connect automatically.

andig, 2016-01-16:
> These will be saved, so next time your sketch runs, it will connect automatically.

You can even call the SSID() and psk() functions and read the credentials back. *Warning:* security hole when giving your esps to other people...

odilonafonso, 2016-01-16:
Hi Igrr... thanks 

A  WiFi.begin("mywifi_ssid","mywifi_psk") call will save the credentials on flash,
Like: AT+CWJAP_DEF="mywifi_ssid","mywifi_psk" ?

That is OK. But have to save in eeprom is not a good idea.

I can use WiFI.Mode(WIFI_AP_STA) for a webserver on Station mode and it will survive on restart ?
I used WiFI.disconnect () and it seemed that the credentials and the mode were reset.
I'm having some difficulty using the API, I found no documentation , only the source code.

I have a application that runs on Arduino (ProMini or Nano with ATMega 328) using ESP8266 (ESP-01) as a wifi. I have tested the AT firmware from espressif, the version:

 * AT version:0.50.0.0(Sep 18 2015 20:55:38)
 * SDK version:1.4.0 BOOT MODE
 * compile time:Sep 18 2015 21:46:52

is working very well.

Now I want to use my application without Arduino hardware, only ESP8266. On a ESP-07 or ESP-12e.

I have to use the libraries from:

https://github.com/esp8266/Arduino/tree/master/libraries  ???

or there is another way to do my job ?

Thank you.

Links2004, 2016-01-16:
you only need to call ```WiFi.begin("mywifi_ssid","mywifi_psk");```
the saving in flash is done in backround you not need to handle it.

if the ESP reboots it starts the WiFi like it where befor the reboot.

if you want to programm the ESP:
https://github.com/esp8266/Arduino#installing-with-boards-manager
for connecting the ESP:
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#generic-esp8266-modules

WiFi examples:
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples
HTTP:
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266HTTPClient/examples

all Arduino libraries that not use registers will work.
some examples:
https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#other-libraries-not-included-with-the-ide 

more info see:
https://github.com/esp8266/Arduino/tree/master/doc
https://github.com/esp8266/Arduino

odilonafonso, 2016-01-16:
Thank you Links2004 !
All settings made ​​in one session will be preserved for the next session ?
I noticed the existence of a variable  (_persistent) with the initial value of "true ".
This variable is used to maintain (or not) adjustments made in a given session ?
With the method:
WiFi.persistent (true / false)

Links2004, 2016-01-16:
```WiFi.persistent(false);``` disables the storage to flash.
as long you not change it (true) the WiFi settings are stored until you override it or deleted it.

odilonafonso, 2016-01-16:
Ok, thank you all guys !!

closed 2016-01-16 by odilonafonso

odilonafonso, 2016-01-17:
Thanks !!

*Odilon Cenamo*
*Vivo (11)95636-1956*

*Skype (11)3042-1956Nextel (11)7720-4140 Id 30*69343*

Goddur, 2017-12-18:
Can someone point to the methodology for reading out the contents of the prom?

devyte, 2017-12-18:
@Goddur if you mean EEPROM, just look at the docs for the EEPROM class.
If you mean the flash in general, there is an api for that. "Use the source, Luke!".