esp8266/Arduino #1450: 2.1.0-rc1: OTA error 4 when uploading SPIFFS on busy device

andig, 2016-01-16:
I've got a neat little sketch that does data collection and pushing to iot server. User interface is served from embedded webserver, loads of ajax calls from browser to device.

I'm using OTA via IP triggered from IDE. I've seen that OTA update reproducibly fails if the UI is open in the browser (i.e. connecting to esp via HTTP calls):

	[webserver] uri: /api/plugins args: 0
	[begin] Update SPIFFS.
	sleep disable
	[begin] _startAddress:     0x00300000 (3145728)
	[begin] _currentAddress:   0x00300000 (3145728)
	[begin] _size:             0x000FB000 (1028096)
	[core] OTA Start
	pm open,type:0 0
	LmacRxBlk:1
	... more of these
	LmacRxBlk:1
	state: 5 -> 2 (2c0)
	rm 0
	pm close 7
	premature end: res:0, pos:946176/1028096
	[core] OTA Error [4]
	ERROR[0]: No Error
	LmacRxBlk:1
	reconnect
	state: 2 -> 0 (0)
	f r-1, scandone
	state: 0 -> 2 (b0)
	state: 2 -> 3 (0)
	state: 3 -> 5 (10)
	add 0
	aid 3
	cnt 
	LmacRxBlk:1
	LmacRxBlk:1
	LmacRxBlk:1
	[webserver] uri: /api/status args: 0

It seems as if the network stack (?) does have problems with multiple connections. Any ideas? Free heap is >30k when OTA starts.

Links2004, 2016-01-16:
try the latest git 2.1.0-rc1 has SDK 1.5.0 there are some bugs in it.

andig, 2016-01-17:
Indeed wfm with git version after quick test. I'll keep an eye on it. Performance serving larger files over HTTP is still horrible.

FWeinb, 2016-01-17:
You can test #1401. It improved HTTP performance for me quite a bit.

andig, 2016-01-17:
Reopening: the Spiffs via OTA problem is still in the current git version. However, it's no longer OTA error 4, the log now looks like this:

	[begin] Update SPIFFS.
	sleep disable
	[begin] _startAddress:     0x00300000 (3145728)
	[begin] _currentAddress:   0x00300000 (3145728)
	[begin] _size:             0x000FB000 (1028096)
	[core] OTA Start
	LmacRxBlk:1
	LmacRxBlk:1
	LmacRxBlk:1
	LmacRxBlk:1
	LmacRxBlk:1
	LmacRxBlk:1
	.. keeps repeating

I will retest with #1401.

me-no-dev, 2016-01-17:
If you are using the ESP8266WebServer, both the server and OTA are blocking and they block each other. Once the OTA starts downloading the new firmware from the computer, the loop is stopped at that position until the Update is over and the ESP restarted (if success).
You say you have lots of ajax requests coming :) maybe see in your JS to detect error and stop the requests. I have also written a simple ajax queue to execute the requests one after another and to stop on error. maybe something like that will help.
You do close the server when OTA starts right? That will also make the new connections be closed instead of waiting to be responded to.

andig, 2016-01-17:
I'm not stopping the webserver on the ota start event as I didn't care for stuck calls (they all have a reasonable timeout set). If ota blocks a running webserver shouldn't be a cause of crashes or ota getting stuck?

andig, 2016-01-23:
tl:dr

I've migrated to the async webserver. After fixing some issues with code using `yield` (e.g. `Stream::readString`) I'm almost overwhelmed with the performance I'm getting. No more issues with parallel requests.

closed 2016-01-23 by andig