esp8266/Arduino #1452: Include OTA capability at compile time (without editing the sketch)

probonopd, 2016-01-16:
ArduinoOTA allows to upload sketches to the ESP8266 remotely without a wired connection. However, for this to work, each sketch needs to be modified to include the necessary calls. While it certainly isn't a huge problem to put those two calls into each sketch manually, it does add some complexity (especially for beginners) and, more importantly, it is easy to forget adding them when uploading random example sketches from the net or from the Examples menu.

So it would be nice if there was a checkbox `[x] Include OTA capability` that would "inject" the required code when compiling a sketch, without the sketch itself needing to be edited manually. @celsworth suggested this could possibly be done in the `user_main()` and/or at `loop_wrapper()`. 

https://github.com/esp8266/Arduino/blob/2b23b005aab605f1fb74bd2bbd388728b69ead86/cores/esp8266/core_esp8266_main.cpp#L99

The idea is to set a `#define` and it could call `.begin` and `.handle`.

Links2004, 2016-01-16:
the include in the sketch will still be needed or the IDE dont include the OTA library,
full automatical Integration Is possible but need some more work (have some ideas for this)

probonopd, 2016-01-16:
Looking for "full automatical integration". :-)

me-no-dev, 2016-01-17:
it actually is not a big deal at all to automate the whole thing. Here are some situations though:
1. What to do when the new sketch is uploaded? How to notify the user? Or should we?
2. What about the password? If it's all auto, how will we define that? There are more than enough security-paranoid guys out there ;)
3. If it's all behind the scenes, that would mean that none of it is running on the loop thread, therefore all callbacks must not spend time and interfere withe the rest of the sketch, which makes implementing a progress indicator somewhat problematic.

Everything else would take me about an hour to write.

toxuin, 2016-01-17:
My opinion: please don't do this. Going fully-automatic on such a device class as ESP8266 would release the kraken of the completely unprotected automation systems EVERYWHERE. It nears the IoT apocalypse that media has been advertising to inevitably come, and soon. My suggestion would be even force one to set a password to enable the remote firmware flash. I'm totally okay to include needed headers and add the calls. Because why not? Two lines of code IS easy enough for such an awesome feature as OTA updates. Seriously.

me-no-dev, 2016-01-17:
+1 on that, but it needs to go async. ArduinoOTA.handle() to be used only for progress callbacks and have the rest in the background.
I have had my fair share of loop screw-ups that would lock the loop waiting for something and me having to resort back to serial instead of just pushing a new firmware over the air.
Not sure for the mandatory password though. I guess if you have developed something that you put out in the open, you should know what you are doing and disable ArduinoOTA all together :)

probonopd, 2016-01-17:
@toxuin I don't think this would necessarily lead to less security. What we would need is a GUI window with widgets like this:

```
[ ] Enable OTA (unchecked by default)
Password: _____________________ (we can think of making this mandatory)
mDNS device name: ______________________ (with a reasonable default like esp8266-$ID)
```
Actually, I'd be very happy if I didn't have to enter my WLAN credentials into each sketch with the risk if inadvertently showing them in a screencast or uploading them to GitHub by mistake. So while we are at it, how about a second GUI where we can enter:

```
WLAN SSID: _____________________ 
WLAN Password: _____________________
```
Sketches should be able to use these by what we would call an "environment variable" in bash.

devyte, 2017-10-18:
This is interesting at first glance, but when you think about it for more serious applications, it is in fact a bad idea. My thoughts:
- ArduinoOTA is insecure, even when using a password
- The functionality is hidden from the developer, hence not too much thought would be given to the implications of its use
- It would mostly be used by beginners, who are supposed to be learning about the system, and not pursuing ease of use
- Consider the last comment about the wifi credentials. Where would the configurability end? We're not looking to make a build front-end for features

Overall, I think the status quo is better: force the developer to add the functionality explicitly, and think about it at least once. Given that at least one other person has expressed an opinion along the same lines, that this would add extra complexity to the core code for insufficient reason, and that there is a lack of feedback in over a year, I consider the request denied.
Closing.

closed 2017-10-18 by devyte