esp8266/Arduino #1453: SerialUploader RunnerException when trying to upload to /dev/ttyUSB0

probonopd, 2016-01-16:
I get this error with a generic ESP-12e module with staging and the most recent hourly of the Arduino IDE, 1.6.8.201601120435-x86_64. I did not get this error with an earlier hourly, but in those OTA was not working when "Upload using" was set to "Serial".

```
"/home/me/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-L/home/me/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/tools/sdk/lib" "-L/home/me/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/tools/sdk/ld" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "/tmp/build13178926a83e9830467010d44d1e824c.tmp/sketch_jan16a.ino.elf" -Wl,--start-group "/tmp/build13178926a83e9830467010d44d1e824c.tmp/sketch/sketch_jan16a.ino.cpp.o" "/tmp/build13178926a83e9830467010d44d1e824c.tmp/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -Wl,--end-group  "-L/tmp/build13178926a83e9830467010d44d1e824c.tmp"
"/home/me/.arduino15/packages/esp8266/tools/esptool/0.4.6/esptool" -eo "/home/me/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/bootloaders/eboot/eboot.elf" -bo "/tmp/build13178926a83e9830467010d44d1e824c.tmp/sketch_jan16a.ino.bin" -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "/tmp/build13178926a83e9830467010d44d1e824c.tmp/sketch_jan16a.ino.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec

Sketch uses 215,356 bytes (20%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 31,182 bytes (38%) of dynamic memory, leaving 50,738 bytes for local variables. Maximum is 81,920 bytes.
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:179)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2417)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NullPointerException
	at java.lang.String.replace(String.java:2240)
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:97)
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:89)
	at processing.app.helpers.StringReplacer.formatAndSplit(StringReplacer.java:37)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:174)
	... 6 more
```

crushedice2000, 2016-01-17:
I can reproduce the problem (polite version of the popular *Me too!*)

probonopd, 2016-01-23:
I also can't upload on the Mac using the hourly IDE build and the staging esp8266/Arduino core.

probonopd, 2016-01-23:
Seems not to be related to esp8266/Arduino, since I can reproduce the issue with an Arduino Genuino Pro Mini too. https://github.com/arduino/Arduino/issues/4471

closed 2016-01-23 by probonopd

vishnursnair222, 2017-06-03:
change the serialport permission 
sudo chmod 666 /dev/ttyACM0