esp8266/Arduino #1455: Missing file when using Arduino core for ESP8266 WiFi chip

weltenbrant, 2016-01-17:
I'm trying to load an arduino sktech to an ESP-07 ESP8266 modul.

When I try to upload the blink example sketch I get the following compiler error:
...AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++": file does not exist
Error compiling.

I use the following setup:
- arduino 1.6.7 with the core for ESP8266 WiFi chips on windows 7 64bit
- arduino setting: Board "Generic ESP8266 Module", Programmer "AVRISP mkII"
- FTDI USB/serial adapter connected to ESP8266 modul (connected RX->TX and TX-> RX, VCC and GND)
- ESP-07 ESP8266 modul with 5V to 3.3 level shifter: http://www.dx.com/p/esp-07-esp8266-serial-wi-f-wireless-module-w-built-in-antenna-compatible-with-3-3v-5v-for-arduino-400559#.Vptqe_l6RD8
- Switch on ESP modul set to "program" (to pulldown GPIO0)
- LED connected to GPIO13

When starting up the board without pulldown of GPIO0 the serial monitor shows that the board is ready. The connection seems to work. But somehow this file is missing... Does anyone has an idea what the reason could be and how to fix it?

Thanks

Links2004, 2016-01-17:
you need to download the build toolchain:
```bash
cd esp8266/tools
python get.py
```
https://github.com/esp8266/Arduino#using-git-version-

weltenbrant, 2016-01-17:
Okay, I used the installation option with the board manager. 

Now I also followed the instructions by using the git version. 
- extracted the zip file to C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266
- used the cmd "python get.py" --> installed properly
- restarted arduino an now I get the next error:

-----------------------------------------------------
Arduino: 1.6.7 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Arduino.h:27:0,

                 from sketch\Blink.ino.cpp:1:

c:\users\scherrer\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\stdlib.h:16:20: fatal error: stddef.h: No such file or directory

 #include <stddef.h>

                    ^

compilation terminated.

exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
-----------------------------------------------------

I have also tried to extract the zip file in the ...AppData\Local\Arduino15\hardware\esp8266com\esp8266 folder and run the "python get.py" command. I always get the same error....and I checked the library is not in folder mentioned in the error message.

weltenbrant, 2016-01-17:
okay, I reinstalled Arduino and used the git version... now it compiles

closed 2016-01-17 by weltenbrant