esp8266/Arduino #1460: Q: how to use  WiFiClientSecure::verify with ip address ?

chaeplin, 2016-01-18:
I am in testing mosquitto/tls1.2( user name and password) with pubsubclient.
It works well.

I can get SHA1 Fingerprint of server using  command "openssl x509 -fingerprint -in  mqttserver.crt".

How can i use WiFiClientSecure::verify with host with ip address ?

ref:
https://github.com/esp8266/Arduino/blob/14b70e9328e78c94ba0c0042766a11854ac389de/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino#L46-L56

https://github.com/esp8266/Arduino/blob/005d9d0bfd79b227409a0f68ae9a35d577090ef9/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp#L354-L385

chaeplin, 2016-01-18:
if (wifiClient.verify(fingerprint, mqtt_server.toString().c_str())) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }

//const char* fingerprint = "70 B2 BF 0D 4E 2A 54 FC DD C3 75 03 CD 42 20 71 9C 4A 97 37";
//connecting to 192.168.10.144
//certificate matches

// test
//const char* fingerprint = "70 B2 BF 0D 4E 2A 54 FC DD C3 75 03 CD 42 20 71 9C 4A 97 36";
//connecting to 192.168.10.144
//certificate doesn't match

closed 2016-01-18 by chaeplin

connercowling, 2016-08-10:
Did you ever figure this out? I'm having a DNS issue which is not allowing me to load the SSL cert using the host name, but does with the IP.  Can't seem to get the right IPAddress type to pass to the second method signature of `connect()`

chaeplin, 2016-08-10:
I did https://github.com/chaeplin/esp8266_and_arduino/blob/master/_33-esp-12-WiFiClientSecure-mqtt-ssl/_33-esp-12-WiFiClientSecure-mqtt-ssl.ino#L54-L68

connercowling, 2016-08-10:
@chaeplin I don't understand how `verifytls()` resolves this. Could you elaborate or provide a snippet of your code?  Again the ESP8266 is having problems resolving the domain name to fetch the SLL cert.   I'm trying provide the ip directly instead.  Was hoping to use the alternative function signature by providing an `IPAddress` type to `connect` but can't seem to instantiate an `IPAddress` type correctly.

chaeplin, 2016-08-11:
tls with ip :https://github.com/chaeplin/esp8266_and_arduino/blob/master/_33-esp-12-WiFiClientSecure-mqtt-ssl/_33-esp-12-WiFiClientSecure-mqtt-ssl.ino
tls with name : https://github.com/chaeplin/esp8266_and_arduino/blob/master/_33-esp-12-WiFiClientSecure-mqtt-ssl/_33-esp-12-WiFiClientSecure-mqtt-ssl.ino

pieman64, 2017-02-22:
@chaeplin may I ask if your sketches still work OK? I don't use mosquitto but I tried fingerprint validation of  an IP address based on your example of:

```
  if (wifiClient.verify(fingerprint, mqtt_server.toString().c_str())) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }
```

Always get certificate doesn't match but when I switch the IP for the domain name it's fine.
Any ideas?