esp8266/Arduino #1461: running out of heap

xnorbi, 2016-01-18:
Hello, I have a sketch, which is using the webserver.

After I while its crash. At the end of the loop function I have code:

Serial.printf("settings heap size: %u\n", ESP.getFreeHeap());

The free heap size is evry time is less. Finally I run out.

Is it exist any code to free heap? What should be observed to avoid running from the heap?

WereCatf, 2016-01-18:
Sounds like your code is allocating new variables at runtime, but not releasing the allocated memory and that's why you eventually run out. And no, there is no magic to freeing heap, you have to free any dynamically-allocated variables when no longer needed.

xnorbi, 2016-01-18:
This is my code:
```
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266mDNS.h>

const char* ssid = "wifi";
const char* password = "";
WiFiServer server(80);
uint16_t timeout=0;

/*------------OTA----------------------------------------*/
  const uint16_t aport = 8266;
  const char* host = "esp8266-ota";
  WiFiServer TelnetServer(aport);
  WiFiClient Telnet;
  WiFiUDP OTA;
/*------------OTA----------------------------------------*/

void html_index(WiFiClient *client);
void OTAhandle();

void setup() {
  Serial.begin(115200);
  Serial.print(F("Connecting to "));
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  uint8_t count=0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    count++;
    if(count>100) {
      delay(1000);
      ESP.restart();
    }
  }
  Serial.println(F("\nWiFi connected"));
  server.begin();     // Start the server
  Serial.print(F("Server started. Use this URL to connect: http://"));
  Serial.println(WiFi.localIP());
  
/*------------OTA----------------------------------------*/
  MDNS.begin(host);
  MDNS.addService("arduino", "tcp", aport);
  OTA.begin(aport);
  TelnetServer.begin();
  TelnetServer.setNoDelay(true);  
  }
/*------------OTA----------------------------------------*/
void loop() {
  delay(100);
  Serial.printf("loop heap size: %u\n", ESP.getFreeHeap());
  OTAhandle();
  WiFiClient client;
  client = server.available();
  
  if (!client) {
   Serial.printf("client heap size: %u\n", ESP.getFreeHeap());
   return;
  }
  
  while(!client.available())   {
    Serial.print(F("."));    
    delay(1);   
    timeout++;
    if(timeout>500) {Serial.println(F("INFINITE LOOP BREAK!"));  client.flush(); client.stop(); Serial.println(F("BREAK NOW!")); break;}
    }
  String request = client.readStringUntil('\r');  // Read the first line of the request
  if(request.indexOf("GET / HTTP") != -1)  {
    html_index(&client);     // Return the response
    Serial.printf("index heap size: %u\n", ESP.getFreeHeap());
    }
}

void html_index(WiFiClient *client){
  String s="abcd";
  client->println(s);
}

void OTAhandle(){
  
  if (OTA.parsePacket()) {
    IPAddress remote = OTA.remoteIP();
    int cmd  = OTA.parseInt();
    int port = OTA.parseInt();
    int size   = OTA.parseInt();

    Serial.print("Update Start: ip:");
    Serial.print(remote);
    Serial.printf(", port:%d, size:%d\n", port, size);
    uint32_t startTime = millis();

    WiFiUDP::stopAll();

    if(!Update.begin(size)){
      Serial.println("Update Begin Error");
      return;
    }

    WiFiClient client;
    if (client.connect(remote, port)) {

      uint32_t written;
      while(!Update.isFinished()){
        written = Update.write(client);
        if(written > 0) client.print(written, DEC);
      }
      Serial.setDebugOutput(false);

      if(Update.end()){
        client.println("OK");
        Serial.printf("Update Success: %u\nRebooting...\n", millis() - startTime);
        ESP.restart();
      } else {
        Update.printError(client);
        Update.printError(Serial);
      }
    } else {
      Serial.printf("Connect Failed: %u\n", millis() - startTime);
    }
  }
}
```
With this output into serial console:
*client heap size: 31552*
*loop heap size: 31464*
*client heap size: 31464*
*client heap size: 31496*
*loop heap size: 31496*
*client heap size: 31496*
*loop heap size: 31408*
*client heap size: 31408*
*loop heap size: 31440*
*loop heap size: 31440*
*loop heap size: 31352*
*client heap size: 31352*
*loop heap size: 31384*
*loop heap size: 31384*

When I put this line into the loop the heap leaking is solved:
TelnetServer.available();

Now the problem is solved! Thank You!

tuxedo0801, 2016-01-18:
I don't see where adding this line prevents a memory leak?!

Isn't it just checking the available bytes that can be read?

Sounds a bit strange to me.

xnorbi, 2016-01-18:
Here:
```
void loop() {
  delay(100);
  Serial.printf("loop heap size: %u\n", ESP.getFreeHeap());
  OTAhandle();
  TelnetServer.available();
  WiFiClient client;
  client = server.available();
```

Check it on your own board. Arduino 1.6.5, ESP8266 Arduino Core: 1.6.5-947-g39819f0

asetyde, 2016-01-18:
Sorry but it’s a good idea create every time a object wifi client ? is the same to reboot every time .. you will not resolve if there is a memory leak but it has you reboot every time .. i ‘ve with mysql connect same problem .. i’m asking for .. but at now every 150 min i reboot and board work

Duality4Y, 2016-01-18:
that object is thrown away when the function loop returns.
there for it doesn't matter, it only impacts performance as you have to create space on the stack for that object every time you enter the loop function.

me-no-dev, 2016-01-20:
first I should point out that this is an old example.
then you are correct, not checking for available "telnet" client will mean that any new client will hang (and your ArduinoIDE is calling that port to check if the ESP is alive) which will lead to a leak.

Please update your core if not updated, and check the BasicOTA example for ArduinoOTA library.

closed 2016-02-29 by igrr