esp8266/Arduino #1462: Many problems with WiFi libraries

odilonafonso, 2016-01-18:
I'm used to use ESP modules to the Arduino libraries and AT firmware.
Recently I decided to use the WiFi ESP8266 libraries and am having a lot of problems .
I have seen many comments about the version 2.1.0.rc1 - I'm using .
I can not use DEBUG , can not reconnect to a wifi ... I am a week ago and not evolution.
I'd better uninstall it and start over with a previous release ?

Links2004, 2016-01-18:
the rc1 uses SDK 1.5.0 which is instabil, try latest git (SDK 1.5.1) it perform much better.

odilonafonso, 2016-01-18:
Hi Links2004. I don't know how to do this. Where do I get SDK 1.5.1 ?
Thank you

Links2004, 2016-01-18:
@igrr will create a rc2 soon (using SDK 1.5.1), check for updates in the next days.

if you want to use the latest git version:
https://github.com/esp8266/Arduino#using-git-version-

odilonafonso, 2016-01-18:
Thank you Markus,

The instructions to use the ESP8266 are:

Instructions

   - Start Arduino and open Preferences window.
   - Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json
   into *Additional Board Manager URLs* field. You can add multiple URLs,
   separating them with commas.
   - Open Boards Manager from Tools > Board menu and find *esp8266*
   platform.
   - Select the version you need from a drop-down box.
   - Click *install* button.
   - Don't forget to select your ESP8266 board from Tools > Board menu
   after installation.

You may optionally use *staging* boards manager package link:
http://arduino.esp8266.com/staging/package_esp8266com_index.json. This may
contain some new features, but at the same time, some things might be
broken.
Using git version

This is the suggested installation method for contributors and library
developers.
Prerequisites

   - Arduino 1.6.5 (or newer, if you know what you are doing)
   - git
   - python 2.7
   - terminal, console, or command prompt (depending on you OS)
   - Internet connection

Instructions

   - Open the console and go to Arduino directory. This can be either your
   *sketchbook* directory (usually <Documents>/Arduino), or the directory
   of Arduino application itself, the choice is up to you.
   -

   Clone this repository into hardware/esp8266com/esp8266 directory.
   Alternatively, clone it elsewhere and create a symlink, if your OS supports
   them.

   cd hardware
   mkdir esp8266comcd esp8266com
   git clone https://github.com/esp8266/Arduino.git esp8266

   You should end up with the following directory structure:

   Arduino|
   --- hardware
       |
       --- esp8266com
           |
           --- esp8266
               |
               --- bootloaders
               --- cores
               --- doc
               --- libraries
               --- package
               --- tests
               --- tools
               --- variants
               --- platform.txt
               --- programmers.txt
               --- README.md
               --- boards.txt
               --- LICENSE

   -

   Download binary tools

   cd esp8266/tools
   python get.py

Well, i have used the first method,

   - Start Arduino and open Preferences window.
   - Enter http://arduino.esp8266.com/stable/package_esp8266com_index.json
   into *Additional Board Manager URLs* field.

Now I have three hardware folders:

/opt/arduino-1.6.5-r5/hardware
/home/Arduino/hardware (empty)
/home/odilon/.arduino15/packages/esp8266

The structure I got when use the Arduino IDE board's manager is completely
different from the structure of the git !!

I have into packages a esp8266 folder with 2 sub-folders:

hardware and tools

into hardware I found another folder esp8266.... esp8266/2.1.0-rc1 and into
2.1.0-rc1:

boards.txt   cores  libraries  package       programmers.txt  tests
variants
bootloaders  doc    LICENSE    platform.txt  README.md        tools

inside the tools of 2.1.0-rc1 there is the get.py file... is that must I
run with python ?

I am completely confused....

What is the binary tools suggested to get with python - get.py

If I change the structure of the ESP8266 files, the Arduino IDE will found
the things ? The libraries, includes, source files ?

May be is better I delete all folders and reinstall everything...

Thank you

*Odilon Cenamo*
*Vivo (11)95636-1956*

*Skype (11)3042-1956Nextel (11)7720-4140 Id 30*69343*

Links2004, 2016-01-18:
remove all existing ESP stuff.
then:

### Using git version [image]

- Install Arduino 1.6.7
- Go to Arduino directory
- Clone this repository into hardware/esp8266com/esp8266 directory (or clone it elsewhere and create a symlink)
```bash
cd hardware
mkdir esp8266com
cd esp8266com
git clone https://github.com/esp8266/Arduino.git esp8266
```
- Download binary tools (you need Python 2.7)
```bash
cd esp8266/tools
python get.py
```
- Restart Arduino

yes you need python, the python script downloads the latest compilers and esptool

odilonafonso, 2016-01-18:
Ok. Must I remove the .arduino15 folder too ?

odilonafonso, 2016-01-18:
Ok, forget the last comment...
I install ESP8266 into /opt/arduino-1.6.7 folder
Into this folder I make the clone and run python.
But now, when I compile the sketch, i got:

Sketch uses 243,396 bytes (56%) of program storage space. Maximum is 434,160 bytes.
Global variables use 35,192 bytes (42%) of dynamic memory, leaving 46,728 bytes for local variables. Maximum is 81,920 bytes.
/home/odilon/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/tools/esptool/esptool -vv -cd ck -cb 115200 -cp /dev/ttyUSB1 -ca 0x00000 -cf /tmp/build79ba95d60ea1a04c49bc5589b2922b96.tmp/WiFiAccessPoint.ino.bin 
java.io.IOException: Cannot run program "/home/odilon/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/tools/esptool/esptool": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:123)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:167)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2430)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more

I think I have to change this:
Cannot run program "/home/odilon/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/tools/esptool/esptool": error=2, No such file or directory
To another local.

Links2004, 2016-01-18:
remove the rc1.
delete /home/odilon/.arduino15/packages/esp8266
and restart the IDE.

odilonafonso, 2016-01-18:
​OK, now it is burning...
but I'm still unable to connect to my wifi.
Well, now I am with the 1.6.7 Arduino IDE release and the ESP8266 last
version, right ?
​I will return to testing to see if I can some progress
Many, many thanks for your help Markus !!​

odilonafonso, 2016-01-18:
Markus,
Do you know WiFiManager ?
https://github.com/tzapu/WiFiManager
I can connect with the AutoConnect example first time, but only first
time...

*WM:
*WM: AutoConnect
*WM: Reading SSID
*WM: SSID:
*WM: PONESA-3G
*WM: Reading Password
*WM: Password: meiadeseda
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.2.103
connected...yeey :)

*(hear I do a hardware reset... ESP8266 RST to GND) or CH_PD floating for a
while*

*WM: AutoConnect
*WM: Reading SSID
*WM: SSID:
*WM: PONESA-3G
*WM: Reading Password
*WM: Password: meiadeseda
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 4
*WM:
*WM: Configuring access point...
*WM: AutoConnectAP
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

*(hear I do another hard reset)*

*WM: AutoConnect
*WM: Reading SSID
*WM: SSID:
*WM: PONESA-3G
*WM: Reading Password
*WM: Password: meiadeseda
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 0
*WM:
*WM: Configuring access point...
*WM: AutoConnectAP
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started

And now, always I do hardware reset, it does the same thing. Get connection
result 0 (WL_IDLE_STATUS)

I'm receiving this for a week... and I don't know anymore what to do.

I thought that with these changes things would get better

Well, thanks for the help so far

odilonafonso, 2016-01-18:
I noted two things:
1. When I use WiFI.mode (WIFI_STA) the occurrence of errors is greater than when I do not use.
2. When I NOT use WiFi.begin (ssid, psk) and use WiFi.begin () without arguments (using saved credentials) decreases the occurrence of errors.
This snippet is working relatively well:

  int wifiMode = WiFi.getMode();
  if ( !(wifiMode == WIFI_STA || wifiMode == WIFI_AP_STA) )
    WiFi.mode(WIFI_STA);
  Serial.print("\nWiFi.getMode():");Serial.println(WiFi.getMode());

  WiFi.printDiag(Serial);

  if ( !(WiFi.SSID().equals("")) ){
    Serial.println("Trying to connect with saved credentials...");
    WiFi.begin();
  }
  else {
    //provide eeprom read credentials...
    Serial.println("Connecting with eeprom credentials..");
    wifiSSID = eepromRead("ssid"); //actually eepromRead = WiFi.SSID();
    wifiPASS = eepromRead("psk");  //actually eepromRead = Wifi.psk();
    Serial.print("ssid ");Serial.println(wifiSSID);
    Serial.print("psk ");Serial.println(wifiPASS);
    Serial.print("Connecting to "); Serial.println(wifiSSID);
    WiFi.begin(wifiSSID.c_str(), wifiPASS.c_str());

closed 2016-01-20 by odilonafonso

MrSaeeed, 2016-05-04:
hello..
I used the git version to add esp8266 to arduino software...
but I have a problem with downloading the binary tools with python 2.7...
when i run the 'get.py' script with python, this error appear:
(Traceback (most recent call last):
  File "C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\tools\get.py", line 114, in <module>
    mkdir_p(dist_dir)
  File "C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\tools\get.py", line 35, in mkdir_p
    os.makedirs(path)
  File "C:\Python27\lib\os.py", line 157, in makedirs
    mkdir(name, mode)
WindowsError: [Error 5] Access is denied: 'dist/')

what does it mean? and what should i do? 
please help

MrSaeeed, 2016-05-05:
no one can give me a hint??  :-(

rousnay, 2017-01-15:
@MrSaeeed , Run the CMD "As a Administrator", Thanks