esp8266/Arduino #1465: SPIFFS erase or overwrite file

aug2uag, 2016-01-18:
I have a file `f.txt` that holds configurations, and I'd like to reset them based on user actions.

The docs don't have a file delete; and I noticed the following fails to erase the data

```
File f = SPIFFS.open("/f.txt", "w+");
// clear configs
f.println("");
ESP.restart();
```

Would this operation be appropriate for the intended outcome or another solution? Thanks Ivan and the team for the excellent resources you've developed.

Links2004, 2016-01-18:
use this one ;)
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h#L119

aug2uag, 2016-01-18:
Hey thanks!

closed 2016-01-18 by aug2uag

romanseidl, 2017-06-18:
```
bool remove(const char* path);
bool remove(const String& path);
```
now on https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h#L120 as lines move...