esp8266/Arduino #1466: Arduino IDE 1.6.7 can not found include files

odilonafonso, 2016-01-19:
Yesterday, after installing the 1.6.7 release of the Arduino IDE, I could run for a while.
For this, I went to the installation directory and directly executed the "arduino" file.
After however, run the shell "install.sh", to generate a desktop shortcut, the compiler no longer find the include files - "ESP8266WiFi.h" and others.
Can anyone tell me what I did wrong?
What did the install.sh shell may have changed so that things no longer work?

odilonafonso, 2016-01-19:
Well, I tried again, something happened - returned to work.
I am seriously considering to downgrade to 1.6.5 .....

asetyde, 2016-01-19:
There is no problem with 1.6.7 , please clean all data  and reinstall it

odilonafonso, 2016-01-19:
Hi asetyde,
Can you compile the .ino attached, please ?
[testfun.zip](https://github.com/esp8266/Arduino/files/95921/testfun.zip)

asetyde, 2016-01-19:
i’m not sure you can do this …

asetyde, 2016-01-19:
i'm not ve this problem because i use c old style common write : 

[image]

odilonafonso, 2016-01-19:
Ok, thanks..

itsjustvenky, 2016-01-20:
@odilonafonso  Things have changed recently. So if you have function and it needs to be declared at the top.

void test();

and then you can refer it any where in the code, else the called function should be on top of calling function.

odilonafonso, 2016-01-20:
Hi itsjustvenky,
Try this:
```
#include <ESP8266WiFi.h>
#define LED 2
void test1(){
  pinMode(LED, OUTPUT);
  test2(LED);
  test3(LED, LOW);
}
void test2(int led){
  digitalWrite(led, HIGH);
  delay(500);
}
bool test3(int led, int what){
  digitalWrite(led, what);
  delay(500);
  return true;
}
void setup() {
  // put your setup code here, to run once:
  test1();
}
void loop() {
  // put your main code here, to run repeatedly:
  test2(LED);
  test3(LED, LOW);
}
```

odilonafonso, 2016-01-20:
And try this:
```
#include <ESP8266WiFi.h>
#define LED 2
void test2(int);
bool test3(int, int);
void test1(){
  pinMode(LED, OUTPUT);
  test2(LED);
  test3(LED, LOW);
}
void test2(int led){
  digitalWrite(led, HIGH);
  delay(500);
}
bool test3(int led, int what){
  digitalWrite(led, what);
  delay(500);
  return true;
}
void setup() {
  // put your setup code here, to run once:
  test1();
}
void loop() {
  // put your main code here, to run repeatedly:
  test2(LED);
  test3(LED, LOW);
}
```
The problem is with the pre-processor...
The "mens" are working to solution.
https://github.com/arduino/arduino-builder/issues/68

closed 2016-01-20 by odilonafonso