esp8266/Arduino #1468: How to broadcasting AP ssid

odilonafonso, 2016-01-19:
How should I use WiFi.softAP () so that the ssid is visible for connection ?
I examined the prototype function in ESP8266WiFiAP.h:

`bool SoftAP (const char * ssid, const char * passphrase = NULL, int channel = 1, ssid_hidden int = 0);`

But not figured out how the sample applications configure broadcast, because I can not see the utilization of the ssid_hidden argument.
The example WiFiAccessPoint from ESP8266WiFi Arduino IDE uses the call:

`WiFi.softAP (ssid, password);`

Without explicitly add ssid_hidden argument.

I figured that because the function take over as the default value 0, by calling the function with fewer arguments, the default was used (ssid_hiden = 0) to broadcast.
However, when I used to call:

`WiFi.sofAT (my_ssid, my_psk)`

that returned true, the AP has been created and yet is not visible connection.

Does anyone know how it works?

thank you

Links2004, 2016-01-19:
with ```WiFi.SoftAP``` it will be broadcast, only when ```ssid_hidden``` is 1 it will hide it (no broadcast).
using:
```
WiFi.SoftAP("APNAME","PASS1234567");
```
will result in a visible AP.

odilonafonso, 2016-01-19:
Ok. The problem must be on my pc, it does not show AP recently created. After a restart it goes to show.
Thank you.

closed 2016-01-19 by odilonafonso