esp8266/Arduino #1469: Flash write failed when updating with HttpUpdate

eyp, 2016-01-19:
Using 2.1.0-rc1 on a esp8266 E-12 I get this error when updating via httpUpdate:
I've activated the debug for HTTPClient + HTTPUpdate + Updater and after doing this call...

t_httpUpdate_return ret = ESPhttpUpdate.update("<hidden>",80,"/nuevofirm.bin","0.1");

I've got this debug traces:

[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 321120
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 700416
[httpUpdate]  - current Sketch Size: 344096
[httpUpdate]  - current version: 0.1
[httpUpdate] runUpdate flash...
sleep disable
[begin] roundedSize:       0x0004F000 (323584)
[begin] updateEndAddress:  0x00100000 (1048576)
[begin] currentSketchSize: 0x00055000 (348160)
[begin] _startAddress:     0x000B1000 (724992)
[begin] _currentAddress:   0x000B1000 (724992)
[begin] _size:             0x0004E660 (321120)
ERROR[1]: Flash Write Failed
[httpUpdate] Update.writeStream failed! (ERROR[1]: Flash Write Failed)
[httpUpdate] Update failed

The return code is 0.

Any ideas of what is causing this error?

Links2004, 2016-01-19:
the write to flash has failed.
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Updater.cpp#L205-L214

most times it comes from a instabile power supply.
try to change it or add 100nF and one elko with 47uF or more.

eyp, 2016-01-19:
What a pity, I already have the 47uF :(

smallworld-network-wupeng, 2016-04-07:
Same problem . Do you solved ？

eyp, 2016-04-07:
No, I don't know if it's a problem of the chip or what. I have two circuits, one accepts the OTAs the other not, so I gave up on this problem.

smallworld-network-wupeng, 2016-04-08:
I solved it !
It is maybe esp8266 2.1 library's bug.  I update to esp8266 library 2.10 , I find can not upload bin file by FTDI232 in MacOS 10.11.4 . 
`esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A400f0Tx
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A400f0Tx at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed`

I use other USB-TO-Serial PL2303, still can not upload . so I google , find this : http://arduino.stackexchange.com/questions/20219/upload-with-esptool-fails-with-espcomm-send-command-cant-receive-slip-payload
I try it . It is work it . But the httpUpdate can not work . Same problem like @eyp . 

Today , I install esp8266 library 2.0 , and use FTDI232 to upload bin file , the httpUpdate worked again ! There is esptool 0.4.6 in esp8266 library 2.0 , but esptool 0.4.8 in esp8266 library 2.1.

murix, 2016-08-17:
I have some board with ota work fine and another lot with ota failing.
HTTP_UPDATE_FAILED Update error: ERROR[1]: Flash Write Failed

eyp, 2016-08-17:
We have the same problem... May be some 'clones' are no so reliable as others.

murix, 2016-08-17:
I think about generic cloned flash id maybe a problem, but the problem can be software only. 
I try print WiFi.RSSI without Wifi connected, and the result are 0xfffffff, when in String(WiFi.RSSI()) a buffer overflow cause lock spi flash and flash write fail. I need verify and paste a small sketch to reproduce de OTA bug. Corrupted buffers in WiFi stack seems the cause of OTA flash write failing

me-no-dev, 2016-08-18:
I just merged [this](https://github.com/esp8266/Arduino/commit/b250381d80c927a72af58c31182b0ea0cad44334) which should help your problems. Please check and confirm

murix, 2016-08-20:
@me-no-dev  I tryed the patch (it's related to sketch size only) but still failing, so my firmware is about 312k and I have about 700k free, that is no problem. Now I'm checking md5 too. Some units are first flashed with espflasher.exe (from nodemcu), git version of httpupdate is sending sketch md5 to php server, and the md5 is failing. During troubleshoot I found the reason of md5 mismatch, that is SPI mode configuration. The espflasher.exe by default use DIO mode, and nodemcu/esp12 build by arduino use QIO mode. So flashing a binary created with 'Sketch export compiled binary' into nodemcu/esp12 board will be corrupt the firmware image, and make use of incorrect **SPI flash settings**, that **can lead to flash write fail.** I fixed the md5 mismatch by setting the espflasher with SPI mode = QIO.

bbourdel, 2017-02-08:
Hello @murix, Do you have some success to fix this  (ERROR[1]: Flash Write Failed) ? 
I got the same and I like to have some idea to quickly solved it.
You'll be greatfull ! ;-)

murix, 2017-02-09:
Hi @bbourdel, the problem seems the low quality flash.
Some units fail in serial programming 921600bps but works with 115kbps or
230kbps.
Maybe the update routine is doing the job too fast or don't care about low
quality flash parameters.

Best regards,
Murilo Rebelo Pontes

devyte, 2017-10-18:
The last comment is most likely correct. Some chinese flash chips don't work well at high speed. Also, flash write time can change over time as the flash ages/wears.
Closing.

closed 2017-10-18 by devyte