esp8266/Arduino #1470: 2.1.0-rc1, can't compile with Debug Level All

eyp, 2016-01-19:
When compiling with Debug Level All I've got this errors:

In file included from /Users/edu/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp:26:0:
/Users/edu/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp: In member function 'wl_status_t ESP8266WiFiMulti::run()':
/Users/edu/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.h:37:66: error: cannot pass objects of non-trivially-copyable type 'class String' through '...'
 #define DEBUG_WIFI_MULTI(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )

                                                                  ^
/Users/edu/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc1/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp:130:25: note: in expansion of macro 'DEBUG_WIFI_MULTI'
                         DEBUG_WIFI_MULTI("[WIFI] SSID: %s\n", WiFi.SSID());

                         ^
I have to choose another level in order to get some debug traces.

Links2004, 2016-01-19:
Duplicate of #1369 and #1448
its fixed in latest git, (rc2 will be released soon).

eyp, 2016-01-19:
OK, I'll close the issue.

closed 2016-01-19 by eyp