esp8266/Arduino #1471: warning: espcomm_sync failed error: espcomm_open failed

kapilpawar, 2016-01-20:


krzychb, 2016-01-20:
Hi kapilpawar,

This message indicates possible issue with:
•	h/w connection between your computer and ESP module
•	selection of board / board reset method in Boards Manager

It may be either one or both above.

The answer depends on the module type and external serial adapter type (if you use one).
There is a dozen module types, three generic adapters types and couple of methods to connect them.

Therefore in order to help you troubleshoot this issue, you need to post some additional details:

1.	ESP board type - https://github.com/esp8266/Arduino/blob/master/doc/boards.md  
2.	Serial adapter type (if you use one) - https://github.com/esp8266/Arduino/blob/master/doc/boards.md#serial-adapter
3.	How you connect your ESP board with adapter - https://github.com/esp8266/Arduino/blob/master/doc/boards.md#minimal-hardware-setup-for-bootloading-and-usage 
4.	Screenshot of configuration for loading you selected in Boards Manager of Arduino IDE

Krzysztof

kapilpawar, 2016-01-20:
Hi krzychb,
I am using ESP1 module,.

I am trying to connect ESP8266 directly to the ARDUINO UNO...
so please give resolution to resolve this problem.

thank you krzychb..........

krzychb, 2016-01-20:
kapilpawar,

I am afraid this not the best place to troubleshoot this issue.

The purpose of this repository is to let you write sketches using Arduino IDE specifically for ESP8266. 

To obtain firsthand experience regarding direct connection of ESP8266 to the ARDUINO UNO I would check  http://www.esp8266.com/viewforum.php?f=8&sid=9ea483e0556ae73240806f83061efbc0 that deals specifically with topics, source code that relate to the Arduino Platform.

Krzysztof

igrr, 2016-01-20:
I think what kapilpawar means is that he uses Arduino UNO as a serial
adapter to flash the ESP. Am I right?

kapilpawar, 2016-01-20:
exactly .........

krzychb, 2016-01-20:
kapilpawar,

The first two questions are now clarified thanks to igrr :smile:
Please answer remaining two.
For third question please provide schematic

Krzysztof

kapilpawar, 2016-01-20:
Hi krzychb,

my connection is .........

ARDUINO UNO ...................> ESP8266
TX...........................................>RX
RX.........................................>TX
VCC......................................>3.3V
GND.....................................>GND
CH_PD ................................>3.3V
RST......................................>3.3V
GPIO 2 ..................................>3.3V

and
laptop to arduino through USB cable.
please help me i have tried different combination of the pins, but it didnt work....

Duality4Y, 2016-01-20:
you know that arduino uno is 5V and that connecting it's TX to your esp8266 RX will blowup the RX on your esp8266 ?

kapilpawar, 2016-01-20:
hi Duality4Y

i tried with 3.3v that is also given in the same arduino board..
please help,,

kapilpawar, 2016-01-20:
while uploading it is showing...

Sketch uses 198,612 bytes (45%) of program storage space. Maximum is 434,160 bytes.
Global variables use 33,014 bytes (40%) of dynamic memory, leaving 48,906 bytes for local variables. Maximum is 81,920 bytes.

warning: espcomm_sync failed
error: espcomm_open failed

Duality4Y, 2016-01-20:
connecting to the 3.3v pin on the arduino doesn't matter it's gpio will still be 5V TTL.
also have you tried removing the avr? because it's in the way of the communication if it's connected to the same pins :)

Duality4Y, 2016-01-20:
have a look at how it's connected here:
http://www.stefanocottafavi.com/discover-the-esp8266/

drmpf, 2016-01-20:
Or just using a diode as in
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html

supersjimmie, 2016-01-21:
Normally, if you want communication between arduino and ESP, you need RX to TX and TX to RX, like you say:
TX...........................................>RX
RX.........................................>TX

But in this case, you want to "abuse" the serial from the arduino to pass-through between your laptop and the ESP. Then you need to connect Arduino RX to ESP RX and Arduino TX to ESP TX (that is pass-through). By doing that, the ESP will be connected to the laptop the same way as the Arduino.
So you need:
TX...........................................>TX
RX.........................................>RX

Also make sure the ESP is in flash mode.

Duality4Y, 2016-01-21:
oh yea good point you need to connect it that way indeed :)

mammenj, 2016-05-02:
I have spent lot of time on this error. What I found is that FTTD usb adapter was not providing enough power to the board to flash the sketch. As soon as I provided a stable power, it works all the time. So its good idea to have stable power if you are still facing the issue,

jyotirajsharma, 2017-01-03:
I am facing the exact issue and below is my configuration. Wil changing to ext. power supply than from Aruduino solve this problem ?
ARDUINO UNO ...................> ESP8266
TX...........................................>RX
RX.........................................>TX
OR
TX...........................................>TX
RX.........................................>RX
VCC......................................>3.3V
GND.....................................>GND
CH_PD ................................>3.3V
RST......................................>3.3V

Arduino: 1.6.12 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: FC instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: FC instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 08 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: FC instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

devyte, 2017-10-18:
This error is documented. Closing.

closed 2017-10-18 by devyte

AliHassan4198, 2018-10-10:
I have the same issue...  I Just Close the Serial Port Monitor of Arduino