esp8266/Arduino #1474: Flash compiled binary using raspberry pi

jdkanani, 2016-01-20:
While using native ESP SDK, build process generates two bin files, 0x00000 and 0x40000. But  `export compiled binary` option generates only one. I flashed it using following command with raspberry pi:

```
esptool-ck/esptool -vv -cd none -cb 115200 -cp /dev/ttyAMA0 -ca 0x00000 -cf main.ino.generic.bin
```

It's working fine. But now, pulling GPIO0 down crashes raspberry pi. I don't know what went wrong. Am I missing something?

Duality4Y, 2016-01-20:
how are you pulling GPIO0 down with the pi?
if so, maybe GPIO0 is configured in such a state that pulling it down with the pi,
causes a short. that might crash the pi :) is the only thing I can think of.

jdkanani, 2016-01-20:
I am connecting GPIO0 to ground as I want to flash it again with new code. Yeah, I did continuity check and it shorts VCC, GND and GPIO0.

jdkanani, 2016-01-20:
Just checked with another chip + native binary, it works with same circuit connections. Any idea?

jdkanani, 2016-01-21:
To flash the chip, I made following connections:

ESP | RaspberryPI
------ | -----------------
VCC, CH_PD | 3.3v
GND, GPIO15 | GND
TX | RX
RX | TX
GPIO0 | GND (To flash new firmware)

GPIO12 is connected to LED to check blink.

While floating `GPIO0`, Blink example works fine. Grounding `GPIO0` directly to flash new firmware causes a hang in RaspberryPI.

devyte, 2017-10-18:
Closing per #3655 .

closed 2017-10-18 by devyte