esp8266/Arduino #1475: opendir failed on root

mytrain, 2016-01-20:
With a 1.6.x version I created a file system with two (virtual) folders f1 and f2 each of them containing files

in 1.6.0 invoking opendir("") returned all the files of the FS.
in 2.1.0 opendir("") returns nothing.
I've tried opendir("/") but returns also nothing
opendir("/f1") returns nothing
but opendir("f1") returns correctly all the files of f1 folder.

Question: Knowing that I do not want to regenerate the File System how can I access the full list of file without know first level folder names ?

Thanks

me-no-dev, 2016-01-21:
make sure your folders and files start with slash when you are creating the filesystem. From then on you will not have any issues.
You can use the plugin for ArduinoIDE to auto generate the image for you as well :)
Also always save files with starting slash!

mytrain, 2016-01-22:
Unfortunately I'm not using IDE for uploading files, my ESP is part of a bigger project and it download information files from a server. 
Does anybody know where, in the code, the regression come from ?

me-no-dev, 2016-01-22:
opendir calles isSpiffsFilenameValid(fileName) which checks if the fileName length is greater than 0 and less than the max filename length. In your case "" has 0 length

mytrain, 2016-01-25:
Now its working.

closed 2016-01-25 by mytrain