esp8266/Arduino #1477: Confused about wdtFeed, esp_yield and yield

supersjimmie, 2016-01-21:
I have some trouble with the latest SoftwareSerial library:
https://github.com/esp8266/Arduino/issues/1426
and I am wondering if this could be fixed with either 
```    
    ESP.wdtFeed();
    esp_yield();
    yield();
```
But what is the difference and how could any of them help to avoid WDT Resets?

supersjimmie, 2016-01-21:
Never mind, if I look closer to where the softwareserial crashes, I don't think a yield or wdtfeed will help.
Closing this.

closed 2016-01-21 by supersjimmie

Defozo, 2017-06-20:
I'd like to know what is the difference between these functions?

gunt2551, 2025-02-24:
I found the comment on line 502 in `Arduino/libraries/ESP8266WiFiMesh/src/EspnowMeshBackend.cpp`
> `ESP.wdtFeed();`  // Prevents WDT reset in case we receive a lot of transmissions without break.

`esp_yield();` and `yield();` is decribed on line 139 in `Arduino/cores/esp8266/core_esp8266_main.cpp`

> // Replacement for delay(0). In CONT, same as yield(). Whereas yield() panics
// in SYS, esp_yield() is safe to call and only schedules CONT. Use yield()
// whereever only called from CONT, use esp_yield() if code is called from SYS
// or both CONT and SYS.

From what i understand, CONT refers to the main function[void loop()] of Arduino IDE structure, which only `yield()` is used for, whereas `esp_yield()` is used inside interrupt function.