esp8266/Arduino #1479: Wifi connect problems

donnib, 2016-01-22:
Hi,
I have a sensor where i take measurement -> deep_sleep (without wifi) -> take measurement -> deep_sleep (without wifi), i do this 10 times then i detect that 10 times past and i deep_sleep with 1 and turn the WiFi on so i can send my data to Cloud. When i wake up from deep_sleep i ran this method 

```
void LaunchWifi()
{
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  WiFi.config(ip,gateway,subnet);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}
```

which many times ended up in the serial console with .................................... meaning it was trying to connect but it went nowhere. This is unacceptable for me because the sensor deadlocks and will drain my battery within no time so i made a correction to the method and made this 

```
void LaunchWifi()
{
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  WiFi.config(ip,gateway,subnet);
  int i = 0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    ESP.wdtFeed();
    if(i>40)
      ESP.reset();
    i++;
    WiFi.begin(ssid, password);
    WiFi.config(ip,gateway,subnet);
    }
```

and now i detect the reset and try to connect again to wifi and send my data and this works 99% if not 100% but i don't like this solution. I am hoping there is a better solution that does not require a ESP.reset().

Any ideas ?

WereCatf, 2016-01-22:
What flag do you supply to deepSleep to wake WiFi up? I think the issue is related to that. The documentation is kind of lacking, but AFAIK you should use WAKE_RFCAL in your case.

donnib, 2016-01-22:
@WereCatf i use WAKE_NO_RFCAL, i can try with WAKE_RFCAL, i understood that i don't have to do RFCAL everytime but i must have misunderstood something.

donnib, 2016-01-22:
@WereCatf i tried WAKE_RFCAL and i still get very often that the ESP can't connect to the WiFi.

donnib, 2016-01-25:
Any other ideas what i can do to make this stable ?

WereCatf, 2016-01-25:
I'm sorry, I completely forgot about this topic. Quite frankly, I'm at a bit of loss myself. You don't have a WiFi.mode(WIFI_STA); anywhere in the snippets you posted here, have you tried adding that before WiFi.begin()?

donnib, 2016-01-25:
@WereCatf , no i don't have it, i will try to add it and get back with the verdict later today

donnib, 2016-01-25:
@WereCatf so i added the line before begin and no change, the unit reboots few times (takes sensor measurement) and turns wifi on and sends data few time then it stops and goes into the loop ............. where it never connects again. Frustrating......

donnib, 2016-01-25:
I have added ````ESP.eraseConfig()```` and that seems to have made it working for at least 1h now. I am thinking that ````ESP.eraseConfig()```` will delete the RF data which might cause something like this ? I know @igrr mentioned something about this RF cal in https://github.com/esp8266/Arduino/issues/1094. I dunno what to say, i am afraid that calling ````ESP.eraseConfig()```` might not be a good thing to call everytime (before) i connect to Wifi. Any thoughts ?

WereCatf, 2016-01-25:
I didn't even know about that. That's interesting. Technically, ESP.eraseConfig() does wear the flash down every single boot and as far as I know there is no wear-levelling on it, but you'd probably be looking at thousands of boots before it became an issue. That still makes me wonder... there's a function for enabling/disabling WiFi-persistence where it saves credentials and stuff on flash and it's enabled by default (personally I always turn persistence off since my sketch has the WiFi-credentials hardcoded anyways and thus writing them in flash every single boot is pointless), but I don't know if this persistence also includes RF-calibration or not -- the question is, does turning WiFi-persistence off also turn RF-calibration persistence off? If not, is there any other way of disabling it?

I don't have any idea about the answer, to be honest.

donnib, 2016-01-25:
@WereCatf how do you disable the persistence ?

WereCatf, 2016-01-25:
WiFi.persistent(false);

chaeplin, 2016-01-25:
@donnib Do you use flash to your data ?

igrr, 2016-01-25:
There is a parameter which tells SDK whether to store RF calibration if flash:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_phy.c#L242

chaeplin, 2016-01-26:
@donnib I ran a test like yours (deepsleep/3sec/nowifi/WAKE_RF_DEFAULT * 10 ---> report --> repaet). With 457 reports/4 hours, no error(reset) occurred. 
Now I am running a test with WAKE_RFCAL.

https://gist.github.com/chaeplin/e78ecaa11ee05c62865c

tzapu, 2016-02-04:
i have seen a similar behavior when using `WiFi.begin(ssid,password);` AND flash size set to wrong size. (nodemcu 1.0 set to 512kB)
in that configuration, it would only work with WiFi.begin() after the intial set of user and pass, or if setting flash size to the proper 4MB... quite weird..

sunepedersen, 2016-03-05:
igrr, how should this parameter be set properly?
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_phy.c#L242

objarni, 2016-12-21:
@igrr what is proposed/documented way to solve these WiFi stability issues today, december 2016?

Erezmu, 2016-12-21:
What should we do with the core_esp8266_phy.c file in the link above?

universam1, 2017-01-26:
@igrr Could you please give a hint how to set
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_phy.c#L241
you mentioned, since I can't find a hook to it!
Thank you!

SimonKlausLudwig, 2017-09-06:
I had the same problem. I need to connect to WLAN at the beggining in the setup method.. 

This doesnt work:

```
configSerial();
configLogging();
configDisplay();
configPins();
statusLED.blink(10, 50); //Blinks the LED 10 Time with a delay of 50 (so 1000ms delay total)
wifiClient.connect(DEFAULT_SSID, DEFAULT_PASSWORD); 
```

but this works:
```
configSerial();
configLogging();
configDisplay();
configPins();
wifiClient.connect(DEFAULT_SSID, DEFAULT_PASSWORD); 
statusLED.blink(10, 50);
```

Erezmu, 2017-09-17:
I don’t see the different.
It supposed to be the same.
Do you still have problem?
 
Thanks,
 
 
Erez Mutzafi | www.RadGreen.com | Technical Director | Office +972- | 3-535-99-35Mobile +972-544-735-011

SimonKlausLudwig, 2017-09-19:
In the first code sample the line

`statusLED.blink(10, 50); //Blinks the LED 10 Time with a delay of 50 (so 1000ms delay total)`

gets executed before

`wifiClient.connect(DEFAULT_SSID, DEFAULT_PASSWORD); `

This statusLED.blink command took about 1000ms .. and these 1000 milliseconds are responsible for the error.. if I connect to the WLAN before this statement everything works fine...

I'm using a Lolin V3 Nodemcu. But I will try this with an ESP-12E as soon as I get it!

devyte, 2017-10-18:
There are different ways to wake up and connect to wifi, and the total time depends on how it is done, e.g.: call begin(), begin(ssid), begin(ssid,pass), begin(ssid, pass, bssid), do/don't do rf cal, etc.
I saw a post somewhere with a comparison table, but don't remember where off the top of my head.
In any case, an issue tracker is not the right place to discuss. Closing per #3655 .

closed 2017-10-18 by devyte