esp8266/Arduino #1480: ESP8266WebServer incompatible with VisualMicro

SebiPanther, 2016-01-22:
I'm ruinnig a complexer IDE Setup:
Arduino 1.6.7
Arduino core for ESP8266 WiFi chip Stable version
Visual Studio Community 2015 Version 14.0.24730.00 Update 1
Visual Micro 1601.21.0
I can compile and upload any other Sketch to the esp8266.
But if i create a new empty project and include just the ESP8266WebServer.h file (over Add Library) in a sketch like this:
```/*
 Name:		WebServerTest.ino
 Created:	22.01.2016 10:53:43
 Author:	sebi
*/

// the setup function runs once when you press reset or power the board
#include <ESP8266WebServer.h>
void setup() {

}

// the loop function runs over and over again until power down or reset
void loop() {
  
}

```
I get the following Error Messages:
```Compiling 'WebServerTest' for 'Generic ESP8266 Module'
WebServerTest.ino:In file included from
ESP8266WebServer.h:45:3: error: 'String' does not name a type
:String  filename
ESP8266WebServer.h:46:3: error: 'String' does not name a type
:String  name
ESP8266WebServer.h:47:3: error: 'String' does not name a type
:String  type
ESP8266WebServer.h:In file included from
WebServerTest.ino:from
RequestHandler.h:6:47: error: 'String' has not been declared
:virtual bool canHandle(HTTPMethod method, String uri) { return false; }
RequestHandler.h:7:28: error: 'String' has not been declared
:virtual bool canUpload(String uri) { return false; }
RequestHandler.h:8:77: error: 'String' has not been declared
:virtual bool handle(ESP8266WebServer& server, HTTPMethod requestMethod, String requestUri) { return false; }
RequestHandler.h:9:51: error: 'String' has not been declared
:virtual void upload(ESP8266WebServer& server, String requestUri, HTTPUpload& upload) {}
WebServerTest.ino:In file included from
ESP8266WebServer.h:62:30: error: expected ')' before 'addr
:ESP8266WebServer(IPAddress addr, int port = 80)
ESP8266WebServer.h:78:3: error: 'String' does not name a type
:String uri() { return _currentUri; }
ESP8266WebServer.h:80:3: error: 'WiFiClient' does not name a type
:WiFiClient client() { return _currentClient; }
ESP8266WebServer.h:83:3: error: 'String' does not name a type
:String arg(const char* name);   \\ get request argument value by name
ESP8266WebServer.h:84:3: error: 'String' does not name a type
:String arg(int i);              \\ get request argument value by number
ESP8266WebServer.h:85:3: error: 'String' does not name a type
:String argName(int i);          \\ get request argument name by number
ESP8266WebServer.h:89:3: error: 'String' does not name a type
:String header(const char* name);   \\ get request header value by name
ESP8266WebServer.h:90:3: error: 'String' does not name a type
:String header(int i);              \\ get request header value by number
ESP8266WebServer.h:91:3: error: 'String' does not name a type
:String headerName(int i);          \\ get request header name by number
ESP8266WebServer.h:95:3: error: 'String' does not name a type
:String hostHeader();            \\ get request host header if available or empty String if not
ESP8266WebServer.h:101:62: error: 'String' does not name a type
:void send(int code, const char* content_type = NULL, const String& content = String(""))
ESP8266WebServer.h:101:89: error: ISO C++ forbids declaration of 'content' with no type [-fpermissive]
:void send(int code, const char* content_type = NULL, const String& content = String(""))
ESP8266WebServer.h:102:49: error: 'String' does not name a type
[... 76 lines cut ...]
ESP8266WebServer.h:candidate is
ESP8266WebServer.h:sendHeader(const int&, const int&, bool) <near match>
:void sendHeader(const String& name, const String& value, bool first = false)
ESP8266WebServer.h:no known conversion for argument 2 from 'const char [5]' to 'const int&
ESP8266WebServer.h:120:28: error: no matching function for call to 'ESP8266WebServer::send(int, const int&, const char [1])
:send(200, contentType, "")
ESP8266WebServer.h:candidates are
ESP8266WebServer.h:send(int, const char*, const int&) <near match>
:void send(int code, const char* content_type = NULL, const String& content = String(""))
ESP8266WebServer.h:no known conversion for argument 3 from 'const char [1]' to 'const int&
ESP8266WebServer.h:send(int, char*, const int&) <near match>
:void send(int code, char* content_type, const String& content)
ESP8266WebServer.h:no known conversion for argument 3 from 'const char [1]' to 'const int&
ESP8266WebServer.h:send(int, const int&, const int&) <near match>
:void send(int code, const String& content_type, const String& content)
ESP8266WebServer.h:no known conversion for argument 3 from 'const char [1]' to 'const int&
ESP8266WebServer.h:121:10: error: '_currentClient' was not declared in this scope
:return _currentClient.write(file, HTTP_DOWNLOAD_UNIT_SIZE)
Error compiling project sources
```
I'm Guessing that the ESP8266WebServer.h don't refer the WString.h correctly. But its existing on the path and even if i open the ESP8266WebServer.h and look for the references file of a String theStudio finds it.
So i'm a little bit out of idears. I see that the include-definitions in other libaries locking different, but i can't point to the failure.
So i would love to get some help to get this thing running.

Greetings Sebi

WereCatf, 2016-01-22:
How about including ESP8266WiFi.h before ESP8266WebServer.h? I mean, you're clearly missing it..

SebiPanther, 2016-01-22:
I'm pretty sure that i've already tryed that before posting my problem here. My first attempt was to add all needed libraries and then add the functionality. I've just post a much smaller sketch to avoid confusions and be sure to havn't done any other mistake. But after a reboot it just works with a new Empty Sketch and all needed Libraries added. Thanks for your help - sometimes it just help to turn it off and on again ;)

closed 2016-01-22 by SebiPanther

megamorphf, 2016-02-05:
I had the same issue and as it seems changing the order of libraries helped me to compile with no errors. 
i'm sending you my order of includes incase someone is interested:
```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <DNSServer.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>      //https://github.com/bblanchon/ArduinoJson
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
```

SebiPanther, 2016-02-05:
Thanks for the advise. I've forgot that the order is important in C/C++