esp8266/Arduino #1481: About Quality of ESP (Nodemcu 1.0)

asetyde, 2016-01-22:
[image]

We want to show a difference from SEED(official product-or) and other chinese clone , because some issue can be folded from bad quality

asetyde, 2016-01-22:
left is chinese clone          -  Right is original 

Bad quality of text on board or other strange connector or different power regulator can say a lot .. but not all ..

2 led on nodemcu and on esp chip must be blue
- specific not visible diciture say ESP12D and not ESP12E 

Memory leak issue can be founded , also we can see if original if board on chip has an Ai cloud

asetyde, 2016-01-22:
[image]

RIGHT is ORIGINAL Ai , other is clone .

WereCatf, 2016-01-22:
Are there any functional differences you know of, though? I have 3 clones, namely "LoLin Nodemcu V3" and the one thing I know of is that on official Nodemcu-kits the Flash is supposed to be running in DIO-mode and GPIO9 and GPIO10 should be available for users to use as they wish, but on these clones the Flash is driven in QIO-mode and GPIO9 and GPIO10 are still wired to Flash and cannot be used.

asetyde, 2016-01-22:
we see in the top of chip esp12e and d and QIO .. i want ask to @igrr or @Links2004  if for our library it can be a problem .. one DIO one in QIO is not same thing .. last time grr say to me not use gpio9 but in standard nodemcu it can be used ? it can do a strange issues ?

asetyde, 2016-01-22:
Lolin has power different regulator and it's not good i think .. it's not real nodemcu amica standard version .. i must do my work and it can't work sometimes or someone yes or no

Links2004, 2016-01-22:
DIO will work on both versions and its configured for "NodeMCU 1.0"
https://github.com/esp8266/Arduino/blob/master/boards.txt#L440

but using GPIO9 and GPIO10 on the fake make problems (since flash is connected on the ESP12e).
may some one with a original can test GPIO9 an 10 and give some feedback,
I personally only have bare ESP12 and ESP01 modules here.

WereCatf, 2016-01-22:
asetyde: "Lolin has power different regulator and it's not good i think .. it's not real nodemcu amica standard version .. i must do my work and it can't work sometimes or someone yes or no"

I'm not sure what you mean with "it can't work sometimes or someone yes or no", but, I at least, haven't had any issues with power-regulator or any actual code running on the chip aside from the fact that I can't use GPIO9/10.

The differences I have noticed are: 1) The clone uses CH340G as its USB-TTL-bridge, not CP2102 2) The clone has only 1 LED and it's on GPIO2 3) GPIO9/10 are still wired to the Flash and are unavailable for use. -- Everything else has worked exactly as I'd expect it to and neither Nr.1 or Nr.2 matter code-wise.

me-no-dev, 2016-01-22:
heck I unsoldered my module from the nodeMCU to swap it for faster QIO :D get me some clones

WereCatf, 2016-01-22:
@me-no-dev Sucks to be y.... err, I mean, "aw shucks!" ;)

The clones ain't expensive on eBay, $5 a pop -- the gentleman is free to help himself to some of them.

asetyde, 2016-01-23:
Sorry my language .. speak too many language can do a little confusion :)

About esp i founded some corrupted board ..therefore if i need 2000 nodemcu i can't test it a long , i must load code and ship ...

i try to find best quality , about difference ESP12D and ESP12E ?? you will see in my photo , 12d come from official producer

hallard, 2016-01-27:
@asetyde 
What's your clues on "good" NodeMCU" or even "ESP12" modules? I need to produce safe sensors boards for customers and just lost one day trying to understand why my sketch was random crashing (and I didn't noticed this before)
I tried GDB because I though it was my code and saw nothing coherent, crashed sometime on SDK calls or other, never the same, and never same time. No way to resolve this issue until I decided to change NodeMCU boards. And did the trick.

That said, what the best deal to have real and good ESP8266 boards? Despite the price, I don't care, I just want things to work for customers and forgot headaches. I was thinking maybe buying dedicated from SparkFun thing, AdaFruit Huzzah, Olimex ESP8266 or even digistump OAK. What's your though?

Anyway, may be a flash test sketch may help to find bad flash ;-)

Got no more idea right now, just thinking ESP8266 is really good, but we can't rely on Flash quality, and this is a real issue !!! I've got lot's from different provider but can I trust them? that's the deal!!

asetyde, 2016-01-27:
reply from nodemcu team :  

Hi, Alexandro

Very glad that you reach out to me.
about the nodemcu clones, I can really do little thing. sorry for that.
since the hardware is opensource, everyone can copy and make it.
some clones use cheaper chips such as ch340, 
some is just an identical clone...only different quality and test maybe.

there is something I can confirm, 
nodemcu board flash is in DIO mode, to use 2 additional GPIO.
because to our firmware, lua code running, no noticeable speed difference.

the module from nodemcu, we choose good quality part. hope to stay at good quality, and no issue from hardware, 
so we can have fun with the software. 

Best Regards
Zeroday 
(Guijie Hong) Chinese name:)

asetyde, 2016-01-27:
We try to understand ... 

AI thinker is chinese but design is open source , therefore difference can be founded on quality (about different power and usb chip ) or on test quality ... about this if i retrive other information i will write

@hallard  Probably we are future competitors on IOT , i'm glad that esp is good therefore i choose good :)

hallard, 2016-01-27:
@asetyde 
Thanks for is useful information, 

Any idea how to detect clones (except the ch340, that is a good chip by the way), My concern is really also on the quality of ESP12E (or new ESP12F) chip (especially flash chip inside), how to know (without desoldering it) if ESP12 module is clone, good or potentially bad one?
Any sketch idea or process to verify if we can use DIO and GPIO9/10 ?

I've got some [Wemos D1](http://www.wemos.cc/wiki/doku.php?id=en:d1_mini) and I'm really happy with them because they are so small and cheap, with 3 different headers ulsoldered, nice, for just $4 and seems with ESP12F, seems to be DIO Mode but GPIO9/10 are not exposed.

By the way, do you know where to buy for sure, official and real NodeMCU ?

Thanks

asetyde, 2016-01-28:
I not use Wemos because it's too small , nodemcu is best choice at moment .
About buy i'm waiting amica team reply.

Also we can use ESP.x functions to retrive data but a good seller can resolve it without many problems

closed 2017-01-20 by asetyde

marcelstoer, 2017-12-27:
Sorry to comment on an old and closed issue. Back in 2015 I collected information about the NodeMCU devkits. At https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/#official-vs-unofficial there's a photo which shows in detail how an "official" 1.0/v2 devkit has to look like.

To my knowledge neither Banggood nor Aliexpress still sells the Amica v2 devkit. Seeed Studio still has them but it's marked as "out of stock". If you [search taobao.com for "NodeMCU cp2102"](https://s.taobao.com/search?q=NodeMCU+cp2102) it turns up quite a few that look genuine.

WereCatf, 2017-12-27:
Meh. I don't see any reason to care, unless one really, really wants to be able to use those GPIO9 and GPIO10. Personally, I'd choose the QIO-mode flash over being able to use the two GPIO. Other than that, though, whether it's an "official" one or a clone is irrelevant.