esp8266/Arduino #1482: Unable to bring up ArduinoOTA on an esp8266 which is also an AP

rDr4g0n, 2016-01-22:
I am attempting to have an esp8266 function as a wireless AP and also be updatable via OTA, but this does not seem to work. If the esp connects to an existing AP, OTA works as expected, but if the esp brings up its own AP, OTA fails with:  

	[begin] roundedSize:       0x0003F000 (258048)
	[begin] updateEndAddress:  0x00100000 (1048576)
	[begin] currentSketchSize: 0x0003F000 (258048)
	[begin] _startAddress:     0x000C1000 (790528)
	[begin] _currentAddress:   0x000C1000 (790528)
	[begin] _size:             0x0003EB40 (256832)
	Start
	Connect Failed
	Error[2]: Connect Failed
	premature end: res:0, pos:0/256832
	rror[4]: End Failed
	ERROR[0]: No Error

In the sample sketch below there is a function `beginAP` which brings up an access point, and a function `connectToAP` which connects to an existing access point. The sketch below works as expected, but if `connectToAP` is commented out and `beginAP` is uncommented, the issue appears.

    #include <ArduinoOTA.h>

    const char* ssid = "clownshoes";
    const char* password = "clownshoes";
    const char* ota_hostname = "clownshoes";
    IPAddress apIP(192, 168, 1, 1);

    // starts an access point
    void beginAP(const char* ssid, const char* password, IPAddress ip){
        //WiFi.mode(WIFI_AP);
        WiFi.mode(WIFI_AP_STA);
        WiFi.softAPConfig(ip, ip, IPAddress(255, 255, 255, 0));
        WiFi.softAP(ssid, password);
        Serial.print("AP is up at ");
        Serial.println(WiFi.softAPIP());
    }

    // connects to an access point
    void connectToAP(const char* ssid, const char* password){
        Serial.print("");
        WiFi.begin(ssid, password);
        while(WiFi.status() != WL_CONNECTED){
            Serial.print(".");
            delay(500);
        }
        Serial.print("\nConnected to '");
        Serial.print(ssid);
        Serial.print("' and got IP ");
        Serial.println(WiFi.localIP());
    }

    // starts OTA server
    void beginOTA(const char* hostname){
        ArduinoOTA.setHostname(hostname);
        ArduinoOTA.onStart([]() {
            Serial.println("Start");
        });
        ArduinoOTA.onEnd([]() {
            Serial.println("\nEnd");
        });
        ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
            Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
        });
        ArduinoOTA.onError([](ota_error_t error) {
            Serial.printf("Error[%u]: ", error);
            if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
            else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
            else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
            else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
            else if (error == OTA_END_ERROR) Serial.println("End Failed");
        });
        ArduinoOTA.begin();
        Serial.print("OTA is up");
    }

    void setup(){
        Serial.begin(9600);
        //beginAP(ssid, password, apIP);
        connectToAP(ssid, password);
        delay(500);
        beginOTA(ota_hostname);
    }

    void loop(){
        ArduinoOTA.handle();
    }

If this is merely a configuration issue on my side, it may be beneficial for it to be documented.

Thanks for the work! This sdk is killer :D

me-no-dev, 2016-01-22:
what version of the core are you using? The sketch looks fine and should by all means work

rDr4g0n, 2016-01-22:
I pull from master before I do any work, so whatever the latest is right now.

me-no-dev, 2016-01-22:
wait... are you starting AP with the same name?

me-no-dev, 2016-01-22:
i got the Serial to 115200 and it all started working. 9600 is waaay too slow :)

rDr4g0n, 2016-01-22:
ill bring up an ap with the same name on my phone so the esp can connect (to test), but take down the phone's ap so the esp can use the same ssid. This is purely for testing as simply as possible.  

Ill try faster serial tonight and see if that does it.

rDr4g0n, 2016-01-23:
OK, just tried it with faster serial and it is still failing. Anything else I can try or anyway to get more details for debugging?

me-no-dev, 2016-01-23:
since you know how to use git, can you try my fork of this repo? I just changed how serial works so it should be OK even on lower speeds. But other than that I have no other clues. I use OTA daily and is working fine (I do not debug it though, progress is visible in terminal)

rDr4g0n, 2016-01-24:
I pulled your fork and recompiled with that code with no success. I also removed all Serial logging from my code and that didn't work either. Ill try to add some better debug messages to the OTA class and see if that yields any insight.

rDr4g0n, 2016-01-24:
I'll go ahead and close this issue until I can produce some better repro steps or evidence that it is an actual issue and not a problem on my end.

closed 2016-01-24 by rDr4g0n

sandrolutz, 2016-04-03:
Did you find a solution for this issue?
I have exactly the same problem now.

gweep, 2016-06-27:
I am able to upload OTA in AP mode.  My code is similar to yours.    To get this to work I need to re-connect to my ESP8266 before each upload as my computer re-connects to my home network when the the ESP8266 reboots after the upload.   Then I also need to re-select the ESP8266 network port each time in the Arduino IDE;  Arduino->Tools->Port->Network Ports->"myESP8266 at XXX.XXX.XXX.XXX".

A bit inconvenient but it works every time for me.

zacharydrew, 2019-03-19:
This might fix it: [PR 5894](https://github.com/esp8266/Arduino/pull/5894)